weewx/weewx #2: Sensor failed event  (SF: #14 )

alco28, 2015-02-18:
Add an event SENSOR_FAILED or something similar, to signal problems with a sensor or station.

[Note: pulled from sourceforge issue 10 ]( https://sourceforge.net/p/weewx/tickets/14/ )

alco28, 2015-02-18:
Should then result in a long sleep and retry. (comment from Tom Keffer)

tkeffer, 2016-11-27:
I'm not sure where this issue came from. Do we really want an event? Or, an exception? If the former, what would be the consumer, and why does it need to consume an event instead of an exception? If the latter, we already have similar exceptions:

`weewx.WeeWxIOError`. If raised, weewx will wait 60 seconds, then restart.
`weewx.HardwareError`. This causes program termination.

I'm not seeing the utility, so I'm closing this unless someone says otherwise.

closed 2016-11-27 by tkeffer