weewx/weewx #5: WU Rapidfire updates may not work for "partial record" stations (SF: #30 )

alco28, 2015-02-18:
For stations that do not create full LOOP packets, the WU's RF protocol may not work.
May have to bundle the partial records together for larger, full posts.

[Note: pulled from sourceforge issue 10 ]( https://sourceforge.net/p/weewx/tickets/30/ )

alco28, 2015-02-18:
Comment from Tom Keffer:

The stations that are most affected by this have long intervals between LOOP records. This means it could take a half a minute or, possibly, much longer, to accumulate enough packets to do a full WU post, thus negating the whole point of RF.

I'm setting this to 'deferred', but if there are stations out there with rapid partial LOOP packets, I'm open to reopening the ticket.

matthewwall, 2015-04-12:
acurite stations have 2 partial packets read at 18 seconds each, so the result would be a rapidfire packet every 18 seconds.  each packet would contain half observations that are current and half that are 18 seconds old.

oregon scientific stations emit packets every 14 seconds for wind and 60 seconds for other sensors.

of course if rapidfire is being deprecated by weather underground then there might be no point in doing this.

W0CHP, 2015-04-13:
> On Apr 12, 2015, at 5:02 PM, matthewwall <notifications@github.com> wrote:
> of course if rapidfire is being deprecated by weather underground then there might be no point in doing this.
> 

Is the RF deprecation something being planned by WU

I ask since I’m working on debug=2 for restx.py patches for logging POST/GET packets. If WU is planning on making changes, I can focus on other protocols.

tkeffer, 2015-06-28:
Merged with issue #31

closed 2015-06-28 by tkeffer