weewx/weewx #9: Temperature differences should be a separate unit (SF #8 )

alco28, 2015-02-18:
Taking the difference of two temperatures really results in a new unit, because the function needed to convert it changes. FtoC(x-y) is not equal to FtoC(x) - FtoC(y).

alco28, 2015-02-18:
### Matthew Wall:

i must not understand the problem here. in the world i know, addition and subtraction cannot change units. only multiplication or division can change units. perhaps we need to be more explicit about 'units' and 'scaling'.

for example, if a is degree_F, b is degree_C, and y is degree_C, then this is a perfectly valid operation:

y = a + b

you should also be able to do this where x is unspecified units:

x = a + b

x would get units of K, F, or C, depending on the defaults, but the quanta of x is temperature.

if r is meters and s is seconds, then

v = r / s

results in a variable with units of speed, i.e., quanta are LENGTH/TIME, which could be expressed in mph or meters per second or whatever the default units of speed.

making differences into new units would mean that you need new units for every kind of difference one might make, for example, megabyte - kilobyte. no?

alco28, 2015-02-18:
### Matthew Wall:

quanta - the building blocks of any unit system. for example, fundamental quanta include (a default unit of measure for each is in parentheses):

LENGTH (meter)
TIME (second)
MASS (gram)
TEMPERATURE (kelvin)
CURRENT (ampere)
QUANTA (none)
LUMINOSITY (none)
CURRENCY (dollar)
ANGLE (radian)
SOLID_ANGLE (steradian)

units - measurement of quanta, e.g., meter, pascal, joule, foot, inch, second

scaling - a unitless multiplier, e.g., kilo, mega, peta, Ki, K

alco28, 2015-02-18:
### Matthew Wall:
On one level you are, of course, right, but unfortunately, within the weewx architecture, temperature is different because conversions do not involve a simple scaling.

Let F_to_C be a function for converting F to C. That is, C = (F-32)*5/9

So, what's the difference between 68F and 50F in C?

It's not F_to_C(68-50), which would be -18C.

It's F_to_C(68) - F_to_C(50), which is 10C.

So, I have to use a different conversion function for the difference, than I do for a single number. In that sense, taking the difference introduces a new "unit."

Within weewx I don't know how else to treat this quirk except by introducing another unit. Any ideas?

CantConfirmOrDeny, 2015-10-17:
This issue, of course, arises from the fact that the classical method of converting back and forth from F to C cleverly involves _two_ conversions - one to account for the fact that C degrees are bigger than F degrees ( 1 C° = 1.8 F° ), plus one to account for the fact that the zero point is different on the two scales. This propery is unique to temperatures - every other unit conversion we're familiar with shares the same zero point, e.g., in the length dimension, zero millimeters = zero light years. Those kind of conversions are a simple matter of applying a linear conversion factor. If temperatures had the same zero point, i.e., if 0° F = 0° C, we wouldn't have this issue.

Well, as @matthewwall sort of hinted at, the "correct" units for temperatures in engineering calculations are based on absolute zero: we use the Kelvin scale for Celsius degrees, and the Rankine scale for Fahrenheit. This way, the conversion between Kelvin and Rankine is a simple, linear one, just like it is with any other common dimension, like ounces to kilos. But, to take advantage of that, we'd have to convert all our data and store all our temperatures internally as K° or R°.  That might be a big job.

There's one other possibility - the F° and C° scales do share a common point where they're equal, at -40°. To take advantage of this little quirk, we would "normalize" all our temperatures by adding 40, regardless of whether they're F or C, it works either way. All temps would be stored this way, and worked with internally in this format. Conversions would then be a simple matter of multiplying C° by 9/5 (or F° by 5/9). Anytime a temperature needs to be sent to an output, whether it's F or C, you simply subtract 40 to to get the "real", human-readable temperature.

In short, there's no super-simple way to handle this. I've used the normalization approach before, and it works well so long as you're not doing anything excessively scientific. You can usually tell when you're looking at one of these "normalized" temperatures, too - it's always obvious with degs. C, perhaps a bit less so with degs. F, but usually obvious from the context. I think the solution is not to introduce a new unit, but a new algorithm. Or, we create a  "temperature object" for working with temperatures that knows how  to deal with these things.

tkeffer, 2015-10-17:
Oh, dear God. Storing temperature with a 40 degree offset would make the "rainRate" wars look like a minor skirmish. :-)

CantConfirmOrDeny, 2015-10-17:
Yeah, I imagine so.  So, that leaves us with a "temperature object", or just hard wiring the algorithm depending on the need, as I suppose is done now.  I'm new around here, so don't really have a full appreciation for all the implications of any proposed solution in this space.  Is there a particular place in the code where this is a notable issue? Might be fun to play around with.

tkeffer, 2015-10-18:
We've found workarounds for most problems, but one place it comes up is the `trend` tag. For example,

```
$trend.outTemp
$trend.barometer
```

Both work, but if you try

```
$trend.outTemp.degree_C
$trend.barometer.mbar
```

only the latter works.

A minor annoyance, but an example of where it comes up.

See the file `tags.py`, class `TrendObj`.

tkeffer, 2016-11-30:
The present system is working well. The benefits of being able to add late unit conversion (like the example above) are too minimal to warrant the complexity. Closing for now.

closed 2016-11-30 by tkeffer