weewx/weewx #13: $current tag should report all observations in a LOOP packet

matthewwall, 2015-03-11:
the $current tag should include all observations, not just those in the archive table schema

tkeffer, 2015-03-24:
This would involve passing in the record to the reporting thread, and use that instead of the most recent record in the database. This is certainly possible.

My only hesitation is that it won't work for wee_reports, which has to rely on the database. 

A work around is to use the database when possible, but if a value cannot be found there, then use the passed in record. If there is no passed in record, return None.

tkeffer, 2016-11-25:
Fixed in commit https://github.com/weewx/weewx/commit/c4ca8b65b6a50da099b1956d47e6b17a7d7ff8fb, although I should note this fix makes all observations in an *archive record* (not LOOP packet) available. I assume that was a typo in the header for this issue.

closed 2016-11-25 by tkeffer