weewx/weewx #17: implement cron-like report generation

matthewwall, 2015-03-24:
Look at Gary's (?) changes to allow cron-like report generation.

(from the TODO file)

gjr80, 2016-02-20:
Just wondering if there is still an appetite for cron like report generation. Attached email exchange is the last correspondence I have on the topic.
[Re weewx implement cron-like report generation (#17).txt](https://github.com/weewx/weewx/files/139156/Re.weewx.implement.cron-like.report.generation.17.txt)

To refresh, I developed some additional code that used 3 parameters that could be specified in `skin.conf`/`weewx.conf` against each report being generated. The parameters were `generate_dom`, `generate_dow`, `generate_time` that allowed the user to specify the day of month, day of week and time that a report will be generated. The parameters were cumulative and each could include multiple comma seperated values. I believe the net effect is to mimmick the standard cron file format but in a 'wordy' manner.

To summarise the state of the code. After the 1st cut of the code I refactored and put the logic that determines whether to generate the report or not in a seperate method (in `weeutil.py` for want of a better place) and added about 10 extra lines to `cheetahgenerator.py`. The code was soak tested for about 6 weeks on a simulator system running 25 odd reports testing various combinations of the 3 parameters and I believe it to operate without error.

I am not sure what weewx version I tested on but given it was April-May 2015 I am guessing 3.1, definitely v3.x. I don't know that there has been any great change to weewx report generation since this time, so I expect the code will still be functional. If there is still a desire to implement this I will go ahead and revalidate the code and submit.

tkeffer, 2016-02-21:
If it does not dramatically increase the complexity of the code and user experience, then I think there is still interest.

davies-barnard, 2016-02-21:
Hi,

Is there scope to include this in some refactoring of the stdreport/restful
sections so that skins are not compulsory etc?   All my ISSWatch, METWatch
and Spacewatch extensions would benefit from only being run once in every 1
or less.

Chris

gjr80, 2016-02-21:
@tk. Ok, I don't believe it does dramatically increase the code complexity. The complexity is in the parsing of the 'cron like parameters' but that is pushed down to a seperate method. The core report generator code (method name escapes me) is not substantially altered. I will integrate ithe code into a current clone, redo the soak testing and offer up a pull request.

@cdb. I am not overly familiar with your extensions, and I could not find them given a quick search this morning. CheetahGenerator already supports a stale_age parameter which runs a reports every x seconds. Maybe that is appropriate.

gjr80, 2016-02-21:
Ok, found `ISSWatch` (much easier when not on the ipad); its a SLE. By definition an SLE is called during the process of generating each report from a report template. The SLE is called for each report in the skin irrespective of whether each report uses the SLE results or not. If the core issue here is to prevent the SLE being called one or more times during each report cycle then that is significantly beyond what I am proposing. I would see a couple of options:

1. A single skin with a single report utilising the SLE with an appropriate `stale_age` setting. Any other reports/pages requiring the output generated by the skin can do it through and include statment. Somewhat inflexible but suitable for small scale projects.

1. Implement the SLE as a service rather than an SLE. The service gathers/processes its data every so often and stores it in its own database. Then create an SLE that pulls data from the database and use the SLE in any report that requires you data. This is exactly what the forecast extension does (think Matthew posted about this recently). I have done it with weewx-wd, its really very easy as weewx has all the API tools there for you, the only real code development is gathering/processing your data and it seems you have already essentially done that. Overall more work but far more flexible in its end use.

tkeffer, 2016-02-22:
There's no reason an SLE can't do a lazy initialization.

-tk

gjr80, 2016-02-22:
Hmm, today's python lesson I guess. I'm keen to understand how this would work in this context given my 15 minutes of knowledge of lazy initialization. Would this mean that you could have a list of SLEs in a given skin and instead of each being called once for each report in the skin, the SLE would (effectivley) only be called as many times as there are reports in the skin that use one or more of the SLE's products?

tkeffer, 2016-02-22:
No, it's way simpler than that.

I'm assuming that the goal is to run the compute heavy stuff in ISSWatch,
etc., only once an hour. So, you save the last time you ran it, and check
to see if an hour has passed before running again.

Something like this:

class ISSWatch(SearchList):
  time_of_last_calc = 0

  def get_extension_list(self, timespan, db_lookup):
    if time.time -
​ISSWatch.​
time_of_last_calc > 3600:
      run_expensive_calculation()

​ISSWatch.​
time_of_last_calc = time.time()
​       ​
       carry on...

OTOH, if the goal is to generate the HTML files from the templates only
once an hour, then you'd need something like the facility that you're
proposing.

​Or, maybe I'm misunderstanding the problem completely!

-tk
​

gjr80, 2016-02-22:
Ok, am I missing something obvious here. Isn't each report cycle run in its own thread and wouldn't each cheetahGenerator instance create its own instance of the SLE concerned? ie is `time_of_last_calc` persistent across all report threads

davies-barnard, 2016-02-22:
Thanks for putting your great minds to my simple problem.

My approach would have been similar to FTP with a last checked stamp but that might not have been the most appropriate.

For what it is worth,  I like the idea of your cron like report generation.  

Chris Davies

Sent from a mobile device.

tkeffer, 2016-02-22:
Good point. The SLE would have to store time_of_last_calc in something more
persistent, such as user.extensions, or a database or file.

-tk

gjr80, 2016-02-29:
I have integrated my old 3.1 code into a current clone and it is soak testing for a few weeks; running over a 29 February should be a good test. In the meantime I sat down to document the 3 new options. Having worked through most of the documentation I now find myself questioning how the options should work. The three options I have implemented are

- `generate_time` - specify time of day to generate file(eg 00:05, 01:00)

- `generate_dow` - specify day of week to generate file(eg Monday, Wednesday)

- `generate_dom` - specify day of month to generate file (eg 1, 10, 21)

Each of the options can be blank, a single value or a comma seperated list of values. So for each template specified in `skin.conf` you could have something like:

    [[[report1]]]
        template = report1.tmpl
        generate_time = 01:00, 02:00
        generate_dow = Monday
        generate_dom = 1,20

The logic I have implemented is such that to work out whether or not a given report will generate at a given time (let's call it `report_time`) 3 key conditions are considered. These are:

1.  (`generate_time` is specified AND `generate_time` contains `report_time`) OR (`generate_time` is not specified)

2.  (`generate_dow` is specified AND `generate_dow` contains the day of week represented by `report_time`) OR (`generate_dow` is not specified)
    
3.  (`generate_dom` is specified AND `generate_dom` contains the day of month represented by `report_time`) OR (`generate_dom` is not specified)

If 1 is true then the report will be generated if either 2 or 3 is true ie 1 AND (2 OR 3). So to use the settings in the example above `report1` will be generated at 01:00 and 02:00 each Monday as well as at 01:00 and 02:00 on the 1st and 20th of each month.

When I first mentioned my idea to Tom (back around mid-late 2014) I think an issue was raised on sourceforge as I rememeber another user commenting that we should look at using a cron/crontab format for specifying when to generate. I think Tom said something along the lines of _however we do it, it needs to be straightforward and simple for the user_ though I am going by memory here as I can't find any emails. This exchange always stuck in my mind and my doubts came came along when I looked at the basic logic behind cron/crontab. My understanding of how crontab is interpreted is that each field that is non-* must match the current date/time for the cron event to fire, so if the example I just used was worked through with 'crontab logic' the report would generate at 01:00 and 02:00 on 1st of the month if it is a Monday and at 01:00 and 02:00 on the 20th of the month if it is a Monday. I know we do not have to implement the same logic as cron/crontab (it is cron like after all not cron exact). I guess those *nix users that are experieced with cron/crontab may have issues with cron/crontab using one set of logic and weewx using something different but to me I see more utility in the current logic I have implemented (ie OR'ing 2 and 3) rather than using cron/crontab logic (ie AND'ing 2 and 3).

Would appreciate thoughts/guidance.

tkeffer, 2016-02-29:
My `man` page for `crontab` says

> Commands  are  executed by cron(8) when the minute, hour, and month of year fields match the current time, and when at least one of the two day fields (day of month, or day of week) match the current  time.

which, I believe, is exactly the logic you're proposing.

gjr80, 2016-02-29:
Hmm, helps when you read the right source... [red face]

gjr80, 2016-03-02:
Noting #108 I have been considering how to apply what I have done so far to the FTP and RSYNC generators. The code I have in `CheetahGenerator` by necessity hooks into the `generate` method so it can handle report generation on a template by template basis. Key in deciding whether to generate or not is knowledge of the report generation time; this is either `gen_ts` (if a specific report time was requested eg `wee_reports`) or the last good archive timestamp (if we are in the normal report cycle). The issue with FTP and RSYNC generators is that they know of `gen_ts` but have no concept of last good archive timestamp; hence we have problems during the normal report cycle. 

I also note that `FtpUpload` uses `time.time()` for recording the time an FTP upload occurred; makes sense since it is threaded. So I guess I could use `time.time()` myself rather than last good archive timestamp but that seems a bit of a fudge. Not sure if it will present some issues with RSYNC since it does not store a 'time of last RSYNC'. I know how to make the FTPand RSYNC generators last good archive timestamp aware but fear I may be overcomplicating things here.

Any advice?

davies-barnard, 2016-03-02:
Given that we are opening the skin conf files on a regular basis can they not be used to store such values?   ConfigObj will do writes.  Or does this go against removing the need for skins in FTP and RSYNC.

gjr80, 2016-03-02:
I would see these type of settings used in the [[FTP]] and [[RSYNC]] stanzas in weewx.conf, primarily because the skin.conf for these skins are basically a 2 line generator spec and secondly as that is where all the other FTP and RSYNC settings reside. For template based skins I see these settings as being no different to any other similar template based setting (eg stale_age) in that they can be specified in skin.conf or overridden in weewx.conf.

tkeffer, 2016-03-02:
I had assumed that generating was an all-or-nothing affair. That is, you either run file and image generation, along with FTP and RSYNC, or you run nothing.

It would be risky to allow users to pick and choose which service to run, because that could lead to inconsistent results. For example, what if the copy generator did not run at all? Or, if FTP ran, but nothing had been generated?

With this in mind, the logical place to put the generate decision is in `StdReport`. There, you have access to the event which triggered the generate, along with the time of the archive record.

gjr80, 2016-03-02:
Understand what you say but I guess the case I always had in my mind is number of files being generated every archive period but a small number of other reports that are either time consuming to generate or slow changing being generated say once every hour or day. Stale_age goes part of the way but you cannot nail down when it runs, only the interval between successive generations. I would also expect an all or nothing approach would make #108 a little difficult.

I had no plans to touch copy generator. Would FTP running with no generated files just result in nothing being FTP'd (if no files generated ever) or the previously generated version being FTP'd? Maybe wasteful of bandwidth but should not lead to any issues.

tkeffer, 2016-03-02:
I see your point. But, pushing back, a common use case is users with expensive WAN connections. They want to archive data often, but FTP / RSYNC only occasionally. 

I guess there's three places to hook this functionality into:

1. At the Report level, as I described above. This would help those with expensive WAN connections and would be very simple to implement. Don't see why it would not solve #108.
2. At the generator level. Here you decide whether or not to fire off a generator. This could be done in `StdReportEngine`. This would also be easy to implement. 
3. At the individual file / image / FTP / RSYNC level. This seems to be what you're proposing, but, as you're discovering, there's no point of commonality between the various generators that you can hook into. Each generator would have its own implementation of the crontab-like functionality, carefully engineered so they would all have the same semantics. Not including the copy generator is probably ok. 

The problem with Option 3 is that it is subtly different from the real crontab. If the real crontab finds a match, it goes and does something. By contrast, in our world, we go do something then check to see if it's OK. This requires each "do something" facility to do its own checking to see if it's OK. 

I'm rambling.

Sorry, but I have to go... I'll try to finish my thoughts later.

gjr80, 2016-03-03:
You are correct about #108, I should learn not to reply at midnight!

Sat down and looked as `StdReport`, `StdReportEngine`, the generators and your thoughts to date, understand you still had more thoughts to get write down but I wonder if perhaps a basic hybrid arrangement as outlined below might allow us to deal with the 2 cases we seem to have.

- Control whether `StdReportEngine` is kicked off by a parameter under `[StdReport]`.
    - let's say there was a `report_time` and/or `report_stale_age` parameter sitting under `[StdReport]`
        - `report_time` is a comma separated list of times of day
        - `report_stale_age` is a period of time in seconds.
    - `StdReport` uses these to determine whether to kick off the `StdReportEngine` thread
        - it fires if the report generation time (ie last good archive timestamp) is in a list of designated times (`report_time`) or
        - it fires every `report_stale_age` seconds
    - don't implement 'days of the week' or 'days of the month' as I have with the `CheetahGenerator` code because the case we are dealing with here does not require it
    - could implement one or the other or both of `report_time` or `report_stale_age`
    - this gives us the all or nothing effect to meet the 'constrained WAN' case

- Control `CheetahGenerator` using the `generate_time`, `generate_dow`, `generate_dom`
    - can be set in `skin.conf` or overriden in `weewx.conf` as per any other `skin.conf` parameter
    - only applies to `CheetahGenerator`
    - fires as per CRON rules
    - this meets the 'generate a resource hungry report less frequently' case
    
- If required, control `ImageGenerator` using the `generate_time`, `generate_dow`, `generate_dom`
    - easily doable, very similar to `CheetahGenerator`
        - would entail about 20 extra line of code in `ImageGenerator`
        - would reuse the core method used in `CheetahGenerator` to determine whether to generate or not
    - arguable as to whether there is much need, though I will guarantee that someone will ask for it

- `CopyGenerator` is left unchanged, it runs each archive period and copies what is available

Do we get any issues/unexpected results?
- If `report_time` and `generate_time` are both in use I guess there is a possibility nothing would be generated ever (highly unlikely) or fewer Cheetah reports would be seen than expected unless the two parameters were deconflicted. Could be mitigated by:
    - some logging in the start up preamble
    - some goods words in the manual(s)
    - leaving the parameters out of the weewx install so the user must consciously include them presumably once they have read the manual (otherwise how do they know to include them)
- if both are omitted then weewx functions as it does now
- if either `report_xxxx` or `generate_xxxx` is used then the user should get exactly what they ask for, no surprises
- `generate_xxxx` parameters are standard `skin.conf` type parameters so can be overridden from `weewx.conf` so that should preclude their use messing up a weewx extension if the user decides to use them with a weewx extension (think this may have been what Chris was getting at). `report_xxxx` parameters are in `weewx.conf` so should not interfere with any weewx extensions.
- `StdReport` implementation is quite simple since it only checks time of day so I don't think software bloat will be an issue
    
I hope I have not railroaded your thoughts, at least this was written during normal daylight hours so hopefully a bit better thought out.

gjr80, 2016-04-08:
Tom, did you ever see my last comment? I think I published it around the time the was a lot of traffic between you and Luc re Luc's changes to the Vantage driver and you appear to have been away a bit since then. Not sure if it may have been lost in the tide of emails.

tkeffer, 2016-04-10:
Haven't forgotten you.

I'm not enthusiastic about having multiple levels of choice in whether to kick something off. Taking an analogy from crontab, it's as if crontab not only allowed you to kick off processes at certain times of day, but also whether crontab got run at all.  

Perhaps we should be thinking of `StdReportEngine` as a kind of crontab. It runs every reporting cycle (that is, with every new archive interval). It goes through the list of possible reports, comparing some yet-to-be-determined set of parameters against the archive time, and decides whether to run the given report.

So, what of the "yet-to-be-determined" parameters? Would `generate_time`, `generate_dow`, `generate_dom` cover all, or at least, 80%, of the use cases? These would be specified in a hierarchical manner, just like other reporting parameters: they would get specified under `[StdReport]`, but could get overridden under individual reports.

There is still the issue of things that want to run more often than once every archive interval, but I don't think `StdReport` is the place to put that.

gjr80, 2016-04-11:
Ok, have been trying to get a clear picture of this in my head. Current (normal) operation is for `StdReportEngine` to launch every archive period. `StdReportEngine` loops through all of the `[[Xxxxxx]]` reports defined in `[StdReport]` and for each report (among other things) launches the appropriate sequence of generators defined in `skin.conf` for the report concerned.

From your description above, some "yet to be determined" parameters are optionally specified under `[StdReport]` or under report `[[Xxxxxx]]`. As per weewx convention any such parameter under `[[Xxxxxx]]` would override the same parameter specified under `[StdReport]`. Report `Xxxxx` would only be 'generated' if the "yet to be determined" parameters say 'fire'.

Following on from this, there would be no "yet to be determined" parameters in the report `skin.conf` so our control over generation would only be on a report by report basis rather than on a template by template basis (and at present only in one particular generator - cheetahgenerator).

So thinking about the use cases. Our user who has the 'expensive' WAN connection and who wants to archive all the time but only wants to generate now and then would set his "yet to be determined" parameters at the [StdReport] level. The user who has some reports that he wants generated at certain times would set  his "yet to be determined" parameters at the `[[Xxxxxx]]` level. The user who has a one or more templates that he wants generated at certain times would need to place only the template(s) concerned in one or more skins and then set  his "yet to be determined" parameters at the `[[Xxxxxx]]` level for the report(s) (skin(s)) concerned. 

One interesting situation came to mind involving overriding, we could have some parameter(s) at the `[StdReport]` level that say 'do not fire this time around', one would expect the whole report cycle would be skipped. Though if there were some parameter(s) at the `[[Xxxxxx]]` level that say 'do fire this time around', that report would be generated. Whilst it follows the weewx convention, it seems sort of counterintuitive to me.

I have also been thinking of the "yet to be determined" parameters. I had been using `generate_time`, `generate_dow`, `generate_dom` and they work fine. I am wondering though if a cron like line might not be better. I liked the former because they are fairly self evident whereas a cron like line is not. Whilst the 3 parameters can specify most of what you can do with a cron like parameter, doing something like 'every 10 minutes' is quite lengthy (would require `generate_time` with 144 csv; self evident but hardly user friendly). I guess not everybody will use this capability, perhaps then it is acceptable to expect those that do want to use it should be able to understand something like cron lines which are a linux standard.

tkeffer, 2016-04-11:
Pretty much what I'm thinking.

Because `FTP` is a separate "report," for those with expensive WAN connections, it would be fired off only occasionally.

A nice side benefit is that any custom report generators could also be controlled this way.

With this strategy you lose per template control, but that would require that every report generator have some knowledge about the cron facility. 

Personally, I find the standard `crontab` syntax easy to understand, but it may not translate well into the ConfigObj-style options. Give it a try and see what it feels like.

gjr80, 2016-04-11:
Yes, FTP could be controlled at report level. I think I have the outline clear in my mind, will see where it takes me.

gjr80, 2016-04-14:
Making some good progress, parsing a CRON line is somewhat easier than I thought!
I thought I recalled an earlier comment that one of the advantages of working in `StdReportEngine` was that the 'timestamp of the report was available' (which is not the case in all the generators when working at the generator level). Looking back I see that comment actually said `StdReport` rather than `StdReportEngine`. As far as I can see the only time when the 'timestamp of the report' is available in `StdReportEngine` is when run from `wee_reports`, otherwise the 'last good database time' is used and that is determined further down at generator level. Have I missed something obvious here? I am working in the `StdReportEngine` `run` method and whilst I have been able to obtain the last good database time it takes a few lines of code to get it.

tkeffer, 2016-04-14:
It's worse than that. With the introduction of database bindings, there is no one database to look at to figure out the "last good timestamp."

Perhaps this is a good opportunity to rethink what we mean by "report time."

The intention here is to kick off a reporting cycle in response to a `NEW_ARCHIVE_RECORD` event, so I'd say the timestamp of that event is what we want.

In most cases it will match the timestamp of the last `WX_BINDING` archive record, but not always. If reports care about that, they should use the database timestamp closest to the report time, perhaps with some "grace" period.

gjr80, 2016-04-14:
I had been using the default `wx_binding` which I can see may not not always be used by the generator. 
Since `StdReport` is bound to `POST_LOOP` are you suggesting a binding to `NEW_ARCHIVE_RECORD` to simply obtain the timestamp? As far as I can tell you get nothing passed in with `POST_LOOP`, just a 'hey the main loop is broken', or do I have a fundamental misunderstanding here?

tkeffer, 2016-04-15:
Hmmm. Good point. Let me think about this one...

-tk

tkeffer, 2016-04-17:
I think the thing to do is have `StdReport` intercept `NEW_ARCHIVE_RECORD` events, and then save the record. It can then pass the record on to `StdReportEngine`.

The default should be `None`, meaning there is no current record. This would be the case when `wee_reports` is being used. In this case, it should fall back to the present behavior (read from the database).

This all feels like a bit of a kludge, but I can't come up with anything better. At least it has the advantage of addressing issue #13 as well.

gjr80, 2016-04-17:
Ok, do you have a view on whether `wee_reports` should honour the "yet to be determined" parameters or should it generate irrespective? I could see reasons for going either way, though I suspect that the case for honouring it would be somewhat weaker than for when generating reports in the normal weewx cycle.

tkeffer, 2016-04-18:
I'd say that if you fire up `wee_reports`, you should get the full reports. It would be very counter intuitive to get some subset, especially if that subset depended on not obvious things like the time of day, or the time of the last archive record.

-tk

gjr80, 2016-04-18:
I have put together a working 'CRON like report' system which has passed my preliminary tests. Given the path taken to get here I thought I would post the code for comment rather than come out with a pull request. Attached (with .txt extensions - hope they can still be easily read) are modified `engine.py` (bind `StdReport` to NEW_ARCHIVE_RECORD), `reportengine.py` (modified run method) and `cronparse.py` (defines `ReportCron` class ie logic to decode/parse a CRON line). I have placed the `ReportCron` class in its own file as I was unsure of the best place to put it - reportengine, weeutil a new file ?

[cronparse.txt](https://github.com/weewx/weewx/files/223851/cronparse.txt)
[engine.txt](https://github.com/weewx/weewx/files/223850/engine.txt)
[reportengine.txt](https://github.com/weewx/weewx/files/223852/reportengine.txt)

Changes summary:
`engine.py`
- added binding to NEW_ARCHIVE_RECORD and associated method to `StdReport`
- added `record` parameter to pass current archive record to `StdReportEngine`

`reportengine.py`
- added `ReportCron` import
- added `record` to \_\_init\_\_
- added code (lines 131-165) to detemine whether to run the current report or not

`cronparse.py`
- `ReportCron` class definition

As with many things it appears that CRON is implemented slightly differently across some Linux flavours. I have documented the report cron syntax used under the `ReportCron` class in `cronparse.py`.

I have run this under 3.4.0 and the current master and the limited testing seems to show it as well behaved. `wee_reports` runs without problem and runs all reports irrespective of `report_cron` settings.

Questions/concerns/issues:
- I am not sure if the logic in `reportengine.py` that controls whether to run the report or not is the best. It feels a little back-the-front as I use the `is_triggered` property to decide to 'not run'. But then I am using a CRON format (which essentially says 'fire now') to determine when to not do something - so I am starting off bac-to-front I guess.
- where should `ReportCron` class be homed.
- logging, I have tried to keep it simple and have only added a couple of debug log calls; one call if a report is skipped due to a `report_cron` setting and secondly if an invalid `report_cron` setting is detected. Too much, too little?
- naming, you have probably worked out I am not the most creative when it comes to naming .conf parameters etc is `report_cron` acceptable.

Still to do:
- more testing of `ReportCron` class to ensure it decodes/parse CRON lines as it should. The other code is very straightforward, the key is in interpreting the CRON line.
- test setting `report_cron` 'up and down' the .conf tree to make sure it behaves as it should.
- test the syslogging/exception code to make sure it picks up and reports what it should.
- documentation.

gjr80, 2016-04-18:
Forgot to say that format for the `report_cron` setting in a .conf file is 5 space seperated values as per a normal CRON line:
report_cron = Minutes Hours Days of Month Months Days of Week

tkeffer, 2016-04-18:
Good progress! Do you have a repository where we can take a look?

gjr80, 2016-04-19:
Forgot the obvious, my weewx [fork ](https://github.com/gjr80/weewx) should have the latest code.

gjr80, 2016-04-23:
Continuing to make progress, I think I have the bugs out of the `ReportCron` class now so that it properly interprets a CRON line. 

I am playing around with the `report_cron` setting (the setting that holds the CRON line) in `weewx.conf`. It works as expected. I see that `StdReportEngine` has access to the skin dict, so I thought that I would be able to access a `report_cron` setting at `CheetahGenerator` level in `skin.conf` and thus support setting the `report_cron` parameter at `CheetahGenerator` level in `skin.conf`. I can see `report_cron` in `skin_dict` under `CheetahGenerator` but I am not sure if I am breaking any `weewx` conventions by reaching down and grabbing it. I don't want to reach down to template level, but thought I should be able to reach down to `CheetahGenerator` level. Sorry, but my understanding of the detailed operation of `.conf` settings and overrides is fundamental at best.

matthewwall, 2016-04-23:
from a conf point of view:  one should be able to specify one report_cron at [StdReport], and that would be used by every section within [StdReport].  then one could specify a report_cron for any one section with [StdReport], and that would override whatever is defined at the upper level.

the root level report_cron should be the backward-compatible default, i.e., run the report on each archive record.

the method reportengine.ReportGenerator.__init__ should take care of reading/settingup report_cron.  that way all of the other generators get it automatically by invoking their super __init__ in their own __init__

btw, i would use a more descriptive term for the parameter, such as 'report_timing' instead of 'report_cron'.  then the docs would read "use the report_timing parameter to control when the report is run.  the report_timing parameter uses cron-like syntax."

or am i misunderstanding this discussion thread?

matthewwall, 2016-04-23:
excellent work on the documentation, btw

matthewwall, 2016-04-23:
`report_cron = 22 * * * *`

would cause the report to be run during the report cycle occurring at 25 minutes past each hour. 

should be '22 minutes past each hour'

gjr80, 2016-04-23:
> or am i misunderstanding this discussion thread?

Just to summarise what I am doing. I have code in `StdReportEngine.run` that is a big switch - it either runs a report or does not run a report. The decision to run a report is based upon a 'CRON like parameter' that is specified at `[StdReport]` to apply to all reports/sections or under individual `[StdReport]` sections to affect just that report/section. The 'CRON like parameter' is compared to the data from the latest archive record and a decision is made to either run the report or skip it. There is no code in any of the generators to in any way act on the 'CRON like parameter'. `wee_reports` ignores the 'CRON like parameter'.

When I saw that the skin dict was available in `StdReportEngine.run` I thought there may have been a simple means to specify the 'CRON like parameter' in say `[CheetahGenerator]` in a `skin.conf` and have the new code in `StdReportEngine.run` make the decision whether to run the report or not. Having thought about this a little more, and having a better understanding of how *.conf settings are fed up and down, I see there is little benefit in doing this.

> btw, i would use a more descriptive term for the parameter, such as report_timing instead of report_cron. then the docs would read "use the report_timing parameter to control when the report is run. the report_timing parameter uses cron-like syntax."

I have said many times I have no great imagination for naming things, more than happy to change the parameter name.

> report_cron = 22 * * * *
would cause the report to be run during the report cycle occurring at 25 minutes past each hour.
should be '22 minutes past each hour'

Actually, no I meant 25. CRON has a 'resolution' of 1 minute, weewx reports effectively have a resolution of 'archive period' minutes. If, as most people (seem to) do, you have a 5 minute archive period what should we do if the 'CRON like parameter' says generate a report at 22 minutes after the hour? If we only consider the timestamp of the last archive record then we would not generate the report. The approach I have taken is to look at the period since the previous report was generated (assumed to be the interval field in the latest archive record) and check whether any of the times in that period would have triggered report generation. So for the case in hand (I did actually specify 5 minute archive period in the draft documentation) the new code would check to see if minute 25, 24, 23, 22 or 21 matches the 'CRON like parameter' and if **any** of them do, the report is run. Easy enough to change the logic to only consider the archive record timestamp if the decision is to go that way.

gjr80, 2016-04-28:
Have completed extensive testing of the CRON like line parser and I am happy it is interpreting CRON like lines as it should. The comments in class `ReportCron` in `cronparse.py` give the best description of what CRON functions are supported. Have exercised the `report_timing` parameter at different 'levels' in `weewx.conf` on a simulator system and it works as expected. Have controlled the FTP report successfully, but did not test rsync - though I have no expectation it will be any different as it is just another report. Have also exercised the logging/error reporting code. It does what I want and I believe it is succinct and accurate.

Documentation has had a few revisions. I have not reviewed it since my last round of changes to the code, that will be the task for the next few days.

There are still a few outstanding issues:
- parameter name. Have called it `report_timing` based on Matthew's input.
- where to put the `ReportCron` class. For no particular reason I put the class in `user/cronparse.py`, clearly not suitable for release. Do we add it to `reportengine.py`, some other file or its own better named/located file?
- what happens for trigger times that fall between 'report times'. What do we do if the `report_timing` paramater is, say `22 * * * *` on a 5 minute period system? Does the report ever run? I have coded so that each minute in the the period from the current report time (inclusive) to the last report time(exclusive) (ie archive period minutes ago) is checked for a match and if one is found the report is run. In this case the report would be run during the report thread kicked off around 25 past the hour only.

closed 2016-05-15 by tkeffer