weewx/weewx #21: Doing 'setup.py install --extension' should prompt for required parameters

matthewwall, 2015-03-24:
in v3.1 the behavior is to inject required parameters
   into weewx.conf.  for passwords and tokens, inject INSERT_XXX_HERE.  when
   service starts up, detect missing or default passwords/tokens and bail out
   with a log message (as the restful services do).

if the installer prompts for the parameters required by an extension then user would not have to do the extra step of modifying the configuration file.

tkeffer, 2015-04-28:
Can we closet this one out?

matthewwall, 2015-04-29:
one way to do this would be to add methods to the ExtensionInstaller class analogous to the methods on the AbstractConfEditor:

default_stanza
get_conf
prompt_for_settings

default_stanza and get_conf are not really appropriate for extensions - a single extension might have multiple config stanzas, not just one.  they work for a driver, since it will only ever have one stanza.

closed 2015-07-13 by tkeffer