weewx/weewx #22: Guard against database locks

matthewwall, 2015-03-24:
locks are still showing up in weewx v3.1.0.  the most repeatable case is on a slower system with forecast reports when the forecast database is configured to retain all forecasts (this is not the default configuration).  since the forecast database is not indexed, the queries can take a long time.

i have also seen locks on a 2012-ish dell dual-processor machine with terabytes of spinning disk on the weewx.sdb database (data since 2009 - around 2 million records).  in that case the locks caused weewx to die when it tried to write weather data to the database (weewx v2.6.4).

this is from 24 march 2015 with weewx 2.6.4:

Mar 23 04:55:52 sailing weewx[19489]: wxengine: Caught unrecoverable exception in wxengine:
Mar 23 04:55:52 sailing weewx[19489]:     ****  database is locked
Mar 23 04:55:52 sailing weewx[19489]:     ****  Traceback (most recent call last):
Mar 23 04:55:52 sailing weewx[19489]:     ****    File "/opt/weewx-2.6.4/bin/weewx/wxengine.py", line 963, in main
Mar 23 04:55:52 sailing weewx[19489]:     ****      engine.run()
Mar 23 04:55:52 sailing weewx[19489]:     ****    File "/opt/weewx-2.6.4/bin/weewx/wxengine.py", line 189, in run
Mar 23 04:55:52 sailing weewx[19489]:     ****      self.dispatchEvent(weewx.Event(weewx.POST_LOOP))
Mar 23 04:55:52 sailing weewx[19489]:     ****    File "/opt/weewx-2.6.4/bin/weewx/wxengine.py", line 210, in dispatchEvent
Mar 23 04:55:52 sailing weewx[19489]:     ****      callback(event)
Mar 23 04:55:52 sailing weewx[19489]:     ****    File "/opt/weewx-2.6.4/bin/weewx/wxengine.py", line 562, in post_loop
Mar 23 04:55:52 sailing weewx[19489]:     ****      self._catchup(self.engine.console.genArchiveRecords)
Mar 23 04:55:52 sailing weewx[19489]:     ****    File "/opt/weewx-2.6.4/bin/weewx/wxengine.py", line 625, in _catchup
Mar 23 04:55:52 sailing weewx[19489]:     ****      origin='hardware'))
Mar 23 04:55:52 sailing weewx[19489]:     ****    File "/opt/weewx-2.6.4/bin/weewx/wxengine.py", line 210, in dispatchEvent
Mar 23 04:55:52 sailing weewx[19489]:     ****      callback(event)
Mar 23 04:55:52 sailing weewx[19489]:     ****    File "/opt/weewx-2.6.4/bin/weewx/wxengine.py", line 574, in new_archive_record
Mar 23 04:55:52 sailing weewx[19489]:     ****      self.archive.addRecord(event.record)
Mar 23 04:55:52 sailing weewx[19489]:     ****    File "/opt/weewx-2.6.4/bin/weewx/archive.py", line 192, in addRecord
Mar 23 04:55:52 sailing weewx[19489]:     ****      e))
Mar 23 04:55:52 sailing weewx[19489]:     ****    File "/opt/weewx-2.6.4/bin/weedb/__init__.py", line 130, in __exit__
Mar 23 04:55:52 sailing weewx[19489]:     ****      self.connection.commit()
Mar 23 04:55:52 sailing weewx[19489]:     ****    File "/opt/weewx-2.6.4/bin/weedb/__init__.py", line 97, in commit
Mar 23 04:55:52 sailing weewx[19489]:     ****      self.connection.commit()
Mar 23 04:55:52 sailing weewx[19489]:     ****  OperationalError: database is locked
Mar 23 04:55:52 sailing weewx[19489]:     ****  Exiting.

here is another example from a raspberri pi:

Jul 30 08:55:22 hummingbird weewx[959]: wxengine: Caught unrecoverable exception in wxengine:
Jul 30 08:55:22 hummingbird weewx[959]:     ****  database is locked
Jul 30 08:55:22 hummingbird weewx[959]:     ****  Traceback (most recent call last):
Jul 30 08:55:22 hummingbird weewx[959]:     ****    File "/home/weewx/bin/weewx/wxengine.py", line 962, in main
Jul 30 08:55:22 hummingbird weewx[959]:     ****      engine.run()
Jul 30 08:55:22 hummingbird weewx[959]:     ****    File "/home/weewx/bin/weewx/wxengine.py", line 189, in run
Jul 30 08:55:22 hummingbird weewx[959]:     ****      self.dispatchEvent(weewx.Event(weewx.POST_LOOP))
Jul 30 08:55:22 hummingbird weewx[959]:     ****    File "/home/weewx/bin/weewx/wxengine.py", line 210, in dispatchEvent
Jul 30 08:55:22 hummingbird weewx[959]:     ****      callback(event)
Jul 30 08:55:22 hummingbird weewx[959]:     ****    File "/home/weewx/bin/weewx/wxengine.py", line 562, in post_loop
Jul 30 08:55:22 hummingbird weewx[959]:     ****      self._catchup(self.engine.console.genArchiveRecords)
Jul 30 08:55:22 hummingbird weewx[959]:     ****    File "/home/weewx/bin/weewx/wxengine.py", line 625, in _catchup
Jul 30 08:55:22 hummingbird weewx[959]:     ****      origin='hardware'))
Jul 30 08:55:22 hummingbird weewx[959]:     ****    File "/home/weewx/bin/weewx/wxengine.py", line 210, in dispatchEvent
Jul 30 08:55:22 hummingbird weewx[959]:     ****      callback(event)
Jul 30 08:55:22 hummingbird weewx[959]:     ****    File "/home/weewx/bin/weewx/wxengine.py", line 574, in new_archive_record
Jul 30 08:55:22 hummingbird weewx[959]:     ****      self.archive.addRecord(event.record)
Jul 30 08:55:22 hummingbird weewx[959]:     ****    File "/home/weewx/bin/weewx/archive.py", line 192, in addRecord
Jul 30 08:55:22 hummingbird weewx[959]:     ****      e))
Jul 30 08:55:22 hummingbird weewx[959]:     ****    File "/home/weewx/bin/weedb/__init__.py", line 130, in __exit__
Jul 30 08:55:22 hummingbird weewx[959]:     ****      self.connection.commit()
Jul 30 08:55:22 hummingbird weewx[959]:     ****    File "/home/weewx/bin/weedb/__init__.py", line 97, in commit
Jul 30 08:55:22 hummingbird weewx[959]:     ****      self.connection.commit()
Jul 30 08:55:22 hummingbird weewx[959]:     ****  OperationalError: database is locked
Jul 30 08:55:22 hummingbird weewx[959]:     ****  Exiting.

tkeffer, 2015-03-24:
Version 3.x already has a guard against this: any database OperationalError exception is caught in the main thread, resulting in a 120 second wait, then a retry.

Do you have an example from a 3.x installation where it's failing?

matthewwall, 2015-03-24:
i just updated the weewx 2.6.4 installation to weewx 3.1.0.  same extensions and databases.  it was failing about once per month starting in january.  so i'll let it run with 3.1.0 to see what happens.

tkeffer, 2015-04-28:
Any update on this?

matthewwall, 2015-04-28:
on the arm installation: no failures since i pruned the forecast database.  pruning the forecast database also stopped the periodic 'segfault' failures i was getting on the arm installation with a ws2080 and a whole bunch of uploaders.

no more locks on the dual processor x86 machine since upgrading from 2.7.4 to 3.1.0 on 24 march 2015.

tkeffer, 2015-04-28:
Can we just document this, then close this issue?

matthewwall, 2015-04-29:
summary of the situation:

when 'database locked' messages appear, it is time to figure out why there is contention for the database.  it might be due to really long database query times, for example when the forecast table gets really big, since it has no index.  it might be due to slow disk, for example on rpi system with slow or failing sd card.

solution is to eliminate the contention.

closed 2015-04-29 by matthewwall