weewx/weewx #31: consider a StdCache service

matthewwall, 2015-04-23:
this service would retain the value for observations so that full packets would be available even if the hardware produces only partial packets.

the service would have a 'max_age' setting to specify how long it would retain values.  it might need a configuration option to indicate which observations it should retain and which it should not.  for example, gauge measurements (e.g., outTemp, inHumidity) could be safely cached, cumulative measurements (e.g., rainTotal) could be safely cached, but incremental measurements (e.g., rain) probably should not be cached.

the service would be inserted just before the StdWXCalculate service.

when would this service be useful?  for any hardware that produces partial packets, this service would make services that expect a full packet work properly, e.g., the wu rapidfire uploads.  of course, for that example the wu rapidfire uploader could be extended to do caching.

would this service be useful anywhere else?

probably do not want this service installed by default for all hardware?

tkeffer, 2015-04-23:
A good idea.

I don't think it's a problem including the service for those stations which emit "full" records (is the Vantage the only station which does this?).

One thing we should explore is its interaction with the StdArchive service, which manages the accumulators. It would have no way of knowing which observations are "real" observations, and which are cached observations and would give them all equal weight.

I can see a couple of ways around this.

1. The record that included cached observations would store them under a different name in the event. So, event.record would be the observed values, event.cached would be the observed+cached values.

2. Introduce a new event NEW_CACHED_PACKET. The StdCache service would create it. Any interested service, such as StdRESTful, would subscribe to it. 

The 2nd approach feels right to me, but I may be missing something...

tkeffer, 2015-05-18:
I realized that we already have the makings of a caching service: the accumulators in StdArchive. They hold the time of the last update for each loop observation.

I created a branch `cache` with an implementation. It has a new event, `UPDATED_ACCUMULATOR`, created whenever the accumulators are updated by a new LOOP packet.

The accumulator has a new method, `get_most_recent()` that takes a single argument, a dictionary with the "stale times." This is how long an old observation is accepted.

I put in an implementation for RapidFire that uses this. The config file, weewx.conf, allows the stale times to be specified:

```
[StdRESTful]
  [[Wunderground]]
    [[[Stale]]]
      barometer = 60
      outTemp = 60
      ...
```

If the `[[[Stale]]]` stanza is missing, all the times default to 60 seconds.

Take a look at the branch [cache](https://github.com/weewx/weewx/tree/cache). I think this is too late for V3.2 (which we really need to get out), but could be the basis for V3.3

tkeffer, 2015-05-19:
Oops. Not so fast. I just realized that my strategy has a fatal flaw: the cache is effectively cleared with every archive record. So, it cannot cache data across an archive interval boundary.

Back to the drawing boards...

tkeffer, 2015-05-19:
OK, reimplemented, this time with an explicit StdCache service.

matthewwall, 2016-04-16:
the cache implementation does not quite work.  i ran it with an acurite bridge as the data source and wu as the sink.  

1) first i ran the cache branch with non-rapidfire updates to wu.  this worked as usual.  

2) then i ran cache branch with rapidfire updates but no StdCache in the services lists. as expected, this failed to upload anything to wu, since wu rapidfire in cache branche requires cache events.  

3) then i added StdCache to process_services after StdCalculate with rapidfire. this did not work as intended - values in the cache were replaced by None values from the calculations, since StdWXCalculate uses LOOP not cache.

4) then i moved StdCache to just before StdWXCalculate.  in this configuration, values were properly cached, but no derived values were in the cache (windchill, etc) since those are done in StdWXCalculate.

perhaps we need to make stdWXCalculate aware of cached packets and use those as its inputs instead of LOOP?  but that introduces some spaghetti dependencies between LOOP and cache.

i made StdPrint aware of cached packets so that it will print them as well as loop and archive.

tkeffer, 2016-04-16:
Hmmm. An unintended side effect of having a separate `LOOP_CACHE_UPDATED` event.

We do not want to archive data derived from the caches, but we want to continue to archive "real" data. So, effectively, there are two data streams, each of which has to be fed through `StdWXCalculate`. 

Sounds like a lot of complications just to support Rapidfire. Right now, I'm not seeing a way around it.

We could just go back to the original idea: have the RF code do its own, simple, cache. But, that would mean it has to calculate any derived variables that it needs (are there any?).

matthewwall, 2016-04-16:
it looks like only one field is derived: dewptf

it could easily use the dewpoint calculation formula in wxformulas.py.  that one bit of duplication would be a lot less complexity that a caching service.

btw, the 'rtfreq' value is hard-coded to 2.5 in restx.py.  the value 2.5 is only valid for a davis station.  the AmbientLoopThread should track the timing of get_record invocations and adjust the value of 'rtfreq' accordingly.

tkeffer, 2016-04-16:
Agreed.

Re: the value of `rtfreq`. This is just following the [WU documentation](http://wiki.wunderground.com/index.php/PWS_-_Upload_Protocol) for the Rapidfire protocol. Frankly, I think it gets ignored.

brentpabst, 2016-05-04:
Just curious if anyone had made any decisions one way or another as to the best implementation path to continue down?

matthewwall, 2016-05-04:
given that wu rf is currently the only part of the system that is broken (wrt partial packets), i think the best approach is to make the wu service do its own caching. we could coopt the meaning of 'stale' for the wu rf case to mean 'oldest cached observation age' instead of 'record age'.  the wu rf implementation would need to track the age of each observation, not just the age of the record.

tkeffer, 2016-05-04:
I agree with Matthew.

Modifying the Rapidfire code in `restx.py` to include a cache would make a good intermediate-level Python project, should someone want to give it a try!

matthewwall, 2016-06-18:
i put tom's cache object in weeutil, then used that to cache both loop packets and archive records for wunderground.

https://groups.google.com/forum/#!topic/weewx-user/U-rr-AwGRcY

one problem with caching and not handling partial packets is that you get no indication of sensor problems until the cache expires.  this is one reason that i think caching should be done by (or at least specified by) whatever is doing the retention and/or calculations (i.e., weather underground).

for example, by using cached values to fake full packets, you won't see evidence of a broken sensor on wunderground until the cache expires.

tkeffer, 2016-06-18:
You are reminding me why I prefer stateless designs. :-)

matthewwall, 2016-06-18:
so a few more problems with caching, specifically as it relates to creating full packets from partial packets:

which observations should be cached?  for example, if you cache 'rain' (or any delta observation) you'll get an incorrect result, but if you cache 'rain_total' (or any accumulated value) its ok.  gauge observations are ok too, such as temperature, pressure, humidity.

should 'None' values be cached?  how to distinguish between "no value for the observation because the sensor was not sampled", "no value because there was an error reading the sensor", or "no value because that sensor does not exist".

i supposed we could make every driver always emit full packets, even if the hardware emits partial packets.

tkeffer, 2016-06-18:
> i supposed we could make every driver always emit full packets, even if the hardware emits partial packets.

All that does is push the caching problem into the driver.

The accumulators in weewx have had to deal with this problem. They solve it by recognizing there's a difference between intensive variables, such as temperature, and extensive variables such as rain. For the former, it uses the average, for the latter it uses the sum or, in some cases, the last value.

I hate to reinvent all this stuff. Perhaps the accumulators could be adapted to be used for the Rapidfire protocol?

matthewwall, 2016-11-02:
caching is now implemented for wu rapidfire.  this addresses the following issues:

- partial packets.  once it fills up, the cache looks like an full packet.  so any hardware that emits partial packets should now work with rapidfire.

- long sensor refresh.  on some hardware with partial packets, one sensor might be refreshed much less often than other sensors, even beyond an archive interval.  for example, on wmr300, the pressure is updated every 15 minutes.  the cache retains those values so this hardware works with rapidfire.  (this is an extreme case of the partial packet problem)

- dealing with None.  the cache object ignores any incoming value that is None; it prefers to retain a previous non-None value.

there are two parts to the None issue: (1) wu does not deal well with None values, and (2) weewx is not consistent about its use of None.

please consider the following to deal with (2):

2a: driver should insert an observation name *only* if there is a known value for that observation.  None is a valid known value, but only if the value is known to be undefined.  for example, if a sensor is known to be reporting a bad value, then None is an appropriate value for that sensor and it should be included in the packet/record.  if the driver does not know whether a sensor exists, then it should *not* include that sensor in the packet/record, and it should definitely not include that sensor with a value of None.

2b: the StdWXCalculate service should insert a value of None for a derived value *only* if it has values for all of the independent variables on which derived depends.  for example, if a packet contains temperature=None and humidity=50 then a dewpoint=None is appropriate.  if a packet does not contain temperature but contains humidity=50, then dewpoint=None is not appropriate; there should be no dewpoint in the packet.

the behavior for StdWXCalculate is different from what we recently implemented to address issue #139.

tkeffer, 2016-11-02:
(Consistent treatment of `None` moved to Issue #175)

tkeffer, 2016-11-27:
I believe this issue has been resolved. Closing.

closed 2016-11-27 by tkeffer