weewx/weewx #33: failures in archive_interval result in no retries

matthewwall, 2015-05-02:
this has happened on ws23xx and fo hardware.  it happens regularly with ws23xx when there is a lot of RF interference (even with shielded cables and ferrite cores in place).  it could happen to any hardware with a logger.

the StdArchive service tries to get the archive interval from the hardware using the archive_interval property.  when archive_interval fails while querying the hardware, weewx dies.

so far the workaround is to disable any logging capability - both catchup and record_generation=hardware.

desired behavior would be for the engine to reload the driver and start over (as it does when other hardware communication failures happen)

tkeffer, 2015-05-11:
The engine cannot possibly know all the different kinds of hardware exceptions that could be raised. Instead, it catches a generic `WeeWxIOError` exception. 

It's up to the driver to convert whatever hardware error that occurs into an `WeeWxIOError` exception. Do you know what exception (or error code) is happening in the ws23xx and fo drivers?

matthewwall, 2015-05-12:
i changed the ws23xx driver so that every known/expected failure will come through as an exception derived from weewx.WeeWxIOError.  will test this for awhile to see if that solves the problem.

tkeffer, 2015-05-12:
Make sure you push your changes --- I'm not seeing them on GitHub.

matthewwall, 2015-05-13:
committed at 41fdba1

closed 2015-05-13 by tkeffer