weewx/weewx #34: vantage driver crash

matthewwall, 2015-05-16:
weewx crashed yesterday with an input/output error.  i thought the designed behavior was to retry a few times then reload the driver.

weewx: 3.1.0
driver: vantage
weather station: vantage pro 2 model 6152C w/weatherlink data logger
os: Ubuntu 12.04.5 LTS
uname: Linux sailing 3.2.0-69-generic #103-Ubuntu SMP Tue Sep 2 05:02:14 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux
uptime:  22:58:15 up 203 days,  4:59,  1 user,  load average: 1.95, 1.73, 1.56

stack trace:

May 14 16:40:26 sailing weewx[30353]: engine: Caught unrecoverable exception in engine:
May 14 16:40:26 sailing weewx[30353]:     ****  (5, 'Input/output error')
May 14 16:40:26 sailing weewx[30353]:     ****  Traceback (most recent call last):
May 14 16:40:26 sailing weewx[30353]:     ****    File "/opt/weewx-3.1.0/bin/weewx/engine.py", line 837, in main
May 14 16:40:26 sailing weewx[30353]:     ****      engine.run()
May 14 16:40:26 sailing weewx[30353]:     ****    File "/opt/weewx-3.1.0/bin/weewx/engine.py", line 187, in run
May 14 16:40:26 sailing weewx[30353]:     ****      for packet in self.console.genLoopPackets():
May 14 16:40:26 sailing weewx[30353]:     ****    File "/opt/weewx-3.1.0/bin/weewx/drivers/vantage.py", line 431, in genLoopPackets
May 14 16:40:26 sailing weewx[30353]:     ****      for _loop_packet in self.genDavisLoopPackets(200):
May 14 16:40:26 sailing weewx[30353]:     ****    File "/opt/weewx-3.1.0/bin/weewx/drivers/vantage.py", line 445, in genDavisLoopPackets
May 14 16:40:26 sailing weewx[30353]:     ****      self.port.wakeup_console(self.max_tries, self.wait_before_retry)
May 14 16:40:26 sailing weewx[30353]:     ****    File "/opt/weewx-3.1.0/bin/weewx/drivers/vantage.py", line 66, in wakeup_console
May 14 16:40:26 sailing weewx[30353]:     ****      self.flush_input()
May 14 16:40:26 sailing weewx[30353]:     ****    File "/opt/weewx-3.1.0/bin/weewx/drivers/vantage.py", line 200, in flush_input
May 14 16:40:26 sailing weewx[30353]:     ****      self.serial_port.flushInput()
May 14 16:40:26 sailing weewx[30353]:     ****    File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 497, in flushInput
May 14 16:40:26 sailing weewx[30353]:     ****      termios.tcflush(self.fd, TERMIOS.TCIFLUSH)
May 14 16:40:26 sailing weewx[30353]:     ****  error: (5, 'Input/output error')
May 14 16:40:26 sailing weewx[30353]:     ****  Exiting.

tkeffer, 2015-05-16:
Yes, it should retry a few times, but in order to do so, the error has to be caught. Unfortunately, the nature of the error is a bit of a mystery to me. It doesn't look like a Python exception was raised. Instead, best I can tell, a generic "errno" error occurred, where the global value of errno was set to 5, then the program exited. 

Do you have any insights?

matthewwall, 2015-06-01:
in the two years or so that this station has been running, this is the second time weewx has failed inexplicably.  i neglected to get the stack trace from the first failure, and i have no idea what might have caused this one.

here are the relevant software versions for reference:

uptime: 11:20:04 up 219 days, 17:21,  1 user,  load average: 1.23, 1.45, 1.43
os: Ubuntu 12.04.5 LTS
uname -a: Linux sailing 3.2.0-69-generic #103-Ubuntu SMP Tue Sep 2 05:02:14 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux
python: 2.7.3
python-serial: 2.5-2.1build1
weewx: 3.1.0

tkeffer, 2015-06-28:
Tracked this one down. Module `serial` can raise an undocumented `termios` exception of type `termios.error`. The Vantage driver now catches these exceptions and converts them into a `weewx.WeeWxIOError` exception, thus allowing retries.

Fixed in commit ef403ae3eb468b28e2ae5948a59bd04121d22153

closed 2015-06-28 by tkeffer