weewx/weewx #40: design a mechanism for defining report defaults in StdReport

matthewwall, 2015-06-06:
leaf nodes in StdReport propagate to reports, but there is not yet a mechanism to propagate non-leaf nodes.  for example, if i want to apply a [Units] section to every report, i must copy/paste the [Units] to every report; if i define a [[Units]] section just under [StdReport], the Units section is treated as a report.

proposed solution: define a [StdReport][[defaults]] section.  anything in the [[defaults]] stanza will be the fallback/default value for every report.

why is this necessary?  to make it easier and less error-prone to apply units across all reports.  to reduce the amount of copy/paste required for labels when using multiple sensors (e.g., renaming pond temperature).  to minimize the size of the StandardReport (and whatever new, adaptive, responsive report we define as the successor to StandardReport).

tkeffer, 2015-06-06:
I think you're on the right track. My only pause is that StdReport refers to specific report generation mechanism, that supplied by the class `StdReport` (and associated thread `StdReportEngine`). 

There are other reporting facilities, such as [`mesowx`](https://bitbucket.org/lirpa/mesowx), which do not use `StdReport`.

matthewwall, 2016-05-03:
try this on for size:

```
[ReportDefaults]
    [[Units]]
        [[[Groups]]]
        [[[StringFormats]]]
        [[[Labels]]]
        [[[TimeFormats]]]
        [[[Ordinates]]]
        [[[DegreeDays]]]
        [[[Trend]]]
    [[Labels]]
        [[[Generic]]]

[StdReport]
    ... # as before
```

when filling in a skin dict, the StdReport plumbing would start with sane defaults, then override with everything from skin.conf, then override with everything found in ReportDefaults, then override with everything in the skin instance section of StdReport.

this exposes some odd naming/organization choices:

- why is 'Almanac' all by itself in skin.conf?
- why is there a 'Generic' section for Labels?  seems extraneous.
- imho, the 'TimeFormats', 'Ordinates', 'DegreeDays', and 'Trend' don't really belong in 'Units'

matthewwall, 2016-05-03:
oops, i got the order of populating skin dict wrong.  it probably should be this:

1. sane defaults
2. [ReportDefaults]
3. skin.conf
4. [StdReport][[instance]]

matthewwall, 2016-12-07:
one gotcha to avoid: stanza named DEFAULT appears to have special meaning in ConfigObj

http://www.voidspace.org.uk/python/configobj.html

don't know if that is specific to configobj 5.  i assume that anything in DEFAULT is used for any child stanza, but i'll need to do some testing (docs i've seen are not obvious about how DEFAULT behaves)

matthewwall, 2017-02-28:
the newskin branch now has an implementation of this.

for example, you can now do this:

```
[StdReport]
    [[default_options]]
        [[[Units]]]
            [[[[Groups]]]]
                temperature = degree_C
    [[Standard]]
        ...
        [[[Units]]]
            [[[[Groups]]]]
                temperature = degree_F
```

this makes it possible to specify units once in default_options and have them apply to all reports.

i also modified the weecfg stuff to do exactly that.

in my admittedly minimal testing (and unit tests), it seems to work as designed.

matthewwall, 2017-02-28:
on the plus side, this was only a matter of some 20 lines of code.

on the negative side, it does not address tom's point about being specific to StdReport.

matthewwall, 2019-01-03:
this issue has been assimilated into issue #248

closed 2019-01-03 by matthewwall