weewx/weewx #41: Refactoring hot to specify ROOT directories.

tkeffer, 2015-06-11:
This is the continuation of a conversation started in commit 493badcf040391ca115414d931f46b57b31ff901

Thinking a bit more about this, moving all ROOT options to the top of weewx.conf may not be such a good idea.

By definition, `CONFIG_ROOT` is where weewx.conf is located, so specifying it in the config file is redundant and could introduce an inconsistency.

`SKIN_ROOT` and `HTML_ROOT` are defined in the context of `StdReport`, so I don't see any upside to moving them to global scope.

The location of `BIN_ROOT` could be useful. It would saving having to search the file system for the weewx Python packages. Perhaps it should be renamed "`LIB_ROOT`"?

That leaves `SQLITE_ROOT`, which could be defined in the `[Databases]` stanza. The present option `root` goes away:

```
[Databases]
    # This section defines the actual databases

    # Where the SQLite databases reside:
    SQLITE_ROOT = /var/lib/weewx

    [[archive_sqlite]]
        database_name = weewx.sdb
        driver = weedb.sqlite
```

alternatively, we introduce a `[SQLite]` (and `MySQL`) stanza:

```
[SQLite]
  SQLITE_ROOT = /var/lib/weewx
  driver = weedb.sqlite

[MySQL]
  host = localhost
  user = weewx
  password = weewx
  driver = weedb.mysql
```

I like the latter approach. It's a more elegant way of introducing new databases.

matthewwall, 2015-06-11:
using [SQLite] and [MySQL] stanzas has a well-defined upgrade path; we can automatically convert existing weewx.conf files.

so the default weewx.conf would look like this:

[Databases]
    [[archive_sqlite]]
        database_name = weewx.sdb
        driver = weedb.sqlite
    [[archive_mysql]]
        database_name = weewx
        driver = weedb.mysql

and a user could override the sqlite database location for a single database like this:

[Database]
    [[custom_sqlite]]
        database_name = /home/username/custom.sdb
        driver = weedb.sqlite

or like this:

[Database]
    [[custom_sqlite]]
        SQLITE_ROOT = /home/username
        database_name = custom.sdb
        driver = weedb.sqlite

or override mysql login credentials like this (still using the same 'localhost' defined in the [MySQL] stanza):

[Database]
    [[custom_mysql]]
        database_name = custom
        user = username
        password = pw
        driver = weedb.mysql

so the minimal definition for an extension (specified in install.py) would be this:

[Databases]
  [[cmon_sqlite]]
    database_name = cmon.sdb
    driver = weedb.sqlite
  [[cmon_mysql]]
    database_name = cmon
    driver = weedb.mysql

tkeffer, 2015-06-11:
Exactly.

One correction: on my earlier post, I proposed including the driver in the [SQLite] and [MySQL] stanzas. That won't work because, right now, the driver name is the only way to identify what kind of database it is. 

We can either do things the way you outline above, or introduce yet another level of indirection:

```
[Databases]
  [[cmon_sqlite]]
    database_name = cmon.sdb
    database_type = SQLite
  [[cmon_mysql]]
    database_name = cmon
    database_type = SQLite
```

where option `database_type` has to match a stanza name.

tkeffer, 2015-06-26:
Fixed in V3.2. With the addition of [SQLite] and [MySQL], the installer has become much simpler.

Closing issue.

closed 2015-06-26 by tkeffer