weewx/weewx #44: 3.2.0a4: no empty line before each report stanza in weewx.conf

matthewwall, 2015-07-06:
installing an extension should put an empty line before the extension's report in weewx.conf.

  1) clean install
  2) wee_extension --install forecast.tgz
  3) forecast stanza in StdReport does not have empty line before it

tkeffer, 2015-07-31:
Any new 2nd level stanza, not just in `StdReport`, will not have a line before it.

How important is this? It would require some substantial surgery to fix.

Frankly, the right way to fix this is to allow extensions to specify any additions to `weewx.conf` as a `StringIO`, rather than only as a dictionary. This would allow comments to be included.

matthewwall, 2015-08-01:
obviously this is not very importan.  i was testing the extension installation process for consistency.

closed 2015-09-02 by tkeffer