weewx/weewx #45: 3.2.0a4: inconsistent use of double quotes for station location

matthewwall, 2015-07-06:
1) clean install with location name 'whatever'
  2) wee_config --reconfigure
  3) location name no longer has double quotes around it

there is logic in the code for dealing with a location that has a comma in it, so there is no need to put the location in double quotes.

tkeffer, 2015-07-31:
I'm not sure there is much we can do about this. ConfigObj immediately converts any value, quoted or not, into a string, stripping off the quotes. Try this:

```python
import StringIO
import configobj

s = StringIO.StringIO('''[Station]
qlocation = "A quoted string"
slocation = An unquoted string
''')

c = configobj.ConfigObj(s)
print c
```

The results will be:
```
{'Station': {'qlocation': 'A quoted string', 'slocation': 'An unquoted string'}}
```

The good news is that, semantically, there is no difference whether the location is enclosed in quotes or not.

matthewwall, 2015-08-01:
thanks for the explanation.  the issue is not very important - it came up as i was diff-ing weewx.conf from before and after changes made by wee_config.

closed 2015-08-01 by matthewwall