weewx/weewx #47: should drivers load when modules they use are not installed?

matthewwall, 2015-07-06:
if you install using setup.py without python-serial or python-usb installed, the driver listing misses some drivers.  not much we can do about the ones that use usb, but i could push the serial imports to a lower level for those that use serial.  

do we want to make driver modules load properly even if they depend on python modules that are not installed?

tkeffer, 2015-07-06:
That seems like a good idea. Then if you do

```
wee_config --list-drivers
```

it would not only give the driver version number, but also whether it needs python-serial or python-usb, and whether it has been installed.

matthewwall, 2015-07-06:
i modified the weecfg driver utilities so that a driver scan lists any import failures.  this is what the output looks like on suse when pyserial and pyusb are not installed:

```
mwall@picosuse:~/src/weewx> wee_config --list-drivers
Module name              Driver name    Version  Status                   
  weewx.drivers.acurite    ?              ?        No module named usb      
  weewx.drivers.cc3000     ?              ?        No module named serial   
  weewx.drivers.fousb      ?              ?        No module named usb      
  weewx.drivers.simulator  Simulator      3.0                               
  weewx.drivers.te923      ?              ?        No module named usb      
  weewx.drivers.ultimeter  ?              ?        No module named serial   
  weewx.drivers.vantage    Vantage        3.0                               
  weewx.drivers.wmr100     ?              ?        No module named usb      
  weewx.drivers.wmr200     ?              ?        No module named usb      
  weewx.drivers.wmr9x8     ?              ?        No module named serial   
  weewx.drivers.ws1        ?              ?        No module named serial   
  weewx.drivers.ws23xx     WS23xx         0.24                              
  weewx.drivers.ws28xx     ?              ?        No module named usb
```

this would be more common on a setup.py installation than deb or rpm.  rpm installs will fail when pyusb and pyserial are not installed, but this can be overridden with the --nodeps option.    

any .py file in the user directory will be scanned as if it is a driver.  however, it will only show up in this list if a module it needs is not installed.  once it loads properly, it will be rejected as a driver since it does not define DRIVER_NAME.  so that is fairly future-proof.

so we do not need to change the current practice of early, global import in drivers.  and the vantage driver illustrates what happens when a driver can be use different ways - import of serial is deferred, so if you use the tcp/ip interface you need not install the serial module.

matthewwall, 2015-07-06:
tangential to this issue, but related: i updated wee_extension (in extension.py) to show more information in the listing of extensions.  this requires that extension authors define name, version, and description fields.

```
mwall@picosuse:~/src/weewx> wee_extension --list
Extension Name    Version   Description
forecast          3.0       Generate and display weather and tide forecasts.
```

closed 2015-07-13 by tkeffer