weewx/weewx #48: syntax for StdWXCalculate settings

matthewwall, 2015-07-07:
what syntax should we use for StdWXCalculate settings?

the default settings for StdWXCalculate look like this:
```
[StdWXCalculate]
    pressure = prefer_hardware
    barometer = prefer_hardware
    altimeter = prefer_hardware
    windchill = prefer_hardware
    heatindex = prefer_hardware
    dewpoint = prefer_hardware
    inDewpoint = prefer_hardware
    rainRate = prefer_hardware
```
however, there are other parameters that might need to be adjusted.  for example, the rainRate period, the algorithm for altimeter calculation, the algorithm and associated parameters for maximum theoretical solar radiation, evapotranspiration parameters.

here are some alternatives:

option 1: separate stanzas for calculations and algorithms
```
[StdWXCalculate]                                                        
    ignore_zero_wind = True                                             
    rain_period = 900           # for rain rate                         
    et_period = 3600            # for evapotranspiration                
    wind_height = 2.0           # for evapotranspiration                
    atc = 0.8                   # for solar radiation RS                
    nfac = 2                    # for solar radiation Bras              
    max_delta_12h = 1800                                                
    [[Calculations]]                                                    
        windchill = hardware                                            
        heatindex = prefer_hardware                                     
        dewpoint = software                                             
    [[Algorithms]]                                                      
        altimeter = aaASOS                                       
        maxSolarRad = RS                                                
```

option 2: parameters in a separate stanza
```
[StdWXCalculate]
    windchill = hardware
    heatindex = prefer_hardware
    dewpoint = software
    [[parameters]]
        ignore_zero_wind = True                                         
        rain_period = 900           # for rain rate                     
        et_period = 3600            # for evapotranspiration            
        et_wind_height = 2.0           # for evapotranspiration         
        rs_atc = 0.8                   # for solar radiation RS         
        bras_nfac = 2                    # for solar radiation Bras     
        max_delta_12h = 1800                                            
        altimeter_algorithm = aaASOS                                    
        maxSolarRad_algorithm = RS                                      
```

option 3: keep parameters with variables.  variable can be scalar (no params) or dict (with params).  if 'source' is not specified, then default to 'prefer_hardware'.
```
[StdWXCalculate]                                                        
    windchill = hardware
    heatindex = prefer_hardware
    dewpoint = software
    [[wind]]
        ignore_zero_wind = True
    [[rainRate]]
        rain_period = 900
    [[ET]]
        et_period = 3600
        et_wind_height = 2.0
    [[pressure]]
        max_delta_12h = 1800                                            
    [[altimeter]]                                                       
        source = prefer_hardware                                        
        algorithm = aaASOS                                              
    [[maxSolarRad]]                                                     
        source = software                                               
        algorithm = RS                                                  
        rs_atc = 0.8                   # for solar radiation RS         
        bras_nfac = 2                    # for solar radiation Bras     
```
option 4: flat namespace
```
[StdWXCalculate]
    ignore_zero_wind = True
    rain_period = 900           # for rain rate
    et_period = 3600            # for evapotranspiration
    et_wind_height = 2.0           # for evapotranspiration
    rs_atc = 0.8                   # for solar radiation RS
    bras_nfac = 2                    # for solar radiation Bras
    max_delta_12h = 1800  # for pressure
    windchill = hardware
    heatindex = prefer_hardware
    dewpoint = software
    altimeter_algorithm = aaASOS
    maxSolarRad_algorithm = RS
```

perhaps something else?

option 1 is currently implemented in 3.2.0, with backward compatibility for the non-grouped calculations.  but it is rather haphazard and should be changed, imho.

tkeffer, 2015-07-07:
Strongly prefer :three:.

But, can we please just get V3.2 out first? :worried:

matthewwall, 2015-07-07:
sorry, just making sure that i did not introduce something in the StdWXCalculate changes for 3.2 that we will have to undo in 3.3.

tkeffer, 2016-10-02:
Option 1 chosen.

Fixed in commit 221222aae8b44097eee4af9607457e6dde2e47fd

closed 2016-10-02 by tkeffer