weewx/weewx #54: schema extension does not work

matthewwall, 2015-07-24:
i tried to extend the wview schema as described in the customization guide:

```
import schemas.wview
schema_with_electricity = schemas.wview.schema + [('electricity', 'REAL')]
```

this fails with an exception:
```
Jul 18 17:07:28 burgos weewx[3921]:     ****  No module named wview
Jul 18 17:07:28 burgos weewx[3921]:     ****  Traceback (most recent call last):
Jul 18 17:07:28 burgos weewx[3921]:     ****    File "/opt/weewx-3.2.0/bin/weewx/engine.py", line 836, in main
Jul 18 17:07:28 burgos weewx[3921]:     ****      engine = EngineClass(config_dict)
Jul 18 17:07:28 burgos weewx[3921]:     ****    File "/opt/weewx-3.2.0/bin/weewx/engine.py", line 75, in __init__
Jul 18 17:07:28 burgos weewx[3921]:     ****      self.loadServices(config_dict)
Jul 18 17:07:28 burgos weewx[3921]:     ****    File "/opt/weewx-3.2.0/bin/weewx/engine.py", line 136, in loadServices
Jul 18 17:07:28 burgos weewx[3921]:     ****      self.service_obj.append(weeutil.weeutil._get_object(svc)(self, config_dict))
Jul 18 17:07:28 burgos weewx[3921]:     ****    File "/opt/weewx-3.2.0/bin/weewx/engine.py", line 504, in __init__
Jul 18 17:07:28 burgos weewx[3921]:     ****      self.setup_database(config_dict)
Jul 18 17:07:28 burgos weewx[3921]:     ****    File "/opt/weewx-3.2.0/bin/weewx/engine.py", line 602, in setup_database
Jul 18 17:07:28 burgos weewx[3921]:     ****      dbmanager = self.engine.db_binder.get_manager(self.data_binding, initialize=True)
Jul 18 17:07:28 burgos weewx[3921]:     ****    File "/opt/weewx-3.2.0/bin/weewx/manager.py", line 822, in get_manager
Jul 18 17:07:28 burgos weewx[3921]:     ****      default_binding_dict=defaults)
Jul 18 17:07:28 burgos weewx[3921]:     ****    File "/opt/weewx-3.2.0/bin/weewx/manager.py", line 952, in get_manager_dict_from_config
Jul 18 17:07:28 burgos weewx[3921]:     ****      manager_dict['schema'] = weeutil.weeutil._get_object(schema_name)
Jul 18 17:07:28 burgos weewx[3921]:     ****    File "/opt/weewx-3.2.0/bin/weeutil/weeutil.py", line 944, in _get_object
Jul 18 17:07:28 burgos weewx[3921]:     ****      mod = __import__(module)
Jul 18 17:07:28 burgos weewx[3921]:     ****    File "/opt/weewx-3.2.0/bin/user/schemas.py", line 2, in <module>
Jul 18 17:07:28 burgos weewx[3921]:     ****      import schemas.wview
Jul 18 17:07:28 burgos weewx[3921]:     ****  ImportError: No module named wview
Jul 18 17:07:28 burgos weewx[3921]:     ****  Exiting.
```
or this:

```
Jul 18 17:12:45 burgos weewx[3936]:     ****  name 'schemas' is not defined
Jul 18 17:12:45 burgos weewx[3936]:     ****  Traceback (most recent call last):
Jul 18 17:12:45 burgos weewx[3936]:     ****    File "/opt/weewx-3.2.0/bin/weewx/engine.py", line 836, in main
Jul 18 17:12:45 burgos weewx[3936]:     ****      engine = EngineClass(config_dict)
Jul 18 17:12:45 burgos weewx[3936]:     ****    File "/opt/weewx-3.2.0/bin/weewx/engine.py", line 75, in __init__
Jul 18 17:12:45 burgos weewx[3936]:     ****      self.loadServices(config_dict)
Jul 18 17:12:45 burgos weewx[3936]:     ****    File "/opt/weewx-3.2.0/bin/weewx/engine.py", line 136, in loadServices
Jul 18 17:12:45 burgos weewx[3936]:     ****      self.service_obj.append(weeutil.weeutil._get_object(svc)(self, config_dict))
Jul 18 17:12:45 burgos weewx[3936]:     ****    File "/opt/weewx-3.2.0/bin/weewx/engine.py", line 504, in __init__
Jul 18 17:12:45 burgos weewx[3936]:     ****      self.setup_database(config_dict)
Jul 18 17:12:45 burgos weewx[3936]:     ****    File "/opt/weewx-3.2.0/bin/weewx/engine.py", line 602, in setup_database
Jul 18 17:12:45 burgos weewx[3936]:     ****      dbmanager = self.engine.db_binder.get_manager(self.data_binding, initialize=True)
Jul 18 17:12:45 burgos weewx[3936]:     ****    File "/opt/weewx-3.2.0/bin/weewx/manager.py", line 822, in get_manager
Jul 18 17:12:45 burgos weewx[3936]:     ****      default_binding_dict=defaults)
Jul 18 17:12:45 burgos weewx[3936]:     ****    File "/opt/weewx-3.2.0/bin/weewx/manager.py", line 952, in get_manager_dict_from_config
Jul 18 17:12:45 burgos weewx[3936]:     ****      manager_dict['schema'] = weeutil.weeutil._get_object(schema_name)
Jul 18 17:12:45 burgos weewx[3936]:     ****    File "/opt/weewx-3.2.0/bin/weeutil/weeutil.py", line 944, in _get_object
Jul 18 17:12:45 burgos weewx[3936]:     ****      mod = __import__(module)
Jul 18 17:12:45 burgos weewx[3936]:     ****    File "/opt/weewx-3.2.0/bin/user/schemas.py", line 60, in <module>
Jul 18 17:12:45 burgos weewx[3936]:     ****      schema = schemas.wview.schema + [('lightning', 'REAL')]
Jul 18 17:12:45 burgos weewx[3936]:     ****  NameError: name 'schemas' is not defined
Jul 18 17:12:45 burgos weewx[3936]:     ****  Exiting.
```
my workaround is to copy the wview.py file to user/schema.py then modify it there.

tkeffer, 2015-07-24:
I just tried this and it worked fine for me.

matthewwall, 2015-07-25:
try this.  put this at the end of your bin/user/extensions.py:

```
import schemas.wview
schema = schemas.wview.schema + [('lightning',   'REAL'),
                                 ('maxSolarRad', 'REAL'),
                                 ('cloudbase',   'REAL'),
                                 ('windrun',     'REAL')]
```
then restart weewx.  i get this:
```
# /etc/init.d/weewx start
Traceback (most recent call last):
  File "/opt/weewx-3.2.0/bin/weewxd", line 12, in <module>
    import user.extensions       #@UnusedImport
  File "/opt/weewx-3.2.0/bin/user/extensions.py", line 20, in <module>
    import schemas.wview
ImportError: No module named wview
```

tkeffer, 2015-07-25:
Again, no problem. You didn't mention it, but I also tried it by modifying the `[[wx_binding]]` section to add the new schema to the binding:

```
    [[wx_binding]]
        ...
        schema = user.extensions.schema
```

matthewwall, 2015-07-25:
what could be causing this to fail in my installation?  my weewx.conf is located at /etc/weewx/weewx-3.2.0.conf.  my weewx is installed at /opt/weewx-3.2.0.  weewx.conf contains

WEEWX_ROOT = /opt/weewx-3.2.0/

is there something that munges the path that would affect the imports in bin/user/extensions.py?

tkeffer, 2015-07-25:
You could include

```python
import sys
print sys.path
```

at the top of engine.py and see where it's looking.

gjr80, 2015-07-25:
Pardon my interruption, I am not sure that it has any realtionship but this sounds somewhat similar to an issue we encountered when we revised Weewx-WD to operate with 3.0. We added a __main__ to one of our python modules and were doing some wee_database like functions. One or two functions ultimately called some manager.py methods which resulted in a weeutil method failing complaining that it could not find wview. It only ever affected a couple of users (I believe more than 2 people use Weewx-WD!), neither Greg or I could ever replicate the issue.

matthewwall, 2015-07-25:
i recall the weewx-wd issue.  at that time i could not replicate the behavior.  now i cannot get rid of it!

i put this in bin/user/extension.py:
```
import sys
print sys.path

import schemas.wview
schema = schemas.wview.schema + [('lightning',   'REAL'),
                                 ('maxSolarRad', 'REAL'),
                                 ('cloudbase',   'REAL'),
                                 ('windrun',     'REAL')]
```
and this is what i get when i run weewx:
```
cd /opt/weewx-3.2.0
./bin/weewxd /etc/weewx/weewx-3.2.0.conf
['/opt/weewx-3.2.0/bin', '/usr/lib/python2.6', '/usr/lib/python2.6/plat-linux2', '/usr/lib/python2.6/lib-tk', '/usr/lib/python2.6/lib-old', '/usr/lib/python2.6/lib-dynload', '/usr/local/lib/python2.6/dist-packages', '/usr/lib/python2.6/dist-packages', '/usr/lib/python2.6/dist-packages/PIL', '/usr/lib/pymodules/python2.6']
Traceback (most recent call last):
  File "./bin/weewxd", line 12, in <module>
    import user.extensions       #@UnusedImport
  File "/opt/weewx-3.2.0/bin/user/extensions.py", line 23, in <module>
    import schemas.wview
ImportError: No module named wview
```
schemas and wview.py are in fact there:
```
ls -la /opt/weewx-3.2.0/bin/schemas/
total 16
drwxr-xr-x  2 mwall mwall  448 Jul 18 17:11 .
drwxr-xr-x 10 mwall mwall 1400 Jul 18 16:31 ..
-rw-r--r--  1 mwall mwall  230 Jul 18 14:16 __init__.py
-rw-r--r--  1 mwall mwall  255 Jul 18 16:31 __init__.pyc
-rw-r--r--  1 mwall mwall 3218 Jul 18 14:16 wview.py
-rw-r--r--  1 mwall mwall 2593 Jul 18 16:31 wview.pyc
```

tkeffer, 2015-07-31:
Matthew, were you able to resolve this?

matthewwall, 2015-07-31:
no, it is still unresolved.

matthewwall, 2016-08-13:
tom,

this issue is still unresolved.  i have a workaround (see first posting), but i'd like to fix this.

what do you get for sys.path when you try to duplicate this configuration?

put this at the end of bin/user/extensions.py (after the locale.setlocale):

```
import sys
print sys.path

import schemas.wview
schema = schemas.wview.schema + [('lightning',   'REAL'),
                                 ('maxSolarRad', 'REAL'),
                                 ('cloudbase',   'REAL'),
                                 ('windrun',     'REAL')]
```

and put this in weewx.conf:

```
[DataBindings]
    [[wx_binding]]
        database = archive_sqlite
        table_name = archive
        manager = weewx.wxmanager.WXDaySummaryManager
        schema = user.extensions.schema
```

then run weewx:

`./bin/weewxd weewx.conf
`

what do you get?

tkeffer, 2016-08-29:
I get

```
./bin/weewxd weewx.conf
['/home/weewx/bin', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-x86_64-linux-gnu', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/home/tkeffer/.local/lib/python2.7/site-packages', '/usr/local/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages/PILcompat', '/usr/lib/python2.7/dist-packages/gtk-2.0', '/usr/lib/python2.7/dist-packages/ubuntu-sso-client']
REC:    2016-08-19 20:35:00 PDT (1471664100) altimeter: None, appTemp: 72.9587073731, barometer: 29.867, cloudbase: 5984.94696181, consBatteryVoltage: None, dateTime: 1471664100, dewpoint: 50.146233368, ET: 0.0, extraHumid1: None, extraHumid2: None, extraTemp1: 66.0, extraTemp2: None, extraTemp3: None, forecastRule: 45, heatindex: 73.4, highOutTemp: 73.9, highRadiation: 0.0, highUV: 0.0, humidex: 75.7685747615, inDewpoint: 53.4760934865, inHumidity: 42.0, inTemp: 78.5, interval: 5, leafTemp1: None, leafTemp2: None, leafWet1: None, leafWet2: None, lowOutTemp: 73.4, maxSolarRad: 0.0, outHumidity: 44.0, outTemp: 73.4, pressure: None, radiation: 0.0, rain: 0.0, rainRate: 0.0, rxCheckPercent: 99.9375, soilMoist1: None, soilMoist2: None, soilMoist3: None, soilMoist4: None, soilTemp1: None, soilTemp2: None, soilTemp3: None, soilTemp4: None, txBatteryStatus: None, usUnits: 1, UV: 0.0, windchill: 73.4, windDir: 202.5, windGust: 3.0, windGustDir: 202.5, windrun: 0.0, windSpeed: 1.0
REC:    2016-08-19 20:40:00 PDT (1471664400) altimeter: None, appTemp: 72.8750072275, barometer: 29.869, cloudbase: 5842.98230485, consBatteryVoltage: None, dateTime: 1471664400, dewpoint: 50.5708778587, ET: 0.0, extraHumid1: None, extraHumid2: None, extraTemp1: 66.0, extraTemp2: None, extraTemp3: None, forecastRule: 45, heatindex: 73.2, highOutTemp: 73.4, highRadiation: 0.0, highUV: 0.0, humidex: 75.767035356, inDewpoint: 53.296634114, inHumidity: 42.0, inTemp: 78.3, interval: 5, leafTemp1: None, leafTemp2: None, leafWet1: None, leafWet2: None, lowOutTemp: 73.1, maxSolarRad: 0.0, outHumidity: 45.0, outTemp: 73.2, pressure: None, radiation: 0.0, rain: 0.0, rainRate: 0.0, rxCheckPercent: 99.9375, soilMoist1: None, soilMoist2: None, soilMoist3: None, soilMoist4: None, soilTemp1: None, soilTemp2: None, soilTemp3: None, soilTemp4: None, txBatteryStatus: None, usUnits: 1, UV: 0.0, windchill: 73.2, windDir: 202.5, windGust: 3.0, windGustDir: 202.5, windrun: 0.0833333333333, windSpeed: 1.0
REC:    2016-08-19 20:45:00 PDT (1471664700) altimeter: None, appTemp: 71.8024464756, barometer: 29.871, cloudbase: 6107.39636169, consBatteryVoltage: None, dateTime: 1471664700, dewpoint: 48.8074560086, ET: 0.0, extraHumid1: None, extraHumid2: None, extraTemp1: 66.0, extraTemp2: None, extraTemp3: None, forecastRule: 45, heatindex: 72.6, highOutTemp: 73.1, highRadiation: 0.0, highUV: 0.0, humidex: 74.3611270777, inDewpoint: 53.1171667581, inHumidity: 42.0, inTemp: 78.1, interval: 5, leafTemp1: None, leafTemp2: None, leafWet1: None, leafWet2: None, lowOutTemp: 72.6, maxSolarRad: 0.0, outHumidity: 43.0, outTemp: 72.6, pressure: None, radiation: 0.0, rain: 0.0, rainRate: 0.0, rxCheckPercent: 98.2291666667, soilMoist1: None, soilMoist2: None, soilMoist3: None, soilMoist4: None, soilTemp1: None, soilTemp2: None, soilTemp3: None, soilTemp4: None, txBatteryStatus: None, usUnits: 1, UV: 0.0, windchill: 72.6, windDir: 202.5, windGust: 3.0, windGustDir: 202.5, windrun: 0.166666666667, windSpeed: 1.0
...
```

No errors.

tkeffer, 2016-10-11:
Anything new on this one?

gjr80, 2016-10-11:
I have been trying to reproduce this on my RPis and any VMs I create and I have been unable to do so. I have only ever experienced it once and that was at least a year ago. I have not run in to the issue since. I think the code is fine and there is some other weird issue that occasionally rears it head.

tkeffer, 2016-11-26:
Question: when you have this problem, do you by any chance have a file named `weewx.py` in the `schemas` subdirectory? I am finding that if I do, I get this error. 

For reasons I don't quite understand, the presence of a module `schemas.weewx`, *even if you don't import it*, screws up everything else in `schemas` subdirectory. I suppose because it is confusing it with the *package* weewx, or .... mumble... something.

Unless I have a gross misunderstanding of Python, this is a bug in Python --- the first I've ever found in 10+ years of usage.

matthewwall, 2016-11-26:
no weewx.py in the schemas directory.  i have two systems on which i have extended the schema using this technique.  on the arm system, i continue to have the problem described above.  on the x64 system, everything works properly.  unfortunately i do not have access to the arm system again until february or march (maybe january if we get a sunny day with no sea ice when i can kayak out to the island :)

tkeffer, 2018-09-28:
Still having this problem? I just tried adding the code you suggested to the bottom of `bin/user/extensions.py` on both an Intel machine and on an RPi. Both worked fine.

-tk

futzle, 2019-02-12:
I got the same error as @matthewwall yesterday while trying to extend the schema by adding to bin/user/extensions.py.  I'm not familiar enough with Python's import model to debug it myself.

futzle, 2019-02-12:
Update: I worked out what was causing the error in my installation.  It was a file left over from weewx 2.6: bin/user/schemas.py.  After I deleted this file the error went away.

zepingouin, 2019-05-09:
Confirmed: After an upgrade from 2.7.0 to 3.9.1, there were /usr/share/weewx/user/schemas.py left and removing this file solved the problem.

tkeffer, 2019-05-17:
OK, now this is all making sense. Take Matthew's original case:

At the end of `bin/user/extensions.py`:

```python
import schemas.wview
schema = schemas.wview.schema + [('lightning',   'REAL'),
                                 ('maxSolarRad', 'REAL'),
                                 ('cloudbase',   'REAL'),
                                 ('windrun',     'REAL')]
```

The import looks like an absolute import, but under Python 2, it is treated as a *relative* import. So, Python imports the module `user.schema`, then tries to find something inside it with the name `wview`, which, of course, does not exist. By getting rid of the file `user/schema.py`, you force Python to go look elsewhere for `schema`. It finds the *package* schema, which does contain a `wview`.

The solution is, as @futzle discovered, delete `bin/user/schema.py`.

Glitches like this is precisely the reason why Python 3 no longer allows implicit relative imports.

I'll work on a fix

tkeffer, 2019-05-17:
After near four years as an outstanding bug, I think this has been fixed in commit https://github.com/weewx/weewx/commit/ad0e4976660052e810d2ea27687c9d892273747f. Thanks, Deborah (@futzle)!

closed 2019-05-17 by tkeffer