weewx/weewx #55: rsync process lingers after weewx says it has shut down

matthewwall, 2015-07-25:
if an rsync is in progress when weewx shuts down, the weewx engine thread does not wait for the rsync process to complete.  the result is misleading log messages and a dangling weewx process.  if a regular (non-rsync) report is still in progress when the engine shuts down, the engine logs something to the effect of "unable to terminate std report thread".  

this is what happens when weewx terminates while an rsync process is running (there are two rsyncs in this example, the first succeeded in 3.34 seconds, the second is retrying due to network issues):
```
Jul 24 19:56:12 burgos weewx[16088]: cheetahgenerator: Generated 8 files for report Exfoliation in 33.72 seconds
Jul 24 19:56:20 burgos weewx[16088]: genimages: Generated 24 images for Exfoliation in 7.49 seconds
Jul 24 19:56:23 burgos weewx[16088]: rsyncupload: rsync'd 32 files (303553 bytes) in 3.34 seconds
Jul 24 19:57:34 burgos weewx[16088]: engine: Shutting down StdReport thread
Jul 24 19:57:34 burgos weewx[16088]: engine: Terminating weewx version 3.2.0
```
but this is what is still running:
```
16088 ?        Sl   265:52 /usr/bin/python /opt/weewx-3.2.0/bin/weewxd --daemon --pidfile=/var/run/weewx.pid /etc/weewx/weewx-3.2.0.conf
18207 ?        S      0:00 rsync --archive --stats -e ssh /var/weewx/www/ weewx@www.bartschi.org:/var/www/weather/wls-ts
18208 ?        S      0:00 ssh -l weewx www.bartschi.org rsync --server -logDtpre.iLsf . /var/www/weather/wls-ts
```
and this is what shows up in the log after killing the ssh process (18208):
```
Jul 24 19:56:12 burgos weewx[16088]: cheetahgenerator: Generated 8 files for report Exfoliation in 33.72 seconds
Jul 24 19:56:20 burgos weewx[16088]: genimages: Generated 24 images for Exfoliation in 7.49 seconds
Jul 24 19:56:23 burgos weewx[16088]: rsyncupload: rsync'd 32 files (303553 bytes) in 3.34 seconds
Jul 24 19:57:34 burgos weewx[16088]: engine: Shutting down StdReport thread
Jul 24 19:57:34 burgos weewx[16088]: engine: Terminating weewx version 3.2.0
Jul 24 20:00:07 burgos weewx[16088]: rsyncupload: rsync reported errors: rsync: connection unexpectedly closed (0 bytes received so far) [sender]. rsync error: error in rsync protocol data stream (code 12) at io.c(601) [sender=3.0.7]
Jul 24 20:00:07 burgos weewx[16088]: rsyncupload: rsync executed in 2600.71 seconds
```

tkeffer, 2016-11-30:
Is there a flag for spawning a process that could change this behavior?

matthewwall, 2021-04-04:
this also happens with `rtl_433`.  do a clean shutdown of weewx, but the `rtl_433` process continues to run.  its a problem, because the next time you try to start weewx, weewx fails because `rtl_433` is still running and has a connection to the usb port.  it seems to be related to `rtl_433` state - it happens only in cases when weewx is not getting data from `rtl_433`.  so i suspect that `rtl_433` might be stuck on a socket and thus ignores any SIGTERM.

prolly need to make weewx understand whether or not its children are zombies, and if they are do a KILL on them, not just a TERM.

this also means we need to ensure that the rc script or systemd script does the same thing - wait awhile after sending weewx a TERM, then after some time (maybe 2 archive intervals?) send it a KILL.  definitely emit something to log about whether the shutdown was due to a KILL or TERM.

bellrichm, 2022-04-14:
As I was trying to figure out why I was seeing two `SIGKILL` from `systemctl stop`, I stumbled across this issue. Here are some ideas on possible 'fixes'. None of these are perfect, but maybe it will trigger some other ideas.

1. Instead of blocking on the process running, use poll in a loop to check the status. 
   This would require the service/driver/report generator to be multi-threaded.
   This has the advantage of being self contained, but adds complexity to each implementor. 
   I also saw some concerns with over running the STDOUT and STDERR buffers. 
   There also might be some timing issues.
  
3. In `weewxd`, send the `SIGTERM` from the signal handler.
   Would require some state so that only sent once
   In theory this would start shutting down the processes before the threads; allowing the threads to react to process shutting down.
   Does not solve the hung process that requires a `SIGKILL`.
   Seems like a solution for a non-existent problem. 
   If the service/driver/report generator 'is that concerned', it can implement the first bullet.
   
4. If there are any processes still running do not exit and have the controlling process (ie systemd) send the `SIGKILL`.
   Would have to determine if any processes are still running.
   Confusion for non linux people on why `kill PID` does not work and `kill -9 PID` is needed.

6. In `weewxd`, after sending the `SIGTERM`, wait/sleep for some period of time and then send a `SIGKILL`.
   Probably the only way to ensure that WeeWX does not leave any zombie processes.
   Simple (other than how long to wait), but I'm not a fan of blindly sending the `SIGKILL`.

If anything here, or a derative, sounds interesting; I'd be happy to do some prototyping to flesh out additional ideas/details.

tkeffer, 2022-04-15:
Good suggestions. 

Another approach would be similar to what `StdReport` does: save the last instance that was launched, then don't launch another until it has terminated. That way, you're guaranteed to have only one instance of `rsync` running. On shut down, if it's still running, then send a `SIGTERM`, followed by `SIGKILL`.

Rich, I would definitely appreciate it if you researched some solutions. It's not a big problem, but it's big enough that it's worth doing.

bellrichm, 2022-04-16:
> Another approach would be similar to what `StdReport` does: save the last instance that was launched, then don't launch another until it has terminated. That way, you're guaranteed to have only one [...]

Since `StdReport` launches a new thread for each reporting cycle, each `rsync` process is in a separate thread. So unless we bubble some information up to the parent thread (which I don't think we want to do), I don't think we can stop the possibility of multiple `rsync` processes this way.
I think we need to be careful on what we mean by 'on shut down'. If you mean 'shut down' of the thread, unless the invocation of the process is nonblocking, the thread's shutDown method is not called. If you mean 'shut down' of the engine, this would require some type of registration of running processes. also, doesn't sound like something we want to do.

bellrichm, 2022-04-16:
> Rich, I would definitely appreciate it if you researched some solutions. It's not a big problem, but it's big enough that it's worth doing.

I'm happy to some research and experiments. It's a great chance for me to learn. I think that conversing about them in this issue will be unwieldy. Unless you have a better idea, I plan to start a development group thread for each experiment.

bellrichm, 2022-04-21:
tl;dr - `systemd` seems to handle sub processes that do not terminate on a SIGTERM.

I have a simple WeeWX service, `testsignal`, that invokes a python program, `sleepy.py`, that runs an infinite loop that consists of a `sleep`. It can be configured to 'ignore' or 'handle' signals. The code can be found in the testsignal directory of https://github.com/bellrichm/experiments
Below is the output of a `start` and `stop`. From the log, at 11:29:55, both `weewx` and `sleepy` receive the SIGTERM. At 11:30:15 `sleepy` receives SIGTERM that originates from `weewx` `os.kill(0, signal.SIGTERM)`. Not sure why there is 2, but if I comment out the code that sends the signal I see none being 'ignored' by `sleepy`.

 ```
pi@raspberrypi:/home/weewx $ sudo systemctl start weewx

pi@raspberrypi:/home/weewx $ ps -ef | grep weewx
pi       23029 22795  0 10:43 pts/2    00:00:00 tail -f /var/log/weewx.log
pi       24218   915  0 11:03 ?        00:00:10 gedit file:///home/weewx/weewx.conf
pi       24554     1 11 11:29 ?        00:00:04 python /home/weewx/bin/weewxd /home/weewx/weewx.conf
pi       24561 24554  2 11:29 ?        00:00:00 /usr/bin/python3 /home/pi/weewx_code/fork.weewx/bin/user/sleepy.py --verbosity=0 --seconds=5 --sigterm=ignore --sigint=ignore
pi       24564 20883  0 11:29 pts/0    00:00:00 grep --color=auto weewx

pi@raspberrypi:/home/weewx $ sudo systemctl stop weewx

pi@raspberrypi:/home/weewx $ ps -ef | grep weewx
pi       23029 22795  0 10:43 pts/2    00:00:00 tail -f /var/log/weewx.log
pi       24218   915  0 11:03 ?        00:00:10 gedit file:///home/weewx/weewx.conf
pi       24598 20883  0 11:32 pts/0    00:00:00 grep --color=auto weewx

pi@raspberrypi:/home/weewx $ sudo systemctl status weewx
* weewx.service - weewx weather system
   Loaded: loaded (/etc/systemd/system/weewx.service; disabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Thu 2022-04-21 11:31:45 EDT; 1min 1s ago
  Process: 24554 ExecStart=/home/weewx/bin/weewxd /home/weewx/weewx.conf (code=killed, signal=TERM)
 Main PID: 24554 (code=killed, signal=TERM)

Apr 21 11:29:55 raspberrypi systemd[1]: Stopping weewx weather system...
Apr 21 11:30:15 raspberrypi python[24554]: weewx[24554] ERROR user.testsignal: Unable to shut down Thread-1 thread
Apr 21 11:30:15 raspberrypi python[24554]: weewx[24554] INFO __main__: Terminating weewx version 4.7.0
Apr 21 11:30:15 raspberrypi wee_sleepy[24561]: Ignoring signal SIGTERM (15).
Apr 21 11:30:15 raspberrypi systemd[1]: weewx.service: Main process exited, code=killed, status=15/TERM
Apr 21 11:30:15 raspberrypi wee_sleepy[24561]: Ignoring signal SIGTERM (15).
Apr 21 11:31:45 raspberrypi systemd[1]: weewx.service: State 'stop-final-sigterm' timed out. Killing.
Apr 21 11:31:45 raspberrypi systemd[1]: weewx.service: Killing process 24561 (python3) with signal SIGKILL.
Apr 21 11:31:45 raspberrypi systemd[1]: weewx.service: Failed with result 'timeout'.
Apr 21 11:31:45 raspberrypi systemd[1]: Stopped weewx weather system.

pi@raspberrypi:/home/weewx $ tail -n 13 /var/log/weewx.log
Apr 21 11:29:07 raspberrypi weewx[24554] INFO weewx.engine: Starting main packet loop.
Apr 21 11:29:36 raspberrypi weewx[24554] INFO user.testsignal: sleeping blocking
Apr 21 11:29:37 raspberrypi wee_sleepy[24561]: starting
Apr 21 11:29:55 raspberrypi weewx[24554] INFO __main__: Received signal TERM (15).
Apr 21 11:29:55 raspberrypi weewx[24554] INFO weewx.engine: Main loop exiting. Shutting engine down.
Apr 21 11:29:55 raspberrypi weewx[24554] INFO weewx.engine: Shutting down StdReport thread
Apr 21 11:29:55 raspberrypi weewx[24554] INFO user.testsignal: SHUTDOWN - initiated
Apr 21 11:29:55 raspberrypi weewx[24554] INFO user.testsignal: SHUTDOWN - thread initiated
Apr 21 11:29:55 raspberrypi wee_sleepy[24561]: Ignoring signal SIGTERM (15).
Apr 21 11:30:15 raspberrypi weewx[24554] ERROR user.testsignal: Unable to shut down Thread-1 thread
Apr 21 11:30:15 raspberrypi weewx[24554] INFO __main__: Terminating weewx version 4.7.0
Apr 21 11:30:15 raspberrypi wee_sleepy[24561]: Ignoring signal SIGTERM (15).
Apr 21 11:30:15 raspberrypi wee_sleepy[24561]: Ignoring signal SIGTERM (15).
```

closed 2024-07-04 by github-actions[bot]