weewx/weewx #63: Lacrosse WS1640 full support

superdevil669, 2015-08-17:
Hello, is it possible to add full support for the Lacrosse WS1640 with driver and all it takes?
Thank you very much

matthewwall, 2015-08-18:
what is the vendor id and product id for the ws1640?  please post the output from 'lsusb -v' while the weather station is plugged in.

superdevil669, 2015-08-19:
Bus 001 Device 004: ID 1130:6801 Tenx Technology, Inc.
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x1130 Tenx Technology, Inc.
  idProduct          0x6801
  bcdDevice            3.02
  iManufacturer           0
  iProduct                1
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10

matthewwall, 2015-08-19:
that looks like a hideki, probably the te838 model.
please try using the te923 driver when you install weewx, then let us know how that goes.

superdevil669, 2015-08-19:
Sure I get good settings thank you
Could you give me a link with the list of all tags that can be displayed like "$ current.outTemp" so I try to create the skin templats and myself?

tkeffer, 2015-08-19:
Please post support questions to weewx-user. And, see your documentation.

matthewwall, 2015-08-21:
the lacross ws2640 is a hideki te838.  apparently it works with the te923 driver.

closed 2015-08-21 by matthewwall

superdevil669, 2015-08-21:
thank you it actually works pretty with te923 drivers but I have no access to the archives it is possible to update the drivers because I saw that the version 0.15 with access to the archive of the station!