weewx/weewx #77: ImportError: No module named weewx

DaveJ61, 2015-10-26:
I have successfully installed weewx and due to me having a 1-wire station and I'm running owfs, I had to install the owfs driver. However, I had to force the dependencies to be installed. I hate doing that and to be honest it should not really be done as I have had major problems in the past because it broke my system.
Part of the installation requires to check what sensors are attached but I had to change the location for owfs.py as on my system it was installed in /usr/share/weewx/user/.

`sudo PYTHONPATH=/usr/bin/python python /usr/share/weewx/user/owfs.py --sensors`

When I tried to run this, I got the following error:-

    Traceback (most recent call last):
      File "/usr/share/weewx/user/owfs.py", line 168, in <module>
        import weewx
    ImportError: No module named weewx

Is there another python interpreter that weewx is using? I searched everywhere and cannot find out why this is happening.

matthewwall, 2015-10-26:
you are using the wrong PYTHONPATH.  assuming an RPM or DEB installation, you should do this:

sudo PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/owfs.py --sensors

as for list-drivers behavior, i see owfs listed when i try list-drivers on a debian8 system with owfs.py in /usr/share/weewx/user and weewx 3.2.1.  perhaps owfs.py is not visible or is not where you think it is?

m

DaveJ61, 2015-10-27:
Ok, but I didn't know what PYTHONPATH to use as there was nothing I could find in the documentation to tell me for a deb package installation.
As far as the location of owfs.py, As per my post, I know the location is in /usr/share/weewx/user/ and it is visible?
When I now check for sensors with the new PYTHONPATH I get the following:-

    sudo PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/owfs.py --sensors
    [sudo] password for david:
    Traceback (most recent call last):
      File "/usr/share/weewx/user/owfs.py", line 629, in <module>
        main()
      File "/usr/share/weewx/user/owfs.py", line 594, in main
        ow.init(iface)
      File "/usr/lib/python2.7/dist-packages/ow/__init__.py", line 224, in init
        raise exNoController
    ow.exNoController

If I reconfigure I still do not see the owfs driver.... it is not listed... Surely it is not that difficult to install this?

matthewwall, 2015-10-27:
ow.exNoController indicates that there is no one-wire controller.

how is your one-wire bus attached the the computer?

exactly what do you mean when you say you had to 'force the dependencies to be installed'?  exactly what did you do?

DaveJ61, 2015-10-27:
There is a controller attached to a serial port and I can use this with owserver / owhttpd to see my 1wire devices.

As far as forcing the installation of some packages, I saw instructions on the weewx website to use apt-get install -f when installing weewx as there were dependencies that were required for weewx such as python-cheetah and a couple of others I can't remember.
The link for this is http://www.weewx.com/docs/debian.htm

matthewwall, 2015-10-27:
you can configure weewx to use the physical interface (e.g., interface=u for usb, interface=/dev/tty/SNX0 for serial) or a one-wire server (e.g., interface=hostname:3003).  use the --iface option to test this when invoking owfs.py directly.

if you had to use -f to install weewx, then something is wrong with your system.

driver=user.owfs is correct.

closed 2015-10-27 by matthewwall

DaveJ61, 2015-10-27:
Sorry, but I did realise how to configure for using physical interfaces. It was the owserver part of it that I was unsure of whether could be done or not.

> if you had to use -f to install weewx, then something is wrong with your system.

My system is not broken and there is nothing wrong with it - I tried this on fresh installation and had the same result. Using the apt-get install -f can indeed break your system if not careful. I have known others in the past it has happened to, from forums and also friends and it has even broken my system before now, hence why I am being very cautious. There is a very good reason why apt-get refuses to install certain packages. It normally resolves dependencies automatically anyway, so if it won't install them, then there is an issue.

I tried invoking owfs.py directly but that didn't work either. It said there is no --iface option.

Also wee_config --list-drivers does not list the owfs driver, so I take that it only lists the drivers that come with the weewx and not extensions.

I am still getting the ow.exNoController exception when trying to list the devices, even after trying both configurations of interface=localhost:4304 and /dev/ttySNX0.

As far as I can see, everything is configured correctly but it still refuses to work.

matthewwall, 2015-10-27:
did you try this:

```PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/owfs.py --iface localhost:4304 --sensors```

the --iface option works for me, and i am unable to duplicate the behavior of the owfs driver not showing up.  i've been testing on ubuntu 12.04, debian 7, and debian 8.  perhaps there is something different about ubuntu 14.04?  or something still different about your configuration?

in the case of weewx, the -f is only needed to get apt-get/dpkg to sort out dependencies.  dpkg knows about dependencies but cannot do anything about them.  apt-get can.  the -f option is a standard approach to dealing with this.  it is not necessary if you install only with apt-get, but you cannot specify a .deb file directly with apt-get, thus the need for both dpkg and apt-get.

matthewwall, 2015-10-27:
the reason owfs was not showing up in your driver listing was due to a missing DRIVER_NAME hint in owfs.py.  owfs 0.16 fixes this.

the --iface option has been in there since long before 0.11 was released.

DaveJ61, 2015-10-28:
Aha.... I didn't realise this is where you meant to the --iface.

    PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/owfs.py --iface localhost:4304 --sensors

Using this, I now get:

    142D0F050000: /09.142D0F050000 DS2502
    09E00F000000: /1F.056505000000/main/1D.09E00F000000 DS2423
    BA8400000000: /1F.CB6C05000000/main/20.BA8400000000 DS2450
    9E5A1B000800: /1F.CB6C05000000/main/10.9E5A1B000800 DS18S20
    604A01000000: /1F.CB6C05000000/main/1D.604A01000000 DS2423
    C09F4A050000: /1F.1B5D05000000/main/28.C09F4A050000 DS18B20
    DA158F010000: /1F.1B5D05000000/main/26.DA158F010000 DS2438

...which is what I expect to get and just when I thought everything was going well:-

    PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/owfs.py --iface localhost:4304 --readings
    142D0F050000
    _usePath: /09.142D0F050000
    _type: DS2502
    _path: /09.142D0F050000
    _useCache: True
      /09.142D0F050000/pages/: page.ALL,page.0,page.1,page.2,page.3
      /09.142D0F050000/family: 09
      /09.142D0F050000/locator: FFFFFFFFFFFFFFFF
      /09.142D0F050000/type: DS2502
      /09.142D0F050000/crc8: F0
      /09.142D0F050000/memory: ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒
      /09.142D0F050000/r_id: 0000050F2D14
      /09.142D0F050000/address: 09142D0F050000F0
      /09.142D0F050000/r_address: F00000050F2D1409
      /09.142D0F050000/r_locator: FFFFFFFFFFFFFFFF
      /09.142D0F050000/id: 142D0F050000
    09E00F000000
    _usePath: /1F.056505000000/main/1D.09E00F000000
    _type: DS2423
    _path: /1F.056505000000/main/1D.09E00F000000
    _useCache: True
      /1F.056505000000/main/1D.09E00F000000/memory: *▒>y{▒8 |▒▒d▒
      /1F.056505000000/main/1D.09E00F000000/type: DS2423
      /1F.056505000000/main/1D.09E00F000000/family: 1D
      /1F.056505000000/main/1D.09E00F000000/locator: FFD2FFFFFFFFFFFF
      /1F.056505000000/main/1D.09E00F000000/pages/:     count.ALL,count.0,count.1,count.2,count.3,count.4,count.5,count.6,count.7,count.8,count.9,count.10,count.11,count.12,count.13,count.14,count.15,page.ALL,page.0,page.1,page.2,page.3,page.4,page.5,page.6,page.7,page.8,page.9,page.10,page.11,page.12,page.13,page.14,page.15
      /1F.056505000000/main/1D.09E00F000000/r_locator: FFFFFFFFFFFFD2FF
      /1F.056505000000/main/1D.09E00F000000/crc8: 15
    Traceback (most recent call last):
      File "/usr/share/weewx/user/owfs.py", line 629, in <module>
        main()
      File "/usr/share/weewx/user/owfs.py", line 598, in main
        traverse(ow.Sensor('/'), display_sensor_info)
      File "/usr/share/weewx/user/owfs.py", line 624, in traverse
        traverse(s, func)
      File "/usr/share/weewx/user/owfs.py", line 626, in traverse
        func(s)
      File "/usr/share/weewx/user/owfs.py", line 608, in display_sensor_info
        display_dict(s.__dict__)
      File "/usr/share/weewx/user/owfs.py", line 613, in display_dict
        display_dict(d[k], level=level+1)
      File "/usr/share/weewx/user/owfs.py", line 619, in display_dict
        print '%s%s: %s' % ('  '*level, d[k], ow.owfs_get(d[k]))
      File "/usr/lib/python2.7/dist-packages/ow/__init__.py", line 159, in _get
        raise exUnknownSensor(path)
    ow.exUnknownSensor: '/1F.056505000000/main/1D.09E00F000000/counter.ALL/'
 
I do have a Hobbyboards 6-port switch on the bus as well, which looks like this is supported by the way it is traversing it but then bang.... the 1D is a counter, which I know is supported but an exception has been thrown.
Incidentally, I can't figure out what the first column in the output is from the sensors scan. An address? 

This is the owfs section from my weewx.conf file (I haven't filled any sensors in yet):-

    # Options for extension 'owfs'
    [OWFS]
        interface = localhost:4304
        driver = user.owfs
        [[sensor_type]]
        [[sensor_map]]

matthewwall, 2015-10-28:
please try weewx-owfs-0.17.  this will handle exceptions and should complete the enumeration while traversing the sensor hierarchy, even if the sensor type is not yet known by weewx or owfs.

matthewwall, 2015-10-28:
with either 0.16 or 0.17 you should be able to get data from the sensors - just specify the sensor paths in the sensor_map.  0.17 simply makes the --readings option more robust.

DaveJ61, 2015-10-28:
Ok, thanks Matthew, I'll give it a whirl.

Another slight issue is:-
    weewx[9993]: engine: 2015-10-28 06:09:05 GMT (1446012545) LOOP value 'outHumidity' 100.979 outside limits (0.0, 100.0)

Is there any way to stop this behaviour of continuously reporting this error every 10 or 15 seconds. I know it is over reading and I have had this problem with various humidity sensors where they would work great for quite a few months and then they would over-read.

matthewwall, 2015-10-28:
that message appears (by design) if an observation is out of bounds.  there is no way to selectively disable it, but if it is really causing you grief you could simply comment it out.  see line 431 in bin/weewx/engine.py

DaveJ61, 2015-10-28:
Ok Matthew, great. It's not really causing grief but I can see that my log is going to plastered with those errors.
Anyway, your weewx-owfs 0.17 works fine as far as I can see.
I'm going to run with this for now and do some more reading to try and suss everything out and fix the humidity sensor.
In the end I think we have resolved a few issues. Thank you very much for your time and patience.

tkeffer, 2015-10-28:
David, if you run into trouble, please continue the discussion in the
weewx-user group, not the issues database. The latter is not really set up
for support type questions.

-tk