weewx/weewx #78: Issue with syslog label

CondormanFr, 2015-11-13:
It looks that this commit : https://github.com/weewx/weewx/commit/84e3d1c824f634d07142d588cc3f79726c75b6d9 *"added option to specify syslog label"* make an issue.

This lines https://github.com/weewx/weewx/blob/master/bin/weewxd.py#L40
has been added to the file https://github.com/weewx/weewx/blob/master/bin/weewxd.py but not https://github.com/weewx/weewx/blob/master/bin/weewxd

And during the install process, a symbolic link is done here : https://github.com/weewx/weewx/blob/master/setup.py#L254 

but the link is done on the weewxd file and this one has not the option *--log-label*.
Then it failed when running weeWX  : 
```
Traceback (most recent call last):
  File "./weewxd", line 61, in <module>
    weewx.engine.main(options, args)
  File "/weewx/bin/weewx/engine.py", line 806, in main
    syslog.openlog(options.log_label, syslog.LOG_PID | syslog.LOG_CONS)
AttributeError: Values instance has no attribute 'log_label'
```

Perhaps, you just have to add https://github.com/weewx/weewx/blob/master/bin/weewxd.py#L40 to the https://github.com/weewx/weewx/blob/master/bin/weewxd file.

matthewwall, 2015-11-15:
tom,
is there a reason to have both weewxd and weewxd.py?  why not simply weewxd as the entry point, just as we do for wee_config, wee_database, etc?
m

tkeffer, 2015-11-15:
If there is, it's been lost to antiquity. I say get rid of weewxd.py.

matthewwall, 2015-11-15:
weewxd.py eliminated as of c28451a

closed 2015-11-15 by matthewwall