weewx/weewx #82: WOW: Unexpected exception of type <type 'exceptions.TypeError'>

toxic17, 2015-12-06:
After upgrading WeeWX to 3.30 from 3.2.1 i am not getting WOW errors.:

Dec  6 20:20:12 conligwx weewx[10939]: restx: WOW: Unexpected exception of type <type 'exceptions.TypeError'>
Dec  6 20:20:12 conligwx weewx[10939]: vantage: DMPAFT complete: page timestamp 2015-11-27 23:05:00 GMT (1448665500) less than final timestamp 2015-12-06 20:20:00 GMT (1449433200)
Dec  6 20:20:12 conligwx weewx[10939]: vantage: Catch up complete.
Dec  6 20:20:12 conligwx weewx[10939]: vantage: Requesting 200 LOOP packets.
Dec  6 20:20:12 conligwx weewx[10939]: reportengine: Running reports for latest time in the database.
Dec  6 20:20:12 conligwx weewx[10939]: reportengine: Running report HTMLPages
Dec  6 20:20:12 conligwx weewx[10939]: *** Traceback (most recent call last):
Dec  6 20:20:12 conligwx weewx[10939]: ***   File "/home/weewx/bin/weewx/restx.py", line 308, in run_loop
Dec  6 20:20:12 conligwx weewx[10939]: ***     self.process_record(_record, dbmanager)
Dec  6 20:20:12 conligwx weewx[10939]: ***   File "/home/weewx/bin/weewx/restx.py", line 349, in process_record
Dec  6 20:20:12 conligwx weewx[10939]: ***     self.post_with_retries(_request)
Dec  6 20:20:12 conligwx weewx[10939]: ***   File "/home/weewx/bin/weewx/restx.py", line 369, in post_with_retries
Dec  6 20:20:12 conligwx weewx[10939]: ***     _response = self.post_request(request, payload)
Dec  6 20:20:12 conligwx weewx[10939]: *** TypeError: post_request() takes exactly 2 arguments (3 given)
Dec  6 20:20:12 conligwx weewx[10939]: restx: WOW: Thread exiting. Reason: post_request() takes exactly 2 arguments (3 given)

tkeffer, 2015-12-06:
Fixed in 3e00a1c66e9e792c1ebd61a1042a0140df4d02b8

closed 2015-12-06 by tkeffer

toxic17, 2015-12-06:
Thanks for the quick fix.  Applied and I can confirm it is now working.

Thanks again!

Regards Simon.
http://www.linksysinfo.org