weewx/weewx #83: Backfill should be broken up into smaller transactions

tkeffer, 2015-12-09:
Right now, a backfill is done as a single transaction. This can require a lot of memory on a MySQL server if the the number of archive records is large.

Instead, a commit should be done every couple thousand records or so.

gjr80, 2015-12-10:
Tom, I toyed with this a while ago when I ran into similar problems building the daily summaries (weewx-wd or weewx, not sure) but I got too wrapped up in transactions, genbatchrecords etc and how it all worked and I let it go (resorted to offsite build in a VM). I am happy to have another go. Any pointers? I just spent half an hour reading the code again, was wondering would building x days as a transaction be simpler than, say, doing x'000 records per transaction? Just thinking of simplfying handling accumulators between transactions.

Gary

tkeffer, 2015-12-10:
That would be a sensible way of doing it. Conceptually it's also appealing: a transaction controls the single write needed to store a day summary.

Basically, you'd need an outer loop that processes a single day's worth of data at a time.

-tk

gjr80, 2015-12-10:
Yep, figured it was just an outer loop with some rejigging of whats already there. Will see how we go.

gjr80, 2015-12-16:
Time for a little more guidance...
Have a reworked backfill_day_summary() that processes records in transcations of x days (max) at a time. Have a reworked wee_database utility that uses a --trans-days parameter to set the max number of days in a transaction (uses a default of 5 atm). Seems to work well and handles the corner cases correctly, just in the final throes of checking it over and will then offer up a pull request.

I have noticed what I think is a logging inconsistency though; on startup with an empty database (or one with no daily summaries) weewx goes through the backfill and syslogs as follows:

Dec 12 10:25:24 debian weewx[3466]: engine: Archive will use data binding wx_binding
Dec 12 10:25:24 debian weewx[3466]: engine: Record generation will be attempted in 'hardware'
Dec 12 10:25:24 debian weewx[3466]: engine: Using archive interval of 300 seconds
Dec 12 10:25:24 debian weewx[3466]: manager: Created and initialized table 'archive' in database 'weewx.sdb'
Dec 12 10:25:24 debian weewx[3466]: manager: Created daily summary tables
Dec 12 10:25:24 debian weewx[3466]: engine: Using binding 'wx_binding' to database 'weewx.sdb'
Dec 12 10:25:24 debian weewx[3466]: engine: Starting backfill of daily summaries
Dec 12 10:25:24 debian weewx[3466]: engine: Daily summaries up to date.

When I do a backfill using wee_database under the same circumstances the following is logged:

Dec 12 10:35:35 debian wee_database[3537]: manager: Created daily summary tables

Of course the user is informed on screen by wee_database what is going on. I see why what happens happens, but I would have thought that if wee_database is going to log the creation of the daily summary tables (albeit indirectly) should it not also log that we backfilled them? Assuming this is a 'yes', at the moment wee_database does not do any explicit syslogging, though it does setup syslog for logging. Are you happy to include some syslogging in wee_database backfillDaily() similar to that in DaySummaryManager __init__ ? If so should wee_database also syslog that it dropped the daily summaries when dropDaily() is called. Know your views on keeping it simple but ...

tkeffer, 2015-12-16:
Having wee_database post the steps it took in the log is a good idea. It could definitely help with support --- instead of asking a user what steps s/he took, we can just look at the log.

With a bit of code refactoring, it might be possible to do both the DaySummaryManager init and the backfill with the same code, thus resulting in identical log entries.

gjr80, 2016-02-14:
Trying to pick up on this after returning from holidays ...
With the addition of 2 syslog lines to wee_database I have wee_database providing the following to syslog on executing a --backfill-daily:
```
Feb 14 10:17:08 jessie2 wee_database[917]: Backfilling daily summaries in database 'weewx.sdb' ...
Feb 14 10:17:08 jessie2 wee_database[917]: manager: Created daily summary tables
Feb 14 10:17:08 jessie2 wee_database[917]: manager: Starting backfill of daily summaries
Feb 14 10:17:24 jessie2 wee_database[917]: manager: Processed 87902 records to backfill 611 day summaries in 16.22 seconds
Feb 14 10:17:24 jessie2 wee_database[917]: Backfill of daily summaries in database 'weewx.sdb' complete
```
for the same command the user will see the following on the screen:
```
gary@jessie2:~$ /home/weewx/bin/wee_database --backfill-daily
Using configuration file /home/weewx/weewx.conf
Using database binding 'wx_binding', which is bound to database 'archive_sqlite'
Backfilling daily summaries in database 'weewx.sdb'...
Backfilled 'weewx.sdb' with 87902 records over 611 days in 16.27 seconds
```
Maybe I have missed or forgotten something here as I am not sure I have captured this:

> With a bit of code refactoring, it might be possible to do both the DaySummaryManager init and the backfill with the same code, thus resulting in identical log entries.

tkeffer, 2016-02-14:
Nice.

In regards to:

> Maybe I have missed or forgotten something here as I am not sure I have captured this:
> 
>  > With a bit of code refactoring, it might be possible to do both the DaySummaryManager init and the backfill with the same code, thus resulting in identical log entries.

My comment was referring to your 15-Dec-2015 comment that the messages you get from starting up `weewxd` and letting it handle backfills, vs the messages you get from using `wee_database` to do the job, are different.

I was suggesting that the logging messages could be collected into one routine, then used by both `weewxd` and `wee_database`, thus insuring identical messages.

BUT, looking at it now, I think that's a bad idea. It's a small point, which would require the introduction of an otherwise useless logging function. I'd just tidy up the messages so they look somewhat similar. A minor point.

Thanks for taking this on, Gary!!

tkeffer, 2016-02-14:
Fixed in 42ec19b68ab5d3533602f4391c531743c0bf916a

closed 2016-02-14 by tkeffer