weewx/weewx #88: non-localized output for formatted(), nolabel() and so on

Zokormazo, 2015-12-28:
When using comma separated decimal locales, the string that Formatter.toString() returns is not suitable for code output like javascript.

Would be interesting to have some methods to get non-localized output of formatted() and nolabel()

I figured out a simple way to code it, but don't know is you are interested on it nor how to call the methods.

tkeffer, 2015-12-29:
Julen,

Glad you brought this topic up. Take a look at the last section in [this document](https://docs.google.com/document/d/1CzuCPu7MsV3Rc3J6GbaMVxtrQ6z5F-xz0glvg3sUWjA/edit?usp=sharing).

The idea here is, rather than force people to assemble their own JSON-formatted output, it would be an optional formatting type, specified with a `.json` suffix:

```
$month.outTemp.max.json
```

or, for comma-separated values:

```
$month.outTemp.min.csv($delimiter=’\t’) 
```

tkeffer, 2016-01-11:
Fixed in commit 805bc5015d592516f92b29191ccc7223bd01ddae.

The "fix" requires using the (so far) undocumented function `toString()`, together with argument `$localize=False`. For example, this would produce a string that always uses a period for the decimal separator, making it suitable for JSON:

    $month.outTemp.max.toString(($addLabel=False, $NONE_string="null", $localize=False)

I'm thinking `toString()` should eventually not only be documented, but replace the existing mess of formatting functions. On the other hand, I'm not sure how it would interact with the proposed `.json` formatting, so I'm leaving it for now.

closed 2016-01-11 by tkeffer