weewx/weewx #89: Acurite driver: 01036 station requires manual configuration of Use_constants.

phillipremaker, 2015-12-31:
In order to get correct barometric pressure with an Acurite 01036, I had to manually add use_constants = True to weewx.config. Is there a way to automatically determine when to set that, and can it be some for a 01036?

USB discovery info for the 01036 (A1):
[   13.329185] hid-generic 0003:24C0:0003.0001: timeout initializing reports
[   13.339224] input: Chaney Instrument as /devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/0003:24C0:0003.
0001/input/input0
[   13.356092] hid-generic 0003:24C0:0003.0001: input,hidraw0: USB HID v1.11 Device [Chaney Instrument] on usb-2098000
0.usb-1.3/input0

phillipremaker, 2015-12-31:
This is on a Rasberry Pi, if that matters.  Also observed on Ubunu: https://groups.google.com/forum/#!topic/weewx-user/sswDL85b8x4

matthewwall, 2016-01-13:
fixed at commit f88f4a1c4ed267c8584b149e89f2e114375af043
use_constants now defaults to True instead of False.
if a 0103x station has bogus constants, one can specify ignore_bounds=True
02032 and 02064 stations should automatically be detected and use the linear function for pressure and temperature.

closed 2016-01-13 by matthewwall