weewx/weewx #91: permit meteorological days other than midnight-to-midnight

matthewwall, 2016-01-13:
for example, france uses 07:00, india uses 08:30, UK uses 09:00

see https://groups.google.com/forum/#!topic/weewx-user/kcblP3PUwU8

morrowwm, 2016-12-18:
Interesting idea. Natural gas utilities in North America use a mid-morning day boundary. Their operations are weather sensitive, and so they prefer meteorological dailies aligned with their operational day boundaries. Electrical utilities have a midnight boundary. There has been talk about aligning the two segments (https://www.aga.org/federal-regulatory-issues-and-advocacy/north-american-energy-standards-board) but it may not make sense.

evilbunny2008, 2018-06-08:
I wanted to do something similar to be able to compare local rain fall before/since 9am with Australian data, Gary is responsible for 99% of the code, I just extended it a little to do before 9am and added a minute option.

[since.py.txt](https://github.com/weewx/weewx/files/2082668/since.py.txt)

Hopefully it's helpful to others as well.

spacelama, 2020-04-25:
> I wanted to do something similar to be able to compare local rain fall before/since 9am with Australian data, Gary is responsible for 99% of the code, I just extended it a little to do before 9am and added a minute option.

evilbunny2008, last time I looked at this code, it lacked any context of how to use it - how to import it, how to call it from a template etc.  Do you have a pointer to docs for it?

evilbunny2008, 2020-04-26:
To get the rain since 9am you'd do the following.

$since($hour=9).rain.sum.formatted

roe-dl, 2021-09-05:
See [weewx-GTS special time spans](https://github.com/roe-dl/weewx-GTS#special-time-spans)

closed 2024-07-04 by github-actions[bot]