weewx/weewx #94: implement usb abstraction for multiple usb programming interfaces

matthewwall, 2016-01-16:
refactor the usb implementation for every driver that talks to hardware that shows up as human-interface device (HID).  libusb works fine on linux, but it does not work with HID hardware on macosx.  there are multiple libusb implementations (1.0.x vs 0.1.x) as well as multiple pyusb implementations (1.x vs 0.1.x vs 0.4.x), and these do not always work seamlessly (despite the efforts of the pyusb folks to provide an api that works with various backends).

as an example, see jim easterbrook's work in pywws, specifically the src/pywws/device_*.py files.

matthewwall, 2016-01-16:
some references for macosx:

http://www.libusb.org/ticket/89
https://github.com/libusb/libusb/wiki/FAQ
https://developer.apple.com/library/mac/technotes/tn2315/_index.html
https://developer.apple.com/library/mac/qa/qa1076/_index.html
http://stackoverflow.com/questions/14302935/codeless-kext-on-os-x
http://stackoverflow.com/questions/3368008/reading-and-writing-to-usb-hid-interrupt-endpoints-on-mac
http://stackoverflow.com/questions/7263648/codeless-kext-loading-problem
http://stackoverflow.com/questions/33545656/codeless-kext-for-usb-storage-device

the pywws implementation:
https://github.com/jim-easterbrook/pywws/tree/master/src/pywws

there are 5 interfaces:
cython_hidapi
ctypes_hidapi
libusb1
pyusb1
pyusb

tkeffer, 2020-11-12:
This issue is getting on 5 years old.

Is it still a problem?

closed 2024-04-25 by tkeffer