weewx/weewx #96: humidex not in unit system

tkeffer, 2016-01-25:
The observation type `humidex` gets added by the service `StdWXCalculate`, but it does not appear to have been added to the unit system. Same with `appTemp`, and, perhaps, others.

gjr80, 2016-01-25:
I always find it interesting that [Wikipedia](https://en.wikipedia.org/wiki/Humidex) states that humidex is a dimensionless number in which case you would expect it should not be defined as a 'temperature', though it is always compared to a temperature. Of course Wikipedia is not authoritative and I've not found any such statement on the Environment Canada site, common sense would say humidex should be a 'temperature'. As I said, interesting.

tkeffer, 2016-02-12:
Well, Wikipedia is wrong. 

Looking at the formula, it clearly has a temperature-based unit.

tkeffer, 2016-02-12:
Fixed in commit 3df343ef9a5a9003ee4cd82ac17273a3f42c08c5

closed 2016-02-12 by tkeffer