weewx/weewx #97: consider making a 'wee_import' utility to import archive records from an external source

gjr80, 2016-02-03:
A number of users have sought advice on how to import Weather Underground PWS data, usually as a last resort disaster recovery measure. Currently this is achieved manually by obtaining the WU data in CSV format then massaging it into a format where it can be imported using an SQL query or some other method.

This process should be able to be automated with a 'wee_xxxx' like utility. Further more the utility should be generic in nature so that additional sources (eg WOW, CWOP or some non-web based source) can be used.

gjr80, 2016-02-03:
M Wall [comments](https://groups.google.com/d/msg/weewx-development/SUOZmDkw-ww/taWcK-xwAwAJ) of 2 Feb (summary):
- use data from an arbitrary source, not just wunderground
- define a mapping between source schema and weewx schema (which may not be the default wview schema)
- schema mapping could probably be done in a config file, so the code to support multiple sources would be minimal
- specify what to do about derived quantities (e.g., apply StdWXCalculate to each record)

gjr80, 2016-02-03:
A few questions/points of clarrification:

1. **Field 'interval'**. 'interval' is an INTEGER NOT NULL field in the weewx schema. I believe weewx accepts 'interval' values that can change from record to record but currently weewx does not take this into account when calculating averages etc. Obs sourced from WU (and I suspect the same for other sources) may have irregular intervals between successive records, depending on the capabilities of the PWS and the configuration of weewx, or simply because there are missing records or the vagaries of WU. What is the preferred approach for handling 'interval'? A few possibilities spring to mind:
a. calculate the interval as the difference between the timestamps of successive records.
b. use the interval from weewx.conf or PWS
c. sample a day of observation and use the mode 'interval' (ie the most common 'interval')
a. would seem the most logical but could have some wild consequences if there are numerous missing records (could limit the max 'interval' I guess). b would seem to introduce some inconsistencies in the resulting archive as would c. Also, 'interval' is defined in whole minutes so any resulting interval would need to be rounded. Looking for some guidance here?

2. **WU WXDailyHistory.asp units**. Does WXDailyHistory.asp return metric units or am I seeing some biased results because I am in Australia? Would appreciate it is someone in USA would confim what units they see in WXDailyHistory.asp query.

3. **UV**. I do not see my UV results appearing in a WXDailyHistory.asp query. Am I special or does this the same for others as well? I have queried other stations I know have UV sensors and see the same resu

tkeffer, 2016-02-03:
I don't mean to discourage anyone, but I'm not wild about the whole idea of compensating for the lack of a proper backup by using the WU as a backup. I can imagine lazy users thinking, "Oh, I really should do a backup, but if all fails, there's always wee_xxxx."  This could be a support headache.

Perhaps if xxxx == 'dummy'?

To answer your questions.

1. This illustrates the problems with wee_xxxx, but I  would choose (a). I hate the idea of "constructing" data, which this would be, but it would result in the daily averages and sums best matching what is on the WU.

    Weewx does not handle changing values of `interval` very well. When calculating aggregates from the raw archive, it uses the SQL aggregators (e.g., `max`), which are ignorant of aggregation intervals.

2. I'm seeing US, but there is also a little gear symbol on the upper-right corner that lets you change.

3. Mine shows solar radiation, but not UV. In fact, here's the header of the CSV file:

```
Time,TemperatureF,DewpointF,PressureIn,WindDirection,WindDirectionDegrees,WindSpeedMPH,WindSpeedGustMPH,Humidity,HourlyPrecipIn,Conditions,Clouds,dailyrainin,SolarRadiationWatts/m^2,SoftwareType,DateUTC
```

matthewwall, 2016-02-03:
two more things to consider:

1) there must be provision for unit conversions.  there's no telling which units or which groups of units will be used for incoming data.  they have to be converted to whatever the weewx database is using.

2) the interval issue is an issue for more than just wee_import.  one use case is when someone has an archive_interval that does not match that of their hardware and they use record_generation=software.  they will have two archive intervals - one from the records read during catchup, and a different one from records generated 'live'.  another case is when someone *changes* their hardware archive interval (or software, for that matter) somewhere along the way, for whatever reason.

davies-barnard, 2016-02-03:
Sorry to throw my few eggs into the basket but how about a seeding facility.  

Rather than import a whole bunch of archive records limit the system to manual entry (configuration) of some known monthly/yearly max/min/avg.

If you lose your data you can seed a new set up with data from wunderground but not import archive records.

I'm sure new users would welcome the ability to make historical comparisons within a few hours of their station being online?

Chris Davies

Sent from a mobile device.

tkeffer, 2016-02-03:
@davies-barnard Unfortunately, this would be difficult with the present architecture of weewx, which assumes that the archive is 'definitive' --- everything else is derived from it. "Seeding" with statistics would require a separate, also definitive source, but only for aggregates, and even then, only for certain time periods. 

Keeping all that straight would be source of complexity, which, IMHO, wouldn't be worth it.

At least, with a wee_dummy approach, it doesn't touch the architecture of weewx itself. It's a totally independent facility.

gjr80, 2016-02-04:
@tkeffer Ok, somewhat discouraged but since I did not hear the STOP word I will not throw out the code. Maybe the way ahead is to specifically not cater for WU but cater for CSV import, MySQL/SQLIte tables with different names etc. Don't think I want to go near some of the other PWS apps.

Re the WU units, I get the gear wheel when I go therough the WU page for my station but when I do a straight up urllib2.urlopen from python I dont see any gear wheel so it would seem that something must be set somwehre. No matter your post gives me the headers for US customary (or is that WU customary).

I agree #61 is more of a priority as are many others, but #61 is probably beyond me, the accumulator does my head in most times. That being said I will trawl through the list and see what is within my capabilities!

@matthewwall At least you can infer units from WU but I guess not necessarilly from another CSV or database source. So I guess the utility will need to get that info from the user.

matthewwall, 2016-02-04:
consider this approach:

1) start with something that reads data from a csv file and sticks it into a weewx database.  it would require two inputs: (1) the csv file, and (2) a configuration file that contains (a) a mapping between csv column indices and schema columns, (b) a definition of what the units are in each csv column, and (c) database configuration in standard weewx form.

2) make that work for a variety of csv topologies (different number of columns, different mappings, different units in csv, different units in weewx database).

3) once that works, you can write a little program that pulls down data from wu and spits out two things: (1) csv file, and (2) the csv side of the column and units mappings.

4) generalize (3) into a class, then you can put everything into a single wee_import app that will handle wu downloads now and can be extended to work with *any* import later.

i'm still seeing this as a generic import tool, although it could be used to restore data in a pinch.

gjr80, 2016-02-04:
Essentially doing all those, apart from 1) (2), now but not necessarily in that order. 
Question re config files, thought I had seen the use of a seperate .conf (other than weewx.conf and skin.conf) but cannot remmeber where. As I am only reading a config file weecfg.read_config is working fine for me, is this an acceptable solution or am I setting myself up for some as yet unforeseen issues?

tkeffer, 2016-02-04:
Offhand, I can't think of any other uses of `.conf` files.

The utility `weecfg.read_config()` looks for a config file "in the usual places." Other than that, it's just `configobj.ConfigObj` with a wrapper. The configuration file you're imagining is a "one off" thing, so it won't really have a "usual place." But, this is a minor point.

I like thinking of this as a general import tool, rather than a "backup from WU" tool.

gjr80, 2016-02-07:
Like to think I've made some good progress. I now have something running that will read a CSV format text file, map CSV fields to weewx, handle either a unit system (if defined in the CSV data) or fields in any mix of know units and then save to weewx archive. I am still to sort out interval (when it is not provided in the CSV data) and I want to include something along the lines of StdWxCalculate to add in any derived obs. At the moment the rough structure other than the normal wee_xxxx front end is an abstract class Source that defines a couple of methods:

- mapRawData(). method that maps an iterable containing all raw obs data for a given time to a weewx format archive record dictionary
- saveToArchive(). method that saves one or more weewx format archive record dictionaries to archive

My CSV class is derived from this abstract class and adds:

- code to \__init\__ to parse a field/units map in a separate .conf file
- getRawData(). method that reads the data source, does any pre-processing/cleaning up of the data etc and returns an iterable or raw data records (for my CSV class this is really just using csv.DictReader()). At present I see each new data source only needing a new getRawData() method with everything else inherited.

At the moment each of the 3 methods I have defined accept or return a list of dicts, This works fine for my small test suite but am not so sure how it would handle many thousands of records. I am wondering if i would be safer implementing these methods as generators that instead of returning lists of records just yield a single record. I don't see a downside other than speed and I expect it will complicate getRawData(). I would appreciate any thoughts/guidance.

I am also looking at using StdWxCalculate to do any derived obs but I would need to override a couple of methods. Given previous expeience with overriding methods and the headaches it introduces down the track I will see where this takes me.

matthewwall, 2016-02-07:
perhaps it will help to make the use cases explicit:

case 1: import data from program X. typically used when migrating to weewx.  extract the data from program X into csv format.  create a configuration stanza that maps the csv columns to the schema used in weewx.  create a configuration stanza that maps the units of the csv columns to the units of the columns in the schema used in weewx.  run wee_import.

case 2: compare historical data. typically used to compare historical data from two or more sources.  for example, someone wants to compare their station data for the past 5 years with data from cwop and weather underground for their location. manual approach would be download data from cwop and wu and save as csv. then run wee_import to import into a cwop.sdb and wu.sdb database, respectively.  then use wee_reports to generate reports that compare data in cwop.sdb, wu.sdb, and weewx.sdb.

case 3: backup of last resort. somehow you lose some or all of your weewx database, but you have been uploading to wu or some other cloudy service.  download data from the service into csv.  use wee_import to put the data into weewx.sdb.

other use cases?

gjr80, 2016-02-08:
I think I have a good understanding of the use cases and I believe that what I have coded so far will support each of the cases above. The 2 issues I am trying to get my head around are:

1. Do I add records to the archive in one potentially very large tranche (which would be one db transaction) or do I do add them individually or in some small tranche (ie multiple db transactions)? I have the issues with `wee_database --backfill-daily` and large archives (ie small PCs running out of memory and crashing) in the back of my mind, and I want to avoid potential crashes if possible. I know there is a lot more going on in `backfill-daily` compared to just adding records to archive but I can see someone trying to import years of data. I have thought about this more overnight and think I now have a way ahead worked out.

2. Using `StdWXCalculate`. I have thought about how to use this as well, and looking through the `engine.py` and `wxservices.py` code it seems like I will need to create an instance of `StdEngine` before I can create an instance of `StdWXCalculate`. Creating and instance of `StdEngine` seems to do a whole pile of things that I don't want to do (eg loading drivers and starting to talk to weather stations), at least that is how it seems to me. Am I trying to make this aspect overly complicated, or is there a simpler way to be able to use the capabilities of `StdWXCalculate`?

matthewwall, 2016-02-08:
there are 4 things that StdWXCalculate needs from the engine: altitude, latitude, longitude, and database manager. it also needs a config_dict so that it knows which calculations to make and which algorithms to use for those calculations.

we could add a layer of abstraction so that the StdWXCalculate (a service) wraps a non-service 'WXCalculate' object. then your code could simply use that inner 'WXCalculate' object.

that might be cleaner than you creating a fake engine object just so you can instantiate a StdWXCalculate service.

matthewwall, 2016-02-08:
for issue 1, consider a tranche_size parameter.  default to a value of None, which results in a single database transaction for all records.  if set to X, then read in X records before doing a database insert.

then you can do some testing to see what the default value should actually be.  perhaps it will be a function of memory available, or fixed number of records, or just leave it at None and let users adjust if they run out of memory.

redvers76, 2016-02-12:
I started using my Aercus WeatherSleuth (aka Ambient IP1400) late in Dec 2015, but have only just got it working with weewx (using the prolabs SocketLogger driver). As such, I would like to export all data that was logged to WUnderground and re-import it in to whew... this utility sounds like it will be ideal, so if you need help in testing it, let me know! :)

peter334, 2016-02-21:
I have been following a few other WxForum related threads to get my Ambient 1400IP where I can intercept the data that normally goes to WU. I started uploading to WU in Oct 2015 and it shows all my historical data on that site. I am anxious to view what you have done in hopes that I can pull this data onto my server and create my own site.

gjr80, 2016-04-25:
> we could add a layer of abstraction so that the StdWXCalculate (a service) wraps a non-service 'WXCalculate' object. then your code could simply use that inner 'WXCalculate' object.

>that might be cleaner than you creating a fake engine object just so you can instantiate a StdWXCalculate service.

@matthewwall. More breaking ground for me, is this what you meant by a wrapper class? The `WXCalculate` class is essentially the the old `StdWxCalculate` class but picking up the config_dict, alt, lat, long and dbm from `StdWxCalculate`. In any case, I have this running in a VM on simulator and as far as I can tell it is doing what it should. Have not tried to use `WXCalculate` separately though.

```python
class StdWXCalculate(weewx.engine.StdService):
    """Wrapper class for WXCalculate.

    A StdService wrapper for a WXCalculate object so it may be called as a 
    service. This also allows the WXCalculate class to be used elsewhere 
    without the overheads of running it as a weewx service.
    """

    def __init__(self, engine, config_dict):
        """Initialize the service.

        Create a WXCalculate object and initialise our bindings.
        """
        super(StdWXCalculate, self).__init__(engine, config_dict)

        self.WXCalculate = WXCalculate(config_dict, 
                                       engine.stn_info.altitude_vt, 
                                       engine.stn_info.longitude_f, 
                                       engine.stn_info.latitude_f, 
                                       engine.db_binder.get_manager('wx_binding'))

        # we will process both loop and archive events
        self.bind(weewx.NEW_LOOP_PACKET, self.new_loop_packet)
        self.bind(weewx.NEW_ARCHIVE_RECORD, self.new_archive_record)

    def new_loop_packet(self, event):
        self.WXCalculate.do_calculations(event.packet, 'loop')

    def new_archive_record(self, event):
        self.WXCalculate.do_calculations(event.record, 'archive')

class WXCalculate(object):
    """Add derived quantities to a record.

    Derived quantities should depend only on independent observations.
    They should not depend on other derived quantities.

```

matthewwall, 2016-04-25:
gary, that is exactly what i imagined.  then we can use WXCalculate for the wee_import utility and for a 'backfill-calculations' option for the wee_database utility. m

gjr80, 2016-05-17:
FWIW I have created a repo [here](https://github.com/gjr80/wee_import) with the code I have put together so far. I have used `wee_import` to import CSV data, though it still needs quite a bit of work to handle things like rain being a running daily total rather than a record by record total. The code is structured such that a base `Source()` class is defined that contains most code for most of the tasks associated with importing data (mapping, parsing, converting and saving). A class is derived from this `Source()` class for each type of external source. This derived class is very simple, it only needs to have a method `getRawData()` defined that gets the data from the external source and returns an interable of records.

Have also utilised a `wee_import.conf` that in the main is used to let the user specify the source -to-weewx archive field mapping.

Intend to further refine CSV import so it is more robust. From there a WU import should be a very simple step.

tkeffer, 2016-05-17:
Looks good!

Never knew StrictVersion existed. Could use for weewx's setup.py

Adding 86,400 seconds to a time cannot be counted on to get you to the same time the next day because of daylight savings time. Instead, you should fiddle with the time tuple. Looks like you don't use the results anyway.

Code like

            csv_src = CSVSource(config_dict, import_config_dict, options)
            csv_raw_data = csv_src.getRawData()
            csv_data = csv_src.mapRawData(csv_raw_data, weewx.US)
            csv_src.saveToArchive(csv_data, config_dict)
 
should be unnecessary. Instead, if you got the abstractions right in class `Source`, you should  be able to instantiate the right subclass using a Factory pattern, then simply calling `run()` (or something similar), to perform the required action for _all_ subclasses. Function `run()` would be in the `Source` class and would be responsible for things like tranching and writing out to the database. It would work for all subclasses of `Source` (although, of course, it could be specialized if necessary).

Put your own copyright on it!

matthewwall, 2016-05-17:
be careful with case sensitivity of .conf stanzas.  i'd go with [CSV] instead of [Csv]

in the sample .conf, consider changing this:

```
    # Map CSV record fields to weewx archive fields. Format is:
    # weewx archive field name = CSV field name, CSV units using weewx unit 
    # names eg
    # dewpoint = dew, degree_C
    # would map the CSV field dew to the weewx archive field dewpoint and the 
    # CSV dew field will use Celsius
    # Weewx archive fields that do not exist in the CSV data should be 
    # omitted/not set
    # Any ommitted fields that can be derived may be derived using the weewx 
    # WxCalculate service

```
to this:

```
    # Map CSV record fields to weewx archive fields. Format is:
    #
    #   archive_name = field_name, unit_name
    #
    # where:
    #   archive_name is an observation name in the database schema used by weewx
    #   field_name is the name of a field from the CSV file
    #   unit_name is the name of the units for the field, as defined in weewx
    #
    # For example,
    #   dewpoint = dew, degree_C
    # would map the CSV field dew, in degrees C, to the archive field dewpoint 
    # 
    # Archive fields that do not exist in the CSV data should be omitted.
    # Any omitted fields that are derived values may be added using the 
    # StdWXCalculate service.
```

for consistency, the `calc_missing` option should be `calc-missing`.  same for `import-config`.  the command-line options uses hyphens, their corresponding variables use underscores.

you should be able to reduce the amount of code considerably by using a factory instead of the if-else

gjr80, 2016-05-17:
Thanks for the comments, seems like the learning topic of the month will be factories!
 
> Never knew StrictVersion existed. Could use for weewx's setup.py

Stumbled on `StrictVersion` when touching up Matthew's `crt` extension a while back, does indeed simplify things, especially when dealing with other than major releases. I know Matthew has used it in a few places since.

> Adding 86,400 seconds to a time cannot be counted on to get you to the same time the next day because of daylight savings time.

Looks like I have a few bits of historical code to revisit :o)

gjr80, 2016-05-19:
I have implemented a Factory pattern, I think I have done it correcty, it is certainly working as I want it. Have also taken on board the comments from 2 days ago. Repo is [here](https://github.com/gjr80/wee_import).

I still need to confirm that the utility works end-to-end, I have only been doing dry runs, but am pretty confident everything is there. I can successfully import CSV and WU data. WU, was interesting, they must gelocate you when you do a http WXDailyHistory call. I found that if I used my VPN and came up in the USA I was presented with US customary units/field names for my station ID but when I came up anywhere else I tried (about 20 countries, including Canada) I was presenteed with metric units/field names - you guys really need to change :). In any case, the WU import decodes and maps the fields accordingly, I only hope there are no other units used other than what I have coded.

I will spend the next few days giving it some more rigorous testing and refining my comments. At the moment it only imports a single day at a time from WU, that is not very practical as it really needs to handle a range of days. I think that will be fairly simple to do with a for loop and a bit of datetime manipulation, but that can wait for now.

One thing I have learnt is to make sure I have good backups and use a logger wih memory - there are so many compromises in getting obs from WU I don't ever want to have to do it!

tkeffer, 2016-05-19:
Looking better!

A few style comments. These are very minor, but why not take an already good program and make it great?

"Push policy up, push implementation down." For example, the class `Source` should not decide whether to exit the program. That's clearly a high-level decision.

"Ask for forgiveness, rather than permission." Somewhat at odds with the above, but the calling program should not check whether `Source.sourceFactory` can handle CSV or WU, it should simply pass on the type of object desired and let `Source` handle it. If `Source` can't, it should throw an exception, which the calling program may or may not handle.

So,

```python
    if options.csv or options.wunder:
        # Get an object for our source
        source_obj = Source.sourceFactory(options, args)
        # Did we get an object from the factory; if so then call its run() 
        # method to do the import.
        if source_obj:
            # Now do the import by running our object's run method
            source_obj.run()
    
    # WOW
    if options.wow:
        print "Option 'WOW' not yet implemented."
        exit(1)
```

becomes simply

```python
    try:
        # Get an object for our source
        source_obj = Source.sourceFactory(options, args)
        source_obj.run()

    except weewx.NotImplemented, e:
        print "Don't know how to do %s" % e
        exit(1)

    exit(0)
```

Similar comments having to do with option `--station` and the like: the specialized class is responsible for deciding an error has occurred, but not what to do about it. It should simply raise an exception, and let someone higher up the stack deal with the problem. Perhaps the caller wants to retry? Or, exit? Or consult with the user? 

These are the techniques that make highly reusable code. Once you're done, classes like `CSVSource` will be reusable outside the context of `wee_import`.

gjr80, 2016-05-20:
I have revised the code and think I have taken on board your style comments. The latest code is on [GitHub](https://github.com/gjr80/wee_import). The factory code no longer makes decisions on what to do with an error, it just raises them. The decision on what to do is made at the higher level. The issue that still puzzles me a little is what level of intelligence should there be at the lower level in terms of raising errors? For example, I could have something as simple as:
```
        config_path, config_dict = weecfg.read_config(options.config_path, args)
        import_config_path, import_config_dict = weecfg.read_config(None, 
                                                                    [], 
                                                                    file_name=options.import_config_path)
        if options.csv:
                return CSVSource(config_dict, import_config_dict, options)
        elif options.wunder:
                return WunderSource(config_dict, import_config_dict, options)
```
in the factory and then pick up the pieces (errors) at a higher level. Or I could have what I have now
```
        # get some key weewx parameters
        # first the config dict to use
        config_path, config_dict = weecfg.read_config(options.config_path, 
                                                      args)
        # Get wee_import config dict if it exists
        import_config_path, import_config_dict = weecfg.read_config(None, 
                                                                    [], 
                                                                    file_name=options.import_config_path)
        # Now create our object
        if options.csv:
            if options.source and os.path.isfile(options.source):
                # Return a CSVSource object
                return CSVSource(config_dict, import_config_dict, options)
            else:
                raise weewx.ViolatedPrecondition("CSV source '%s' could not be found." % os.path.isfile(options.source))
        elif options.wunder:
            if options.station:
                # Return a WunderSource object
                return WunderSource(config_dict, import_config_dict, options)
            else:
                raise weewx.ViolatedPrecondition("No station name specified.")
        else:
            raise weewx.UnsupportedFeature("No option specified or requested option not yet implemented.")
```
which puts a bit more intelligence in the factory in terms of its error detection and reporting. Obviously this makes handling the errors at a higher level somewhat easier.
Does it really boil down to horses for courses but keep the decision making on how to handle an error at the higher level?

tkeffer, 2016-05-20:
The reason to put the `if` / `else` clause down in the Factory class is that's what makes it the Factory pattern. The pattern is that you tell it what kind of object or service you need, and it goes and manufactures the appropriate object and return it. If it can't, then it either raises an exception (which would be the Python style), or returns an error code.

Part of the documentation of the Factory object is what kinds of exceptions it can raise and under what circumstances. 

This pattern applies to the next level down: the specialized classes. So, instead of

```python
        if options.csv:
            if options.source and os.path.isfile(options.source):
                # Return a CSVSource object
                return CSVSource(config_dict, import_config_dict, options)
            else:
                raise weewx.ViolatedPrecondition("CSV source '%s' could not be found." % os.path.isfile(options.source))
```
you have
```python
        if options.csv:
            # Return a CSVSource object
            return CSVSource(config_dict[CSV], import_config_dict, options)
```

Note how it's `config_dict[CSV]` that gets passed down. There is no reason for the `CSVSource` object to be aware of the rest of the `config_dict`.

Finally, you do not want to include things like reading the config file in the Factory: that's outside its domain. 

I guess if I could summarize, the objective is to make things like `CSVSource` totally self-contained and unaware that they are being used in a weewx context. That's how to make them maximally reusable.

gjr80, 2016-05-20:
Ok, I think I see, I get the `CSVSource` bit.To summarize the factory: (1) keep it simple, (2) do the minimum it needs to to manufacture whatever it is that it manufactures and (3) let the manufactured object worry about 'reporting' whether it has what it needs to do the job. Sorry this has turned out to be a Factory 101 lesson.

tkeffer, 2016-05-20:
Exactly.

Good class design is subtle. This is basically the core expertise of [Rogue Wave Software](http://www.roguewave.com/), the company I started now 26+ years ago.

gjr80, 2016-05-23:
In the course of implementing multi-day WU imports I noticed some (more) strange WU behaviour. In using the http `WXDailyHistory` call to obtain a day's worth of observations I found that subsequent calls for the same day would return different numbers of observations. Variation was only ever a handful of obs, sometimes I would get 288 (as I woudl expect) for a given day and at other times I would get something less (283 was the smallest I recall). Makes me think that once `wee_import` is done maybe we need a `wee_import_wunderfixer` !

tkeffer, 2016-05-23:
...and it's been this way for years and years. They need a new DBA.

gjr80, 2016-06-09:
A Python question if I may.

I have a base class that is never used for anything other than creating child classes. I have a property, let's say `self.prop`, that is used by one or more of the methods in the base class. Each of my child classes sets `self.prop` in their `__init__`. Should I be setting `self.prop` in the base class' `__init__` or is it acceptable to leave it out of the base class' `__init__` ? I appreciate the base class could be used by someone else for some unknown purpose, but it is not much use as it is without the methods added by the child classes. I feel that by leaving `self.prop` out I am making things confusing, but on the other hand I feel like adding it is just adding nugatory code.

tkeffer, 2016-06-12:
Don't think in terms of the mechanics, "How do I initialize this thing?" but rather in terms of what abstraction the base class represents. If `prop` is a natural property of this abstraction, then it deserves to be completely encapsulated by the base class. In that case, the base class's `__init__` would be used to initialize it.

On the other hand, if `prop` just happens to be used by the subclasses, but is not really a core property of the base, then it's probably best left out of the base class completely.

To make the abstract concrete: suppose that the base class represents an abstract source of bytes, while the subclasses represent concrete implementations. The property `prop` is the IP port to be used. In this case, `prop` is not really part of the abstraction. It's just the concrete way a particular source gets its bytes.

OTOH, say the base class is meant to be a little less abstract, and truly does represent an IP source. We don't know exactly what that source is, but like all IP sources it will include a port. In this case, `prop` would belong in the base class.

Think through the exact abstraction your base is designed to encapsulate, then where to put its properties will become obvious.

gjr80, 2016-06-15:
> Don't think in terms of the mechanics, "How do I initialize this thing?" but rather in terms of what abstraction...

Thanks, that makes sense, think I have translated the theory into practice.

I think I have `wee_import` at a stage where I can let it loose for someone else to try out/review.

To summarise where it is at:

- have implemented both CSV file and WU PWS imports
- can import multiple days of WU data through specifying `--date` on the command line
- `--date` supports hours/minutes
- have tested almost all exceptions I am trying to catch, though some recent code changes probably mean I need to revisit some of these
- the `WXCalculate` commit at https://github.com/weewx/weewx/commit/bf354c49dfb8675c868b05e1a29376557be72820 is needed to run `wee_import`, so in due course I expect `REQUIRED_VERSION` will need to be `3.6.0`
- I have found the error handling and logging/informing the user particularly challenging due to the myriad of possible combinations/issues that can arise. I am sure I don't have it 100% yet but good enough I think for some testing by others.
- I did some tests early on pulling WU data from a variety of sites (using my VPN to emerge in a variety of countries) so I think I have the WU import tied down, but it would be good for someone in another country, particularly the USA, to confirm the WU import works as expected. I will do some more testing with my VPN in due course.

I still have a number of questions (mainly about expected behaviours) but I won't complicate this comment with those questions, they will keep.

Latest version is [here](https://github.com/gjr80/wee_import)

matthewwall, 2016-06-16:
i had to modify WXCalculate so that database initialization would work properly.  i pushed back initialization of the db manager as long as possible, otherwise weewx failed to start up when there was no database.  i also made the database binding an explicit option for StdWXCalculate, with 'wx_binding' as the default.

gjr80, 2016-08-02:
In developing `wee_import` I have tried to provide the user with succinct information on what, if anything, has been 'done to the archive'. I can certainly count the records `wee_import` tries to add, I can even identify the number of unique records the `wee_import` tries to add (some real life Cumulus monthly log files I obtained had a number duplicate timestamped records - the disadvantage of using a csv/text file as the record of fact rather than a database) but at the end of the day when `wee_import` uses `Manager.addRecord()` to add a record or a sequence of records, `wee_import` cannot tell how many, if any, were actually added to archive as `addRecord()` handles any duplicate record exceptions internally and provides no result to the calling method.

At the moment the best I can do is to tell the user how many records `wee_import` processed, how many of these were unique and that the logs should be checked to see how many were actually imported (or how many were duplicates). Not really practical if importing 1000's of records. I was wondering if there is anyway that some sort of result could be returned from `addArchive()`.

I was not sure if a method that returned a value could be called both with and without any assignment of the returned value. For example, if `addRecord()` was modified (some comments left out for brevity):

```
     def addRecord(self, record_obj, log_level=syslog.LOG_NOTICE):
        record_list = [record_obj] if hasattr(record_obj, 'keys') else record_obj
        
        min_ts = None
        max_ts = 0
        recs_added = 0
        with weedb.Transaction(self.connection) as cursor:

            for record in record_list:
                try:
                    self._addSingleRecord(record, cursor, log_level)
                    recs_added += 1
                    min_ts = min(min_ts, record['dateTime']) if min_ts is not None else record['dateTime']
                    max_ts = max(max_ts, record['dateTime'])
                except (weedb.IntegrityError, weedb.OperationalError), e:
                    syslog.syslog(syslog.LOG_ERR, "manager: unable to add record %s to database '%s': %s" %
                                  (weeutil.weeutil.timestamp_to_string(record['dateTime']), 
                                   self.database_name,
                                   e))

        self.first_timestamp = min(min_ts, self.first_timestamp)
        self.last_timestamp  = max(max_ts, self.last_timestamp)
        return recs_added
```

For backwards compatibility would something like:

```
    db_manager.addRecord(rec)
```

still work while at the time something like:

```
    added = db_manager.addRecord(rec)
```

could be used where the number of actually records saved to archive was required.

I tried making the above change to `Manager.addRecord()` and assignment and non-assignment calls both seem to work as expected with no error (when called from `wee_import`). I expect something like this would be frowned upon as bad python (or programming for that matter), or is there some other more acceptable manner or piece of python wizardry that could return the number of records actually saved to archive? (I had thought of adding a property to the `Manager` class, say `recs_added` which would be set in `addRecord()`, I guess this would implicity give the number of records added rather than explicity returning the value)

matthewwall, 2016-08-28:
gary,

how about putting sample wee_import config files in the util directory.  perhaps something like this:

weewx/util/
weewx/util/import/
weewx/util/import/cumulus.conf
weewx/util/import/wu.conf

ideally these would work with no modification.

for example,

```
cd /home/weewx
./bin/wee_import --config util/import/cumulus.conf --input-dir x
./bin/wee_import --config util/import/wu.conf --id XXXX

```
but a user could copy/modify the sample conf files and put them wherever they need.

the wee_import utility would live in the bin director with wee_config, wee_device, etc.

tkeffer, 2016-08-28:
I like this. It gets the config file out of the same directory as `weewx.conf`, as well as gives the user at least a head start on writing a proper config file for whatever import s/he is considering.

gjr80, 2016-08-28:
`weewx/util/import` sounds like a good place for the conf file(s). 

I am not sure I see the advantage of the individual conf files for 'sources'. I don't think it has simplfied the command line. The single conf file can be made to be just as 'out of the box' ready as the individual conf files and of course it can be used as a starting point just as the individual conf files. Sure it is simplified in as much as it will only have options that apply to that particular config but it will need to somehow say what type of import it is configuring. I think the `[CSV]`, `[Wunderground]` etc structure keeps the user in line to some degree.

Would avoid `--config` and keep that for specifying `weewx.conf` location/name.

matthewwall, 2016-08-28:
so far there are three different input sources: csv, wu, and cumulus.  typically wee_import will be used for only one.  so having cumulus settings in the same file as csv settings complicates things for someone importing from csv.

gjr80, 2016-08-28:
So you are suggetsing a 'usage string' like:

```
usage = """wee_import --help
       wee_import --import-config=IMPORT_CONFIG_FILE
            [--config=CONFIG_FILE]
            [--date YYYY/MM/DD|YYYY/MM/DD hh:mm-YYYY/MM/DD hh:mm]
            [--calc-missing] [--dry-run]
            [--log=None|LOG_FACILITY] [--verbose]
       wee_import --version
"""
```

rather than the current:

```
usage = """wee_import --help
       wee_import --csv
            [--config=CONFIG_FILE]
            [--import-config=IMPORT_CONFIG_FILE]
            --source=CSV_SOURCE_FILE
            [--date YYYY/MM/DD|YYYY/MM/DD hh:mm-YYYY/MM/DD hh:mm]
            [--calc-missing] [--dry-run]
            [--log=None|LOG_FACILITY] [--verbose]
       wee_import --wunder
            [--config=CONFIG_FILE]
            [--import-config=IMPORT_CONFIG_FILE]
            --station=PWS_NAME
            [--date YYYY/MM/DD|YYYY/MM/DD hh:mm-YYYY/MM/DD hh:mm]
            [--calc-missing] [--dry-run]
            [--log=None|LOG_FACILITY] [--verbose]
       wee_import --cumulus
            [--config=CONFIG_FILE]
            [--import-config=IMPORT_CONFIG_FILE]
            --source=MONTHLY_LOGS_FOLDER
            [--date YYYY/MM/DD|YYYY/MM/DD hh:mm-YYYY/MM/DD hh:mm]
            [--calc-missing] [--dry-run]
            [--log=None|LOG_FACILITY] [--verbose]
       wee_import --version
"""
```

gjr80, 2016-08-28:
Disregard my last

gjr80, 2016-08-28:
So are you suggesting a usage string like:

```
usage = """wee_import --help
       wee_import --import-config=IMPORT_CONFIG_FILE
            [--config=CONFIG_FILE]
            [--id=WU_STATION_ID]
            [--input=FILE_OR_FOLDER]
            [--date YYYY/MM/DD|YYYY/MM/DD hh:mm-YYYY/MM/DD hh:mm]
            [--calc-missing] [--dry-run]
            [--log=None|LOG_FACILITY] [--verbose]
       wee_import --version
"""
```

rather than the existing

```
usage = """wee_import --help
       wee_import --csv
            [--config=CONFIG_FILE]
            [--import-config=IMPORT_CONFIG_FILE]
            --source=CSV_SOURCE_FILE
            [--date YYYY/MM/DD|YYYY/MM/DD hh:mm-YYYY/MM/DD hh:mm]
            [--calc-missing] [--dry-run]
            [--log=None|LOG_FACILITY] [--verbose]
       wee_import --wunder
            [--config=CONFIG_FILE]
            [--import-config=IMPORT_CONFIG_FILE]
            --station=PWS_NAME
            [--date YYYY/MM/DD|YYYY/MM/DD hh:mm-YYYY/MM/DD hh:mm]
            [--calc-missing] [--dry-run]
            [--log=None|LOG_FACILITY] [--verbose]
       wee_import --cumulus
            [--config=CONFIG_FILE]
            [--import-config=IMPORT_CONFIG_FILE]
            --source=MONTHLY_LOGS_FOLDER
            [--date YYYY/MM/DD|YYYY/MM/DD hh:mm-YYYY/MM/DD hh:mm]
            [--calc-missing] [--dry-run]
            [--log=None|LOG_FACILITY] [--verbose]
       wee_import --version
"""
```

matthewwall, 2016-08-28:
this is a review of the options to wee_import.

these arguments are common to any source: --source, --verbose, --dry-run, --calc-missing, --date, --config, --import-config

how about --source-format to specify CSV, WU, or Cumulus.  when we add another, there is no need for a new option, just a new value.  make source_format a parameter at the root level of the import config.  value of source_format would match the [CSV], [WU], or [Cumulus] names (analogous to the way driver works in weewx.conf).

then --source for csv would indicate a file, --source for cumulus would indicate a directory, and --source for wunder would indicate a station.

could you get rid of the --log option?  just output to stdout (and perhaps stderr?), then user can redirect to file.

exactly which parameters are required from weewx.conf?  it would be nice if we could eliminate that option, but i guess that means duplicating the database parameters in the import config.

gjr80, 2016-08-29:
Agree on the common arguments and --source-format/source.

>could you get rid of the --log option? just output to stdout (and perhaps stderr?), then user can redirect to file.

Of course, though `wee_import` can do some damage to the database so I thought it best to default log to the same location as weewx. I guess I saw a few queries coming up "but it didn't work". The only way we will solve it is to look at the logs and there might only be one chance to capture it.

I am happy to change it.

>exactly which parameters are required from weewx.conf? it would be nice if we could eliminate that option, but i guess that means duplicating the database parameters in the import config.

From memory the binding and `archive_interval` if you use `interval=conf` on the command line.

matthewwall, 2016-08-29:
i was thinking something simpler, like this:

```
wee_import --help
wee_import --version
wee_import [--source-format=(CSV | WU | Cumulus)]
    [--config=IMPORT_CONFIG_FILE]
    [--weewx-config=WEEWX_CONFIG_FILE]
    [--source=CSV_SOURCE_FILE | PWS_NAME | MONTHLY_LOGS_FOLDER]
    [--date YYYY/MM/DD|YYYY/MM/DD hh:mm-YYYY/MM/DD hh:mm]
    [--calc-missing]
    [--dry-run]
    [--verbose]

```
config would be necessary if --source-format is not specified, and in that case source_format must be specified in the import config file.

weewx-config would use the weecfg code to look for weewx.conf in the standard places, the complain and exit if not found.

tkeffer, 2016-08-29:
I can see two advantages to putting all the 'state' in the config file and doing away with command line options like `--csv`:

1. It simplifies debugging. The user need only mail the config file s/he used and we have all the info we need. This is the main reason why `weewxd` has so few command-line options.
2. What happens when we come up with a fourth, fifth, ... twelfth source? It would be simpler just to introduce a new config file, rather than an expanding list of command line options.

matthewwall, 2016-08-29:
it could be simplified even further like this:

```
wee_import --help
wee_import --version
wee_import [--source=(CSV | WU | Cumulus)]
    [--config=IMPORT_CONFIG_FILE]
    [--weewx-config=WEEWX_CONFIG_FILE]
    [--date YYYY/MM/DD|YYYY/MM/DD hh:mm-YYYY/MM/DD hh:mm]
    [--calc-missing]
    [--dry-run]
    [--verbose]
```

then the actual source must be declared in the import config file, with parameter name appropriate to the source.  for example:

```
[CSV]
    file = foo.csv

[WU]
    station = XXXX

[Cumulus]
    dir = /var/lib/cumulus
```

matthewwall, 2016-08-29:
then the util/import directory would contain one conf file for each source.  a user could put multiple source configurations into a single conf file if s/he wanted to, then distinguish using the --source option to wee_import.

gjr80, 2016-08-29:
Removing the --source argument means the user will almost certainly need to edit the import config file.

So are we agreed to drop the logging?

matthewwall, 2016-08-29:
you have a good point about the need to capture the log for diagnostics.  so how about having the log default to a file import-log-YYmmddHHMMSS in the current directory?  then make the --log option specify a different file, if desired.  and '--log -' would send the log messages to stdout.

gjr80, 2016-08-29:
>you have a good point about the need to capture the log for diagnostics. so how about having the log default to a file import-log-YYmmddHHMMSS in the current directory? then make the --log option specify a different file, if desired. and '--log -' would send the log messages to stdout.

Agree with above but current directory might cause some issues. We will likely be dealing with new users importing data from a windows app. I can see putting it in the current directory might mean never being able to find it again later. /var/tmp or /var/log ?

matthewwall, 2016-08-29:
another option would be to put the input-log in the same directory as the weewx.sdb file

gjr80, 2016-08-29:
works for SQLite but not for MySQL

matthewwall, 2016-08-29:
as for command-line options versus config file...  

i like tom's approach in weewx where everything goes in the config file - that ensures reproducibility.  however, that is at odds with one-offs or simple overrides, and is probably better suited toward daemons (weewxd) than command-line tools (wee_import).

so it should be possible to make wee_import work with only the weewx.conf config file (no import config necessary).

for that i see two approaches:

make entry points generic, using the mount approach for source-specific options:

```
wee_import --source=CSV --options=file,/var/tmp/filename.csv

wee_import --source=Cumulus --options=dir,/var/lib/cumulus

wee_import --source=WU --options=station,XXXX
```

or permit only the required source-specific options directly on the command line:

```
wee_import --source=CSV --file=/var/tmp/filename.csv

wee_import --source=Cumulus --dir=/var/lib/cumulus

wee_import --source=WU --station=XXXX
```

gjr80, 2016-08-29:
We will always need an import config file, CSV needs a field map and Cumulus needs a units definition. I would expect that sources from other weather apps woudl need something similar too. Surprisingly, WU is the easy one as it provides all you need (in an easy to parse format) in the downloaded file.

gjr80, 2016-08-29:
Have to head out so to capture this mornings exchnages. 

I think I have enough to 'refactor' to use individual import config files, a revised/reduced command line and a revised logging arrangement. Will see where that takes me. Pity, the days are warming up and all those outside jobs are calling me!

gjr80, 2016-08-29:
After a day of thinking this over perhaps I am not ready to go ahead just yet. In the end I have come to these conclusions:

1. We need to have use an import config file; CSV needs a field map and units need to be specified for Cumulus. I don't see a practical way of specifying these on the command line, nor do I see them being put in any other file.

1. Individual config files are the go.

1. Almost every first time import will require editing of an import config file:

    - The way in which `interval` is determined will need to be set; I have found during testing that for WU and Cumulus imports the most reliable method is to specify it rather than derive it. There are just too many holes in the data.

    - The CSV field map will need to be set/confirmed. The user could change the field headings in their source but that would seem to somewhat defeat the purpose of the utility.

    - The source file/folder/station ID needs to be specified. File and folder might get by with a default but station ID won't

1. The implemenation of `--source=(CSV | WU | Cumulus)` would appear to achieve two things:

    - Give the user a simple means of doing an inport with the default import config files. But since the almost every user will need to modify a config file this would not seem to achieve that intent.

    - Give the user the ability to choose between multiple import configs in a single file. Given that a user will normally only want to do a single type of import there is probably not much call for this from a user perspective.

As for questions. Unless I have missed some other benefit is there really any value in keeping `--source`? Do we drop it and accept that we won't do a multi-import config file.

Another thing that just popped into my head; if the user modifies an import config file in `util/import` will it be overwritten at next upgrade? And if so is that an issue?

matthewwall, 2016-08-29:
does the --source option aid in development/testing/debugging?  if there is no --source option, there must still be a source=(CSV|WU|Cumulus) in the config file so wee_import knows which module (code) and block (conf) to look for, right?

all of the non-source-specific options could be specified at the root level of the conf, then specialized (overridden) in any CSV/WU/Cumulus/whatever block.  this would follow the pattern used by options in weewx.conf and skin.conf.

so the suggested workflow would be:

1. create a working directory
`  mkdir /home/weewx/import`
2. make a copy of util/import/csv.conf
`  cp /home/weewx/util/import/csv.conf /home/weewx/import`
3. modify the import config
`  vi /home/weewx/import/csv.conf`
4. test run of wee_import
`  wee_import --config=/home/weewx/import/csv.conf --weewx-config=/home/weewx/weewx.conf --dry-run`
5. do it for real
`  wee_import --config=/home/weewx/import/csv.conf --weewx-config=/home/weewx/weewx.conf --log=/home/weewx/import/YYmmdd.log`

matthewwall, 2016-08-29:
during upgrade, any filenames in util/ that match originals will be silently replaced.  so we need to encourage people to make a copy of the sample conf files.  they could be copied to the user directory, but i think making a separate import directory is a better way to keep import-related conf and log from interfering with other parts of an install.

[21 more replies]

closed 2016-09-10 by tkeffer