weewx/weewx #102: Noob/nob Has issue ...

ArjanHarre, 2016-02-18:
Hi
I do not know much about linux/Ubuntu but I thought I could do this. I downloaded the files and ran the install but it has failed by the looks and I have no idea what any of it apart from the last line means. 
Weewx version is 3.4.0 
Any suggestions?

arjan@arjan-Desktop:~/Downloads$ wee_extension --install weewx-crt-0.13.tgz
Request to install 'weewx-crt-0.13.tgz'
Extracting from tar archive weewx-crt-0.13.tgz
Traceback (most recent call last):
  File "/usr/bin/wee_extension", line 83, in <module>
    main()
  File "/usr/bin/wee_extension", line 75, in main
    ext.install_extension(options.install)
  File "/usr/share/weewx/weecfg/extension.py", line 115, in install_extension
    self.install_from_dir(extension_dir)
  File "/usr/share/weewx/weecfg/extension.py", line 173, in install_from_dir
    shutil.copy(source_path, destination_path)
  File "/usr/lib/python2.7/shutil.py", line 119, in copy
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: '/usr/share/weewx/user/crt.py'
arjan@arjan-Desktop:~/Downloads$

tkeffer, 2016-02-18:
This issues list is for bug reports, not support questions. You should post your question to weewx-user.

In any case, you have a permissions problem. Try

```
sudo wee_extension --install weewx-cert-0.13.tgz
```

closed 2016-02-18 by tkeffer

ArjanHarre, 2016-02-23:
Ah K ... sorry ... will do