weewx/weewx #103: Document different types of atmospheric pressure and associated troubleshooting steps

gjr80, 2016-02-19:
User issues frequently arise relating to the different atmospheric pressure obs used by weewx, their meaning, use and troubleshooting of associated issues. Explanations of the 3 pressure obs are located variously throughout the forums and in a wiki entry. The Users's Guide also contains some limited troubleshooting steps. This user forum [post](https://groups.google.com/d/msg/weewx-user/QnmBNRbVkfo/Y5GFC7d_AwAJ) agreed to place an explanation of the 3 pressure obs and where the obs are used in the troubleshooting section of the User's Guide.

This issue was raised to facilitate discussion of the proposed additions to the User's guide.

matthewwall, 2016-02-19:
perhaps just add an entry to the troubleshooting section of the user guide?  something like "the pressure on my console does not match the pressure reported by weewx"

we might need a "meteorological references" section in the user guide as well.  that section would contain the definitions of pressure (then remove the wiki entry), definitions of humidex/apptemp/heatindex/windchill.

tkeffer, 2016-02-19:
I'm concerned about how big the User's Guide is becoming. Downright intimidating for the newbie. Its purpose should be on configuring an instance of weewx and how to get it up and running.

I know that I'm the one who suggested putting a pressure explanation in the Troubleshooting section but, since then, the Wiki entry has gone up and I'm satisfied with it.

There's also more room to put all this stuff in the Wiki.

davies-barnard, 2016-02-19:
Sorry to hijack this thread but could we move lots of the more advanced
stuff into the wiki rather than adapt the userguide to suit more devices?

tkeffer, 2016-02-19:
I agree that it's probably time for a docs "refactoring," but can you give me an example of "more advanced stuff?"

davies-barnard, 2016-02-19:
I'm guessing for me it does not make sense that stuff relating to
customising weewx and writing extensions is split across two sites/formats.

I would expect to see everything to do with setting up Weewx in the user
docs with a basic troubleshooting guide.  I would not expect this to need
rewriting every version. Everything else I would expect to see in the
wiki?  The wiki would make it much easier to separate basic need to know
knowledge from deeper science/developer knowledge.  Currently the user
guide pages are sooooo long!

One addition I would be keen to add to a wiki is an explanation of where to
add searchlists in order to make them available to different templates.  I
was confused a bit the other day when trying to access searchlist
extensions in the main template which were working in the extension
template.  I solved in the end with a face palm moment.

Sorry, that was supposed to be a short reply.

Chris

gjr80, 2016-02-19:
Well, this is interesting, created this issue so I could discuss a few things I was considering adding to the User's Guide and now I feel like I am hijacking my own thread :) I appreciate the original conversation where adding the '3 pressures' started was really about adding a single paragraph to the troubleshooting section, but when I sat down to write this it seemed to beg the question if you are putting pressure info in the troubleshooting section what about the questions that seem to come regularly? (eg 'my console displays XXXX but weewx displays YYYY' or 'I have a calibration expression for barometer but it has no effect'). I saw these issues as worthy of including in the troubleshooting section, which would then require an explanation of the '3 pressures', even though one may consider adding the '3 pressures' by themselves as belonging somewhere other than the troubleshooting section.

If things have moved on and the new wiki entry is all that is required that is fine with me.

gjr80, 2016-03-10:
I appreciate this issue is closed but the section added by Matthew is somewhat similar to what I had in mind when I opened the issue and started work on until the above seemed like it might take the docs down a new path. I have first drafts of the following topics under a general 'pressures' section in troubleshooting:

- Weewx atmospheric pressures explained
- Pressure/Barometer/Altimeter corrections in [StdCalibrate] do not work
- Weewx displays XXXXX, my console displays YYYY
- No barometer data
- Problems with uploads to external services
- AWEKAS quality control indicates my pressure is incorrect

these headings were based on a search through the forums for pressure related issues raised by users. I will rework what I had based on Matthew's _Meteorological problems_ section and offer up a pull or new issue. Not the brainstorm mentioned above but I think useful info.