weewx/weewx #110: StdWXCalculate does not honor 'None'

tkeffer, 2016-03-21:
The docstring leads one to believe that setting a derived variable to `None` would prevent it from being calculated:

```
            [[Calculations]]
                windchill = hardware
                heatindex = prefer_hardware
                dewpoint = software
                humidex = None
```

However, it is always calculated.

matthewwall, 2016-03-21:
fixed at commit https://github.com/weewx/weewx/commit/872a53272cd3a48765f163b1f9a25cd5d1ca98d1

closed 2016-03-21 by matthewwall

tkeffer, 2016-03-21:
Matthew: a question. Is the intention that if a derived type is not listed in `[[Calculations]]` then the default is `prefer_hardware`? Or, that it not get calculated at all?

matthewwall, 2016-03-21:
the intent is that if nothing is specified it will default to prefer_hardware

one must explicitly set to 'none' to disable a calculation