weewx/weewx #111: TypeError: coordinate list must contain exactly 1 coordinate

aroundmyroom, 2016-04-06:
cheetahgenerator: Generated 7 files for report StandardReport in 6.58 seconds
Apr  6 11:00:43 raspberrypi weewx[15746]: genimages: Generated 24 images for StandardReport in 16.68 seconds
Apr  6 11:00:43 raspberrypi weewx[15746]: reportengine: copied 0 files to /var/www/html
Apr  6 11:00:53 raspberrypi weewx[15746]: GaugeGenerator: Generated 7 images for HTMLPages in 9.28 seconds
Apr  6 11:00:53 raspberrypi weewx[15746]: historygenerator.py: Generated 6 tables in 0.34 seconds
Apr  6 11:00:58 raspberrypi weewx[15746]: cheetahgenerator: Generated 10 files for report HTMLPages in 5.35 seconds
Apr  6 11:00:58 raspberrypi weewx[15746]: reportengine: copied 0 files to /var/www/html/Bootstrap
Apr  6 11:00:59 raspberrypi weewx[15746]: reportengine: Caught unrecoverable exception in generator user.translategenerator.ImageGeneratorTranslated
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****  coordinate list must contain exactly 1 coordinate
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****  Traceback (most recent call last):
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****    File "/usr/share/weewx/weewx/reportengine.py", line 140, in run
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****      obj.start()
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****    File "/usr/share/weewx/weewx/reportengine.py", line 170, in start
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****      self.run()
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****    File "/usr/share/weewx/weewx/imagegenerator.py", line 31, in run
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****      self.genImages(self.gen_ts)
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****    File "/usr/share/weewx/weewx/imagegenerator.py", line 218, in genImages
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****      image = plot.render()
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****    File "/usr/share/weewx/weeplot/genplot.py", line 178, in render
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****      self._renderTopBand(draw)
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****    File "/usr/share/weewx/weeplot/genplot.py", line 383, in _renderTopBand
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****      font=unit_label_font)
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****    File "/usr/share/weewx/weeplot/genplot.py", line 578, in text
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****      return ImageDraw.ImageDraw.text(self, position, string, **options)
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****    File "/usr/lib/python2.7/dist-packages/PIL/ImageDraw.py", line 274, in text
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****      self.draw.draw_bitmap(xy, mask, ink)
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****  TypeError: coordinate list must contain exactly 1 coordinate
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****  Generator terminated...
Apr  6 11:00:59 raspberrypi weewx[15746]: reportengine: Caught unrecoverable exception in generator user.translategenerator.ImageGeneratorTranslated
Apr  6 11:00:59 raspberrypi weewx[15746]:         ****  coordinate list must contain exactly 1 coordinate

dunno what the issue can be .. but some graphs are not made

matthewwall, 2016-04-06:
please try this patch and let us know if it fixes the problem:

at line 63 of bin/weeplot/genplot.py change this:

        self.unit_label_position    = (10, 0) * self.anti_alias

to this:

        self.unit_label_position    = (10 * self.anti_alias, 0)

matthewwall, 2016-04-06:
fixed at commit 4353da9..d2c5911

closed 2016-04-06 by matthewwall