weewx/weewx #114: consider a replacement for StandardReport

matthewwall, 2016-04-10:
the existing StandardReport is long in the tooth:
- use of buttons instead of links
- duplication in html, especially tables
- requires customization, even for basic installations
- information layout is less than ideal, e.g., arcane station info has higher priority than weather data, oversized title bar, inconsistent navigation controls

implement a streamlined StandardReport to replace the existing one.  it should:
- use modern (but not bleeding edge) html conventions
- eliminate the buttons
- display well on any device
- use tufte-esque principles for information display
- display more elements from the wview schema so customization is not required for basic installs (e.g., automatically conditionally display all of the 'extra' data)
- display station status (e.g., current rf signal, battery status) when available

this is a good starting point, but it needs to be pruned:
https://groups.google.com/forum/#!topic/weewx-development/NDAt327kbts

davies-barnard, 2016-04-10:
Hi,

I've been busy with end of year exams and other stuff but aim to have some
time to finish this off over the summer if that suits.

Chris.

matthewwall, 2016-12-07:
latest candidate is in newskin branch.  snapshot of simulator output is visible here:

http://lancet.mit.edu/mwall/projects/weewx-newskin-161104

matthewwall, 2016-12-07:
as part of the refactoring, i pulled the smartphone and mobile reports out of StandardReport

i propose that these still ship with weewx, but as separate, standalone reports.  they would be disabled by default (fully configured in weewx.conf, but enable=false)

i propose to add one more report in the weewx distribution.  this is tentatively named 'iconic' and is very simple.  should have candidate ready later this week...

matthewwall, 2016-12-09:
the 'iconic' view is more appropriate for forecasting.  i added it to the forecast extension.

matthewwall, 2017-02-28:
the newskin prototype now has 3 pages: index.html, statistics.html (table of hi/lo values for week, month, year), and sensors.html (current sensor state plus history for battery status, voltage, etc).  there is also a google map block (disabled by default).

matthewwall, 2017-08-20:
this issue has been superceded by issue #248

closed 2017-08-20 by matthewwall