weewx/weewx #115: consider shipping weewx with a schema in addition to wview schema

matthewwall, 2016-04-11:
the wview schema is long past its prime - its naming of fields is not consistent, its field names are biased toward davis hardware, it does not handle multiple sensors well, it does not include many commonly requested fields (cloudbase, humidex, appTemp, maxSolarRad).

although modifying/extending the schema is not terribly difficult, it is for a noob, and it is yet another step.

some non-exclusive options to this problem follow:

option a: publish an 'expanded' schema on the weewx wiki.  provide instructions for how to replace the wview schema with that one. is it as easy as putting the new schema in the user dir, changing the 'schema' directive in weewx.conf, stopping weewx, deleting old db, starting weewx?

option b: make $current.X work properly in templates.  this does not address plotting of historical values not in wview schema, but it would help in the cases where someone just wants to display the current value of observations not in the schema.

option c: publish an expanded schema in the bin/schemas directory.  provide option at install time to choose wview schema or the expanded schema.

option d: do option a, but package as an extension.  is it possible to make the extension installer do all of the steps, including moving the old database aside?  or even copying over data from the old database into a new one?

of course this also raises the question of what the expanded schema should look like.  is it the wview schema plus fields?  how future-proof can we make an expanded schema so that this does not have to happen again in a few years?

m

tkeffer, 2016-04-12:
Clear preference for option b, plus c. Make the "expanded" schema the default. 

I don't see any reason to even mention the wview option at install time --- it would only confuse people. Instead, we can document its existence.

matthewwall, 2016-05-03:
there is a prototype of an extended schema in the 'schemas' branch.

- added schemas/weewx.py
- change to the default weewx.conf
- the experimental/transfer_db.py is also affected, but i did not modify that code
- the vantage driver had an explicit reference to the wview schema in dump_logger
- weecfg/__init__.py
- weewx/manager.py
- gen_fake_data.py has explicit reference to the wview schema

so changing the default schema is easy enough, but i think we need to figure out a better way to extend the units.  the prototype schema weewx.py does this at the end of the file.  it is a kludge.  we need a mechanism that is more modular and that will be respected no matter how the schema/units are used.  for example, weewxd should end up with the same schema/units as wee_reports, and if an extension provides its own schema/units, those should be respected/used by both weewxd and wee_reports.

tkeffer, 2016-05-04:
1. No idea why `dump_logger` in the vantage driver has an explicit value for the option `default_binding_dict`. The default value would work just fine. I've deleted it.

2. The reference to wview in `weecfg__init__.py` is used when upgrading from something earlier than V3.X. It can be changed to the weewx schema.

3. The references in `weewx/manager.py` can be changed without a problem.

4. Any schema that contains the tested types will work for `gen_fake_data.py`. 

Regarding defining units. The problem of types in a database schema is orthogonal to the problem of which units a type uses. Specifying units should be an independent facility. But, you probably already reached that conclusion.

I don't see any reason why all the stuff in `units.py` can't be specified in a configuration file instead of in Python code. Then it's more accessible to extension writers.

How to implement? Two possibilities come to mind.

1. Extensions could add new unit types to a configuration file by doing a merge at install time. This is similar to what we do with `weewx.conf` now. Indeed, we could press `weewx.conf` into service for this, but it's obese enough. I'd favor a new configuration file, call it `units.conf`. See more below.
2. Alternatively, we could have a units subdirectory. On startup, `weewxd` (and `wee_reports`) would scan the directory for all configuration files and compile the full set of unit types. This would make writing extensions much easier (just plop your config file into a subdirectory),  but startups slower.

I think I prefer approach 1. Quicker startup, more predictable results. 

Frankly, there's a lot of Python code that could be moved to config files. For example, the accumulator extractors. 

I've always liked that all of the configuration state in weewx is in one file, easily emailed. I've always thought this was a better approach than wview's many config files. But, perhaps it's time to add a few specialized config files, useful only to extension writers. No way would a user be expected to go into them.

matthewwall, 2016-05-04:
try this on for size.  weewx.conf contains the definitive definition (or reference) for schema, units, and aggregation methods.  for any one of those, the defaults can be overridden or extended.  but this must be done via weewx.conf, not by arbitrary code in a driver (e.g., the way klimalogg extends the units).  that way *anything* that uses a given weewx.conf file will use the same configuration.  that probably also means that we deprecate the use of user/extensions.py, or, if not, then ensure that it is loaded by every utility.

for example, the schema.  default would be specified as it is now:

```
schema = schemas.weewx # the default schema
schema = user.extensions.myschema # extension of the default schema
schema = user.klimalogg.schema # complete override of schema

```
for units, it could look like this:

```
[Units]
    units = weewx.units.unit_systems # default unit definitions
    units = user.extensions.my_unit_systems # extend the defaults
    units = user.klimalogg.unit_system # extend the defaults with klimalogg-specific observations

```

i'm not sure yet what the aggregators look like.  it needs to be a syntax that lets you say "things in group_rain should be summed; things in group_temperature should averaged; things in group_count should not be averaged; things in group_x should be the last value"

the point here is that there is a single entry point for each of units, schema, and aggregations, yet each can be extended or overridden.  still a single weewx.conf so no ambiguity, but some of the definitions live outside of the .conf file itself.

i don't know how we could make a directory-based approach work, especially for a mutually exclusively extending something like units (and the order would matter too).

i'm not sure yet whether the unit labels and other skin-specific unit stuff belongs in the weewx.conf [Units] stanza.

tkeffer, 2016-05-04:
A couple of things worry me about all this.

1. What happens when two extensions try to define the same observation type? A namespace may be needed.
2. Conversions. Adding a new unit frequently requires defining conversion functions (dictionary `weewx.units.conversionDict`, which must be in Python.

tkeffer, 2016-11-26:
With the resolution of issue #13, it's now possible to do your option b (see your first post).

It occurs to me that your option c is possible right now. There's nothing stopping us from adding new observation types to the existing wview schema. Code in the database manager already makes sure that we only insert the intersection of types in the schema and types in the archive record.

matthewwall, 2016-11-26:
excellent.  the file bin/schema/weewx.py in the 'schema' branch is my best guess at what the new schema should be:

https://github.com/weewx/weewx/blob/schema/bin/schemas/weewx.py

we could ship weewx with both weewx.py and wview.py in the schemas directory, but everything would default to using the weewx schema.

tkeffer, 2016-11-26:
> https://github.com/weewx/weewx/blob/schema/bin/schemas/weewx.py

This needs definitions for `obs_group_dict`, `extract_dict`, and `accum`. Unfortunately, including them tickles the bug you mention in issue #54.

The fix is to rename the weewx schema file to something other than `weewx.py`. Frankly, I don't know why we can't just add stuff to the wview schema. Am I missing something?

matthewwall, 2016-11-26:
because as soon as we add to it, it is no longer the wview schema? :)

how about wview-extended.py?

i'd rather not start the life of a new schema with bad naming...

perhaps more important is how we deal with the obs_group_dict, extract_dict, and accum are more important.  although they are not part of the schema, they are closely coupled with the schema, and at some point we need to figure out how to make the process of extending them more manageable.

extending them is easy, but supporting them when extended is anything but.

matthewwall, 2016-12-24:
although we could easily introduce the extended schema in the next weewx release, i would like to see whether it is reasonable for us to introduce a mechanism for extending (and, to a lesser extent, overriding) the unit groups and accumulators.

first of all, we should ensure that the unit group and accumulator are specified for every field in the extended schema.

this means there is one and only one definition of unit groups and accumulators.  so there would be only one place an extension could modify if it were to introduce a new group or accumulator.

then, if someone wants to override or extend those, then they could do it in the config file like this:

[Units]
    [[Groups]]
        [[[US]]]
            group_temperature = degree_F
        [[[METRIC]]]
            group_temperature = degree_C
        [[[METRICWX]]]
            group_temperature = degree_C
    [[Observations]]
        inTemp = group_temperature
        outTemp = group_temperature
        inHumidity = group_percent
        outHumidity = group_percent
        rxCheckPercent = group_percent
[Accumulators]
    snow = sum
    hail = sum

if a group is specified in [[Observations]] but not in any unit system, then processing of the units/groups should fail hard.

accumulator types are 'sum', 'last', 'avg'.  'noop' is used for any field that has no accumulator specified.

(these are all fields in the extended schema, so they should already be defined in the code.  i use them here as examples for config file syntax)

that way the units, accumulators, and schema are all specified in one place - weewx.conf.  so every entry point should pick up a complete units/accum/schema specification, whether that is weewxd, wee_reports, or anything else.

tkeffer, 2016-12-24:
I like this.

My one concern is this will add a lot of bulk to `weewx.conf`. While I like having all the config information in one place, it may be time to introduce a separate file, `units.conf`, which will hold this information. Perhaps it should even be generalized to hold all config info that really should not be messed with by the average user (`system.conf`?). We could put things like `[StdConvert]` in there.

tkeffer, 2016-12-24:
Something like this?

~~~
[Observations]
    [[outTemp]]
        schema = real
        group = group_temperature
        accumulator = scalar
        aggregate = avg
    [[rain]]
        schema = real
        group = group_rain
        accumulator = scalar
        aggregate = sum
    [[ ... ]]

[Units]
    [[Groups]]
        [[[US]]]
          group_temperature = degree_F
          group_rain = inch
          ...
        [[[METRIC]]]
          group_temperature = degree_C
          group_rain = cm
          ...
~~~

matthewwall, 2016-12-24:
the [Observations] stanza as you posted makes much sense.

i'd like to figure out a syntax that says "METRICWX is the same as METRIC, but with these overrides" - i'm not sure how to do that in weewx.conf-speak.

the default set of units and accumulators does not need to be specified in weewx.conf.  only the overrides and additions would be in weewx.conf, so in most cases, the [Observations] and [Units] stanzas would be empty.

i'm a bit leery of having implicit defaults, but i am resigned to the fact that this is just the nature of the beast: obvious-to-understand vs. too-much-to-read.  either you put *all* of the default values in the conf (possibly commented to show they are defaults), or you put *none* of the defaults.

matthewwall, 2016-12-24:
btw, having the defaults hard-coded (so only overrides and additions in weewx.conf) is robust in that we ensure that everyone starts with a sane unit definition (unless they modify the python code).  if we made units defined *only* from conf, we might have more support issues.

matthewwall, 2016-12-24:
i'm not crazy about a separate units.conf, and definitely not crazy about arbitrary distinctions about what an 'average' user might or might not do.

remember that we need a 'defaults' section for StdReport, and that will contain a units stanza.  if there is a separate units.conf, that will confuse many people (starting with me :)

tkeffer, 2016-12-24:
Thinking about this a bit more, I'm liking the idea of organizing around observation types less, and the idea of organizing around facilities more. 

The rest of weewx is organized around services and facilities, not types. Organizing around facilities also makes it easier to add new facilities, or subtract old ones. So, I'd rather see:

~~~ini
[Schema]
    outTemp = real
    rain = real
    ...

[Groups]
    outTemp = group_temperature
    rain = group_rain
    ...

[UnitSystems]
    [[US]]
        group_temperature = degree_F
        group_rain = in
        ...
    [[METRIC]]
        group_temperature = degree_C
        group_rain = cm
        ...

[Accumulator]
    [[outTemp]]
        accumulator = scalar
        aggregation = avg
    [[rain]]
        accumulator = scalar
        aggregation = sum
    [[ ... ]]

[Derived]
    dewpoint = weewx.formulas.dewpoint
    wind = weewx.formulas.wind
    ...
~~~

NB: I've added a bonus section `[Derived]` for observations calculated from other observations. Functions listed here would have a signature that looks like:

~~~python
def calc(record, obs_type)
~~~

Also, upon reflection, I agree with you about just listing deviations from the defaults. In that case, the example above is just illustrative of the structure and not to be taken literally: types such as `outTemp` and `rain` would be included in the default configuration, and not in the configuration file.

I suspect you're also right about not using a separate `system.conf` configuration.

tkeffer, 2016-12-25:
I implemented the accumulators part of this in commit https://github.com/weewx/weewx/commit/344d32fb90ab11942e7568e1f92a41b88a69fa8c

It's pretty much as I outlined above, with one exception: I forgot that in addition to an accumulator type and an extract function, there's also an 'add' function. So, for the record, it looks like:

~~~ini
[Accumulator]
    [[outTemp]]
        accumulator = scalar
        adder = add
        aggregation = avg
    [[rain]]
        accumulator = scalar
        adder = add
        aggregation = sum
    [[ ... ]]
~~~

The `weewx.conf` file need only list non-default values (so, the above is not really necessary).

tkeffer, 2016-12-25:
Occurs to me, one problem with the config file only including overrides and additions to the default is that it would make it impossible to remove things from the schema. This is desirable in many applications in order to save some space. 

For the schema we might want to either stick with the Python-based solution we have now, or switch to a separate config file that the user can edit.

vinceskahan, 2019-10-05:
It's been 3 years since you guys gave this one a lot of thought, I was wondering if it's still alive strategically....

In trying to follow the threads above I might have misread where you got to at that time in terms of design, but I wanted to throw out a couple thoughts from my DevOps point of view.  In short - I love componetization and feature flags.   I'm wondering if it's possible to reach some kind of additive solution.  How'bout this:

- leave the current wview-derived schema as the baseline
- permit people to 'add' to it via the equivalent of an include file or equivalent
- don't worry about size.  Few people have complained about the baseline schema and resulting size

So if there's a new station type, using WeatherFlow as an example since I have one, the driver writer would need to either:

- map the available observations into the default schema, excluding what doesn't map
- or provide their own 'additive' schema to add elements to the db (example - lightning distance)
- or even maybe provide a complete 'alternate' schema and turn a switch to disable the default one

I guess what I'm suggesting is more componentization at the db level and less of an (extensible) monolithic schema.

That said, another alternative might be to leverage the multiple-db capability we already have:

- a weatherflow station might ' provide+enable its own 'complete' custom schema, likely writing to a wf-unique secondary db
- downside is this breaks all the skins since they're bound to the primary db.  Maybe that's something that's fixable under the hood at a high level.  Provide a switch saying which db is the primary one that a default skin should look into for data

The additive thing is where I'd go personally, if possible.  That would let somebody with custom MQTT sensors they want to add be able to add elements they need at their location only (ie, barnTemperature or upsVoltage or whatever) without needing to roll their own custom extension and secondary database(s)

Of course it's way easier to say than do, but I thought I'd throw a couple alternatives out there and get the kevlar on :-)

matthewwall, 2019-10-06:
how about the following:

1) ship weewx 4 with the 'wview-extended' schema as the default.  you can see this schema in the 'schema' branch: https://github.com/weewx/weewx/blob/schema/bin/schemas/wview_extended.py

2) ship additional schemas for common monitoring configurations.  i know of one examples right now: solar/battery monitoring.  this is useful for drivers that should share the same schema, so the driver author does not have to invent a new schema, and so that users can expect common schema definitions across hardware from different vendors.  for example, there are drivers for solar controllers from sunny web box, victron, outback, etc 

3) for other configurations, encourage the design pattern of shipping a schema with your extension.  this can be done by embedding the schema in the .py file for the extension (see weewx-cmon or weewx-swb), or it can be done by including a separate schema file with the extension.

i am not sure about (2).  as i review the drivers for all of the different solar controllers, the observations they produce vary wildly.  any common subset would be very small.  but by providing that as a 'standard' schema, at least the common names would be consistent across drivers.

as for examples for (3) i know of at least these: cmon, pmon, per-circuit energy monitor (GEM), solar/battery monitor, tide depth, tank level, propate/water usage.

tkeffer, 2019-10-08:
I like the idea of switching to a superset of the wview schema, but with about 120 new types, the wview_extended schema introduces too many that would be of marginal usefulness. 

New types:
```
appTemp
appTemp1
appTemp2
appTemp3
appTemp4
appTemp5
appTemp6
appTemp7
appTemp8
batteryStatus1
batteryStatus2
batteryStatus3
batteryStatus4
batteryStatus5
batteryStatus6
batteryStatus7
batteryStatus8
batteryVoltage1
batteryVoltage2
batteryVoltage3
batteryVoltage4
batteryVoltage5
batteryVoltage6
batteryVoltage7
batteryVoltage8
cloudbase
co
co2
dewpoint1
dewpoint2
dewpoint3
dewpoint4
dewpoint5
dewpoint6
dewpoint7
dewpoint8
extraHumid3
extraHumid4
extraHumid5
extraHumid6
extraHumid7
extraHumid8
extraTemp4
extraTemp5
extraTemp6
extraTemp7
extraTemp8
forecast
hailBatteryStatus
heatindex1
heatindex2
heatindex3
heatindex4
heatindex5
heatindex6
heatindex7
heatindex8
humidex
humidex1
humidex2
humidex3
humidex4
humidex5
humidex6
humidex7
humidex8
inDewpoint
leafTemp3
leafTemp4
leafTemp5
leafTemp6
leafTemp7
leafTemp8
leafWet3
leafWet4
leafWet5
leafWet6
leafWet7
leafWet8
lightning_distance
lightning_disturber_count
lightning_energy
lightning_noise_count
lightning_strike_count
luminosity
maxSolarRad
nh3
no2
noise
o3
pb
pm10_0
pm1_0
pm2_5
signal1
signal2
signal3
signal4
signal5
signal6
signal7
signal8
snow
snowBatteryStatus
snowDepth
snowMoisture
snowRate
so2
soilMoist5
soilMoist6
soilMoist7
soilMoist8
soilTemp5
soilTemp6
soilTemp7
soilTemp8
uvBatteryStatus
windrun
```

I would trim this list to:
```
appTemp
appTemp1
cloudbase
co
co2
dewpoint1
heatindex1
humidex
humidex1
inDewpoint
lightning_distance
lightning_disturber_count
lightning_energy
lightning_noise_count
lightning_strike_count
luminosity
maxSolarRad
nh3
no2
noise
o3
pb
pm10_0
pm1_0
pm2_5
signal1
signal2
signal3
signal4
signal5
signal6
signal7
signal8
snowBatteryStatus
snowDepth
snowMoisture
snowRate
so2
uvBatteryStatus
windrun
```

The `signal[1-8]` types would be general-purpose "spare" types.

vinceskahan, 2019-10-08:
Wondering if we're worrying too much about making the field 'names' fit all
use cases, rather than making the gross 'quantity' of elements big enough
to fit all use cases.

What if you did this:

   - leave the default wview-based items in the TBD extended schema as-is,
   keeping the wview compatibility for legacy reasons in skins etc.
   - simply add 32 new elements, call them something generic ala sensor[1-32]
   or even user[1-32] to make it clear they're not core elements
   - require the extension user to map extension-expected-name 'foo' to a
   actual-database-field-name 'sensorN'
   - require the extension author to validate this was done by their
   customer, throwing the right errors if they skipped that mandatory step
   -

So for a purpleair extension, that .py code wants to write to a pm1_0_cf_1
element name.   If it can map that to sensor23, the code will work.   If it
can map it to any database element name the code will work.

Genericize the heck out of it.  Just add more fields and then you don't
need to do an additive schema, you just made yourself a couple dozen
generic available db fields to map and use.

Picking a number out of thin air, I'd say 32 new generic fields.

matthewwall, 2019-10-09:
i would suggest using only specific fields in the database.

the problem with generic fields is that they are ... generic.  the problem with a generic field is that we cannot specify units, we cannot specify aggregators, and they are not directly usable in skins.

tkeffer, 2019-10-09:
In any case, can someone else push this to completion? I have my  hands full with the Python 2 to 3 transition, as well as extensible types.

vinceskahan, 2019-10-09:
Guess I'm not understanding why the database can't have a bunch of generic
elements in it, assuming the user 'would' need to lightly edit weewx.conf
to salt to taste for how they used those fields for their custom purpose.

If I wanted to use a generic 'sensor23' to capture a temperature in degF
wouldn't I just define sensor23 to be in group_temperature and be good to
go in the skins etc. ?

I guess I'm missing all the moving parts in the .conf file somehow.

The customization guide is written around needing to add to the schema.
I'm trying to mentally walk the steps to use that element in a custom
user-defined way, once the schema element is present in the db.   Assuming
a generic sensor23 is in the extended schema, what moving parts would the
user need to edit into weewx.conf to salt to taste to pick what kind of
thing that is, its units and aggregators, etc ?

Is this a no-go completely ?

vinceskahan, 2019-10-09:
Of course - you're working far more important things.

I was just hoping to get the 'what is the plan' stuff talked through a bit
as this one's been open a while.    Currently it seems like the
really-extended one has some downsides, the genericized extended one has
downsides (or blockers), so we're kinda at status-quo re: a path forward.

Sorry to clutter up the github issue with replies.  Matthew if you want to
talk via email or even the -dev group, let me know.

matthewwall, 2019-10-09:
the reason that generic columns are not desirable include:

* a generic column requires some kind of modification before it can be used.  someone must specify units and aggregation method, at the very least.  this is still a non-trivial modification in weewx.

* there is no way to build a reliable skin that refers to a generic column.  what if a user used generic column X for one purpose, but the skin wants to use the same column for a different purpose, with different units/aggregation?

these were my design goals with the wview_extended schema:

1) add fields that seem like they should be there but that wview left out.  for example, extraTemp1-3 are there but only extraHumid1-2

2) use naming that matches wview conventions for names that were already there, and all lower-case or 'standard' names for new items.  for example, the particulate names and signal strength.  wview is inconsistent about case, and case matters a lot when using other databases.  lowercase or snake_case is far more robust than camel case.  (i supposed one might argue for a complete break from wview that uses all snake_case.  but that breaks backward compatibility with a lot of things.)

4) provide a few of the derivative fields that everyone asks for - apparent temperature, humidex

5) provide fields for all of the observations used in the seasons skin, for hardware with up to 8 additional temperature/humidity sensors, 8 soil temperature sensors, 8 soil moisture sensors, 8 leaf temperature sensors, 8 leaf wetness sensors.

6) provide the 'secondary' fields for extended sensors too.  that means battery, signal strength, heatindex, dewpoint.  that results in a lot of 'extra' fields - battery, signal strength, temperature, humidity for up to 8 additional sensors.

i supposed we could eliminate the extra leafWet, leafTemp, soilMoist, soilTemp fields.  if someone has that many of those sensors, they must extend the schema.

but i would rather not eliminate the extra t/h sensors (and their battery and signal values).  i use those all the time on sites where we measure refrigerator and freezer temperatures, or where there are accurite temperature/humidity sensors scattered all over a site with sdr picking up the signals.

tkeffer, 2019-10-09:
Just for the record: both SQLite and MySQL are case-insensitive when it comes to column names. Either of these are valid:

```
SELECT min(outTemp) from archive;
SELECT MIN(OUTTEMP) FROM archive;
```

Of course, `outTemp` and `out_temp` cannot be interchanged.

vinceskahan, 2019-10-10:
I'll of course defer to your guys' decisions, but I can't personally use
the extended schema as-is because some of the extensions I use don't use
the same naming you picked.  This is why I wanted to explore the 'could it
be additive' or 'how could it be genericized' possible paths forward.  That
would almost certainly let folks meet conceivable needs that I can think of.

I don't really buy Matthew's two reasons why generic doesn't work, as you
say you already 'do' this with a custom schema meaning a custom skin for
your custom use already.   I do the same things, albeit with secondary
databases since whew upgrades would stomp my edits to the default schema.

I always expected that if we went the generic route, there could be
namespace collisions among skins if everybody wanted to use extraTemp5 or
sensor23 or whatever.    Genericizing names vs. pre-canned ones doesn't
stop namespace collisions among skins (for example - how many skins assume
extraTemp1 is 'theirs' ?). By definition almost, if you use the generic
fields you're taking the task to make your skin(s) work with them.  That
seemed ok to me personally, no clue if it's a good global expectation or
not.

But if we continue to ask folks to either cook up alternate db (perhaps an
example minimalist template extension pattern to follow would help?) or to
continue to alter the default schema (hard for some folks, would be
overwritten by a weewx update), we didn't really solve all of the
problem....but certainly making the default schema bigger would help
greater-than-zero for sure.

Again, my admittedly easy-when-you-don't-have-to-do-it vision includes
Matthews plus:

   - do so in a componetized way that can be easily added to by users with
   custom needs
   - and have their customizations (additions) not altered or invalidated
   by a weewx upgrade

But if we don't get those last two, any kind of reasonably thought out
extended schema will definitely scratch some folks' itches and be goodness
if it can be done without breaking the as-is.

matthewwall, 2019-10-10:
i think that tom's xtypes work will make the extensibility easier.  but we should decide on the minimal number of fields in an extended schema that result in maximal improvement to out-of-the-box experience.  by that i mean "weewx and the seasons skin just works" - no need to extend the database to make it work with most hardware and 'normal' hardware configurations.  so 'normal' will mean things like "up to the number of T/H sensors that the station hardware supports" and "all of the observations that the hardware supports".

just to be explicit about some of the use cases i have in mind:

use case 1: make seasons skin support N extra T/H sensors out of the box.  the te923 supports up to 5, the vantage supports up to 8, klimalogg supports up to 8, fine offset hp3000 supports up to 8.  so lets say that N=8.  each one has one or more of battery level, signal strength, dewpoint.

use case 2: extension author for solar/battery driver does not want to reinvent the wheel.  if weewx ships with a `bin/schema/power.py` schema, then the extension can use standard field names and units.  any skin that uses those field names will work with charge controller from any vendor.

use case 3: user wants to add snow depth, UV, luminosity, particulate sensors.  if these are added as a service, then user will want schema with these fields.  if these are done with individual drivers, then they can be standalone databases.

use case 4: per-circuit power monitoring.  this one requires a schema defined by the driver author.  schema is included as a separate file in `bin/user` or a definition in the driver py file itself.

use case 5: gw1000 and other recent fine offset devices.  the latest fine offset systems are perhaps more extensible than davis.  in addition to the standard weather fields, they may also have UV, lux, particulates, soil moisture.

use case 6: netatmo.  the netatmo has co2 and noise data.  these are reported in loop data, but are not saved to database since there are no fields for them.

use case 7: tide sensor.  the tide database needs to track tides at the sensor location as well as predicted values for the sensor location and nearby locations.  this is a custom database schema defined by the driver (in this case the maxbotix driver, using a separate file to define the schema)

btw, the case sensitivity is for postgres and for mysql (when used on a platform with case-sensitive filesystem, i.e., macos in some configurations).  i know i should just give up on this, since the camelCase basically screwed up our hopes for full cross-platform multi-database support, and the inconsistency (heatindex versus extraHumid1) basically screwed up our hopes for consistent naming to make things easier for users.

matthewwall, 2019-10-10:
in some ways i wish weewx used a less rigid local storage like influx - then you don't worry nearly so much about the schema

vinceskahan, 2019-10-10:
Agree. I have a nice grafana dashboard overlaying influxdb data from weewx
weatherflow and multiple Arduino ds18b20 rigs and weewx vp2 data. Super
flexible.

Let me get you the current purpleair json elements, the extension is a
little behind what the gear puts out natively now. If you could add a few
aqi elements that would be helpful. The hardware reports it now, so you no
longer need to calculate it in the driver.

tkeffer, 2019-10-10:
> in some ways i wish weewx used a less rigid local storage like influx - then you don't worry nearly so much about the schema

We could switch to a 6th normal form schema that has the sensor name as a column:

| *dateTime*     | *sensor_type*  | *usUnits* | *interval* | *value* |
|:----------------|:--------|:-------|:---|:---------------------------------|
| 1570712892      | outTemp  | 1 | 5 | 55.2|
| 1570712892      | outHumidity  | 1 | 5 | 44|

I experimented with this once and found it super slow.

It would also require a major rewrite.

vinceskahan, 2019-10-10:
Update on the purpleair extension...re: the extended schema, it would be
helpful if you could add one element =>  "aqi2_5" please.

https://github.com/bakerkj/weewx-purpleair has what he currently saves in a
separate db, which is a small subset of what the JSON provides.  I can get
you the full list if it matters.

note: I have 'not' talked with Ken recently although we did talk some weeks
ago about aqi calculations and that they're done by the hardware now, so
most of the heavy lifting he currently does seems no longer required for
most folks who don't need the underlying source data used to calculate
AQI.  There is also a weewx-aqi extension that calculates a lot of
different variants of AQI based on source data it can get.  Presumably
anybody who needs that kind of fancy calculation would need weewx-aqi and
the current standalone weewx-purpleair extensions linked together
appropriately per their docs.

But if you add this one element, for the usual case a very simple extension
could snarf the JSON, pick off the desired elements, average them, and save
them in the new field in the db.  Much lighter lift.

matthewwall, 2020-01-13:
> In any case, can someone else push this to completion? I have my hands full with the 
> Python 2 to 3 transition, as well as extensible types.

i think we might want to include this as part of the v4 release.  there is a lot of hardware now that will support more than just extraTemp1-extraTemp3.

on 08oct2019 tom proposed a short list of additional fields.  i would suggest that we go with tom's suggestion, plus a full suite of T/H sensors

at some point we should release an 'official' air-quality schema, and an 'official' power schema.  but those should probably be done as extensions so that we don't slow down the release of weewx4.

full complement of T/H sensor parameters:
```
extraTemp4
extraTemp5
extraTemp6
extraTemp7
extraTemp8
extraHumid3
extraHumid4
extraHumid5
extraHumid6
extraHumid7
extraHumid8
batteryStatus1
batteryStatus2
batteryStatus3
batteryStatus4
batteryStatus5
batteryStatus6
batteryStatus7
batteryStatus8
```

these are the fields proposed by tom:
```
uvBatteryStatus
cloudbase
windrun
dewpoint1
heatindex1
humidex
humidex1
inDewpoint
snowBatteryStatus
snowDepth
snowMoisture
snowRate
lightning_distance
lightning_disturber_count
lightning_energy
lightning_noise_count
lightning_strike_count
luminosity
maxSolarRad
noise
co
co2
nh3
no2
so2
o3
pb
pm1_0
pm2_5
pm10_0
signal1
signal2
signal3
signal4
signal5
signal6
signal7
signal8
```

note that WU now officially supports these air-quality fields:
```
AqNO
AqNO2T
AqNO2
AqNO2Y
AqNOX
AqNOY
AqNO3
AqSO4
AqSO2
AqSO2T
AqCO
AqCOT
AqEC
AqOC
AqBC
AqUV-AETH
AqPM2.5
AqPM10
AqOZONE
```

matthewwall, 2020-01-13:
wview_extended schema has been updated as a candidate for consideration to release in v4.

one way to do it would be to put both 'wview.py' and 'wview_extended.py' in the 'schema' directory, but make 'wview_extended' the default schema.

vinceskahan, 2020-01-13:
For users who are upgrading (either way), would they have any action to
take to catch up to a modified schema file superseding their legacy
existing installation?

matthewwall, 2020-01-13:
since the schema file is used only when creating a new database, any existing installations would be unaffected.  users would only notice when they create a new database.

here are the scenarios:

* users who migrate from wview.  weewx 'just works' with the old wview database.
* users who have wview schema and upgrade to v4.  weewx 'just works' - existing database is untouched, and weewx continues to work with it.
* users who create a new database.  the new database will have extended schema.  existing sensor_maps and other elements will continue to work, since new schema uses old names and only adds new names (no name changes)
* users who have customized wview schema.  weewx 'just works' - existing database is untouched, and weewx continues to work with it.

i suppose the only surprise would be for someone who has a customized wview schema, with the customized fields matching one or more of the extended schema field names.  then there might be a clash.  on second thought, even in this case i think the user will be unaffected, since the user would have extended schema.wview (which still exists) and any extensions would have been explicit (e.g., in user/extensions.py), at least if they followed the guide.

tkeffer, 2020-01-13:
I'm nervous about trying to cram in a new feature so late in the release schedule, but I guess this is a benign one.

Note that while V4 supports the old style, it introduces a new style of specifying schemas where the day summaries are listed explicitly.

vinceskahan, 2020-01-13:
If a site chose to upgrade to the extended schema to start using the
additional fields, is it as simple as "wee_database --reconfigure" to catch
up after doing their v4 upgrade ?

tkeffer, 2020-01-13:
I've implemented the `wview_extended` schema, and updated the test suites to follow. This results in 114 types up from 52. Unfortunately, this also results in a huge increase in the size of the database. The old test sqlite database was 468kB, now it's 5.4MB. I don't know how general that is.

Commit https://github.com/weewx/weewx/commit/ad3546ad2baaae2fb018e75432d3cec575dbf965

closed 2020-01-13 by tkeffer

tkeffer, 2020-01-13:
Oops. I was not comparing apples to apples. The actual size increase was from 3.2MB to 5.4MB. Still a significant increase (68%), but not as bad as I'd feared.

chaunceygardiner, 2020-01-14:
> Commit ad3546a

Hi Tom,

I’m not sure if you are open to changing any of this, but, for air quality, you might want to support both CF_1 (CF=1) and ATM (CF=ATM) concentrations.  It is my understanding that, typically, sensors measure and report both.

Something like: 
    ('pm1_0_cf_1','REAL'),
    ('pm1_0_atm','REAL'),
    ('pm2_5_cf_1','REAL'),
    ('pm2_5_atm','REAL'),
    ('pm10_0_cf_1','REAL'),
    ('pm10_0_atm','REAL'),

The air quality index (AQI) is derived from pm2_5_cf_1 (in the USA).  I’ve been saving the AQI directly as:
    ('pm2_5_aqi', 'INTEGER'),

I’m not sure if that should be added; but it might not be straight forward to derive it.  Also, it varies by country.  As such, if you do add it, perhaps another field is needed to specific the type of AQI (e.g., USA, CAN)

Lastly, perhaps there should be room for multiple sensors (e.g., the outdoor PurpleAir sensor has two).

If you leave as is, perhaps a field to specify CF_1 or ATM concentrations would be good.

Just some thoughts.  I can continue to save in a separate database.

Cheers,
John

tkeffer, 2020-01-14:
The extended schema is already borderline obese. I'd actually prefer to eliminate some types.

We should follow @matthewwall suggestion of specialized schemas, including one for air quality. They would then have their own binding (*e.g.*, `aq_binding`), making them easy to use in reports.

chaunceygardiner, 2020-01-14:
No problem.  I already have a separate binding for air quality and use it in reports.