weewx/weewx #124: consider renaming of labels in skin.conf

matthewwall, 2016-05-16:
in skin.conf, there is a [[Generic]] section within [Labels].  Generic contains labels for observations.  i propose renaming 'Generic' to 'Observations' or perhaps even 'Schema'.

```
[Labels]
    some_key = a string that goes with the key
    latlon_formats = "xxx", "xxx", "xxx"
    [[Observations]]
        inTemp = Inside Temperature
        outTemp = Temperature
```

what is driving this?  i want to add a Labels search list in cheetahgenerator.py, analogous to the Extras and Almanac that are already there.  that would make all of the labels directly available to skins.  that in turn means that we do not have to duplicate strings.  right now one must put them in [Labels] so that the image generator picks them up, then do it again in the actual .tmpl files.  not good.

this will also facilitate translations.

for backward-compatibility we make the image generator look for [[Observations]] first, then fallback to [[Generic]]

matthewwall, 2016-05-16:
the [[Generic]] section is currently used in two places.  imagegenerator uses it to determine the label for each observation, and units uses it for the $obs.label.XXX search list.  in the context of these two, it is more appropriately called 'Observations' than 'Generic'.

as for getting to other entries in the [Labels] section, that is more of a 'what is the best practice' question.  it is already possible to put arbitrary labels in extras, for example:

```
[Extras]
    wind_avg = Average Wind Speed
    [[titles]]
        day = Current Conditions
        week = Week View
```

i guess i'm wondering of we want to have users (or skin designers) put all of their non-observation labels in Extras, or do we want [Labels] to be accessible so that they put non-observation labels right in the [Labels] stanza.  or perhaps something else?

tkeffer, 2016-05-22:
My original thinking was that `[[Generic]]` would be a very, well... generic resource that could be used for something other than observation types. Basically, it's a dictionary between any (arbitrary) key and a text label. The image generator uses this facility as a way of mapping an observation type to a label.

It sounds like you're proposing a more proper categorization of the "generic" labels into observation types and, perhaps, some other things. 

I admit that there is an overlap with `[Extras]`. I think that happened because of their different histories.

matthewwall, 2016-05-24:
i think i'm circling in on a couple of small adjustments:

1) formalize the [Labels][[Observations]] section
2) encourage extension writers to either use [Extras] (for a few items) or their own stanza (for skins that have many labels and/or parameters)

this is what it would look like:

```
[Extras]
    [[titles]]
        current_title = Current Conditions
        daily_title = Summary of Day
        ...
[Forecast]
    [[Labels]]
        zambretti_title = Zambretti Forecast
        nws_title = National Weather Service
        ...
[Units]
    [[Labels]]
        centibar          = " cb"
        cm                = " cm"
        ...
[Labels]
    hemispheres = N, S, E, W
    latlon_formats = "%02d", "%03d", "%05.2f"
    [[Observations]]
        outTemp = Outside Temperature
        ...
[Almanac]
    moon_phases = New, Waxing crescent, First quarter, Waxing gibbous, Full, Waning gibbous, Last quarter, Waning crescent
```

what am i trying to do here?
1) establish 'best practices' for where to put things in skin.conf, specifically for new reports
2) eliminate duplication across skins.  skin.conf should not have to contain entries for all of the default values.  for example, skin.conf should work just fine with no [Units] section - it would get the defaults.  this means we need to document what the defaults are.
3) explore patterns for translations and localization, i.e., make it easier by putting all strings in skin.conf instead of hard-coding into each .tmpl file.  this won't work for every report, but it should be possible for simple reports.

tkeffer, 2016-10-04:
I'm very concerned about backwards compatibility. In fact, there is a matrix of four possibilities we would have to worry about: 

1. old template with old skin.conf
2. new template with old skin.conf
3. old template with new skin.conf
4. new template with new skin.conf

We can always say, "Don't do that" and require users adhere to only combos 1 and 4, which simplifies things. The report generators would have to be modified (perhaps using a version number encoded in `skin.conf`) to figure out which is active.

Alternatively, we translate old skin.conf files into a new style, preferably on the fly (there are too many skin.confs out there to do it statically). 

Frankly, this seems like a lot of work, a lot of maintenance, for too little value. I'd rather just add new sections to skin.conf on an as-needed basis.

matthewwall, 2016-10-04:
how about this:

1) we change [Labels][[Generic]] to [Labels][[Observations]] in the Standard skin.conf.  make the imagegenerator look for Generic first, then fall back to Observations if there is no Generic.  deprecate the use of Generic.  (any label in [Labels] is generic)

2) we make $Labels.x accessible to any template via the default search list.  that would mean a template could do $Labels.Observations.barometer to get the text for barometer.  if the skin.conf for a report contains:

```
[Labels]
    current_title = Current Conditions
```

then $Labels.current_title would return 'Current Conditions'

3) values for the wview schema would be fully populated in english for [Labels][[Observations]], so a new skin would not have to specify any of them.  same goes for Labels.hemispheres and Labels.latlon_formats.  the defaults would be documented in the customization guide so that anyone doing a translation would know the full set of defaults that need to be translated.  (at some point there should be a utility that parses all the templates in a report and spits out a string list of labels that need to be translated)

4) make unit labels accessible too.  a template should be able to do $Units.Labels.inch to get " in" (the leading space is an anachronism we can thank wview for, but oh well).  $Labels.Units.inch would be better from the pov of someone writing/modifying a template, but the indirection for doing that might be too confusing.

i'm working on some dynamic reports, and right now there is *way* too much duplication and not enough parameterization of the labels for units and observations.

matthewwall, 2016-11-27:
perhaps this belongs in a separate issue, but here goes anyway: i18n adds another dimension to this problem.  from the i18n point of view, all of the translatable strings should be in a single location.  currently there are translatable strings in various skin.conf stanzas (Units.Labels, Units.Ordinates, Labels, Labels.Generic, ImageGenerator*label) as well as individual .tmpl files.  that makes translating tedious and error-prone.

one approach is to push all strings into stanzas in skin.conf.  so any translatable string in a .tmpl file is referenced by variable, e.g., $obs.label.rain or $Extras.titles.current_conditions.  translating a report would then be a matter of having one skin.conf for each language.

another approach is to gettext the strings.  any translatable string in skin.conf or .tmpl files would be wrapped in an i18n indicator, e.g., _("xxx") or %(xxx)s.  then an i18n step would be added whenever a tagged string is requested.  strings would be saved to a separate file for each translation, e.g., lang/en_US or lang/en_GB

ideally one could emit a single report in three languages like this:

```
[StdReport]
    ...
    [[StandardReport]]
        skin = Standard
    [[StandardReport-french]]
        skin = Standard
        lang = fr_FR
    [[StandardReport-german]]
        skin = Standard
        lang = de_DE
```

more generally, we might want locale = xxx instead of lang = xx_xx.

this is all in the context of the new standard skin issue #114

tkeffer, 2016-11-27:
I first considered using gettexxt when writing the skin algorithms, but chose not to, and, instead, rely on straight translations of the templates The reason is that, having had some first-hand experience writing C++ libraries to be used in I18N applications, I came to hate gettext. 

The biggest problem is that the same word or phrase may have to be translated different ways, depending on context. So, you need to either artificially add something to the lookup name so you know the context, or rely on an explicit context argument. Either way, it's easy to make a mistake and you end up with some very complicated looking gettext calls.

It's also worth noting that gettext is intended for writing I18N *code*, not templates, where you can't really duplicate everything when porting to a new language. 

Finally, with a gettext approach, you have *three* things that have to be kept in harmony: `skin.conf`, the templates, and all the different translation tables. Given that we're writing templates, and not code, I concluded it's simplest just to merge the last two items and resolve to translate the templates when localizing.

tkeffer, 2016-11-30:
Matthew and I talked this one out and concluded there is too much complexity for minimal benefit. Closing.

closed 2016-11-30 by tkeffer