weewx/weewx #126: Upload to Windguru

vhornik, 2016-05-30:
Hi!
Vaclav from Windguru here. One user asked me how he could upload data from weewx to Windguru, I don't have any experience with weewx, but would like to help him and offer uploads to WG. Is there some developer who could write the extension for Windguru upload? Basically the procedure would be that first the user would register his station here: https://stations.windguru.cz/register.php to make sure he has a unique ID and then use his ID in weewx config, weewx would send data using GET requests to a script on windguru site. It's easy to implement, I would provide details about how the GET request should look like. We already support meteobridge and WOSPi and weewx would be nice too. If you are interested please contact me (maybe better by mail to vh@windguru.cz)
Thank you and best regards!

Vaclav
www.windguru.cz

davies-barnard, 2016-05-30:
> <javascript:_e(%7B%7D,'cvml','vh@windguru.cz');>)

Hi Václav,

It's Chris here from the Wordpress/Joomla plugins.    I'm happy to help and
have some experience of working with Weewx.

Give me a shout on weewx@davies-Barnard.co.uk

Regards

Chris

matthewwall, 2016-06-12:
i added a link in the weewx wiki to claud's windguru extension for weewx

https://github.com/weewx/weewx/wiki

closed 2016-06-12 by matthewwall