weewx/weewx #127: weewx french standard skin

yannick56, 2016-06-01:
Ticket to upload the skin archive for weewx french standard skin and add the archive URL in the weewx's wiki.

yannick56, 2016-06-01:
[Standard-FR-v1.0.tar.gz](https://github.com/weewx/weewx/files/293536/Standard-FR-v1.0.tar.gz)

tkeffer, 2016-06-01:
This is something you can easily do yourself. 

We can host the tar file on weewx.com, but I'm hoping you're willing to maintain it as releases come out.

yannick56, 2016-06-01:
Hello Tom,

>This is something you can easily do yourself.
Yes the wiki web page is available here https://github.com/weewx/weewx/wiki/weewx-french-standard-skin and I have added a link to it in the main wiki page (skin part) : https://github.com/weewx/weewx/wiki

>We can host the tar file on weewx.com, 
OK
>but I'm hoping you're willing to maintain it as releases come out.
Yes.
I have uploaded Weewx French Standard skin in weewx wiki because I found several weewx french web sites with a french translation but no link or URL to download the weewx skin files with this translation.
And I try also the beautiful french skin from http://meteosaintsulpice.free.fr/SaintSulpice.tgz but it does not work with weewx 3.5.0 !
I hope that this translation will be useful for french users and help weewx to have more success.

Thank you for developing of the weewx software.

yannick56, 2016-06-01:
weewx french screenshot:
[image]

tkeffer, 2016-06-02:
great start.

A few comments:
1. It should not be put in a subdirectory with the name `Standard`. This is the name for the, well, Standard skin. Your original name, `Standard-FR` is much better.
2. It is not necessary to change the database units. They have nothing to do with the units displayed by the skin. Indeed, if the user changes database units, they will end up with a database with two unit systems, which weewx does not support.
3. If you follow comment 1 above, then it will not be necessary to do your "Part 2" step (where you have the user edit `weewx.conf`). You can put the correct target units directly in your french `skin.conf` configuration file.

Thanks so much for agreeing to maintain this. We try very, very hard to keep backwards compatibility with skins, so it should not be much work.

yannick56, 2016-06-02:
Hello Tom,

>1. It should not be put in a subdirectory with the name Standard. This is the name for the, well, >Standard skin. Your original name, Standard-FR is much better.

If I have a Standard-FR folder, with french translation, in /etc/weewx/skins/ , then I need to edit  /etc/weewx/weewx.conf file and  in [[StandardReport]] part
I need to replace the line
        skin = Standard
by
        skin = Standard-FR
Is the good choice/solution for you ?

>2. It is not necessary to change the database units. They have nothing to do with the units >displayed by the skin. Indeed, if the user changes database units, they will end up with a database >with two unit systems, which weewx does not support.

There is no need to change the parameter for "target_unit =" in the /etc/weewx/weewx.conf file, is it what you want to say to me ?
 
>3. If you follow comment 1 above, then it will not be necessary to do your "Part 2" step (where you >have the user edit weewx.conf). You can put the correct target units directly in your french skin.conf >configuration file.

I don't understand what you want to say to me. It's not clear for me.
Do you have an exemple ?
  
>Thanks so much for agreeing to maintain this. 

Yes, and if I have not the time, a new contributor can get my archive, update the file and upload a new release in the weewx wiki. 

>We try very, very hard to keep backwards compatibility with skins, so it should not be much work. 

Thank you

tkeffer, 2016-06-02:
> If I have a Standard-FR folder, with french translation, in /etc/weewx/skins/ , then I need to edit /etc/weewx/weewx.conf file and in [[StandardReport]] part
> I need to replace the line
> skin = Standard
> by
> skin = Standard-FR
> Is the good choice/solution for you ?

Yes. Your install program can do this. Look at some of the other extensions to see how this is done.

> There is no need to change the parameter for "target_unit =" in the /etc/weewx/weewx.conf file, is it what you want to say to me ?

Yes. The units used in the database have nothing to do with the display units. It is best not to change the database units.

> I don't understand what you want to say to me. It's not clear for me.
> Do you have an exemple ?

There are two ways of changing the display units. You can specify then directly in `skin.conf` (as explained in the section [Changing options in skin.conf](http://weewx.com/docs/customizing.htm#_________Changing_options_in_skin.conf_____)), or you can "override" in `weewx.conf` (as explained in the section [Overriding options](http://weewx.com/docs/customizing.htm#Overriding_options_in_skin.conf_from_weewx.conf)). Each has its advantage.

The default install uses the override method. 

I am suggesting that because a French reader is likely to always want metric units, you might as well just specify metric units in `skin.conf` and not do the override. But, either way will work.

yannick56, 2016-06-28:
[Standard-FR-v1.0.1.tar.gz](https://github.com/weewx/weewx/files/338169/Standard-FR-v1.0.1.tar.gz)
changelog since 1.0:
* translate the following missing text strings in french :  label = Rain (daily total), label = Low Temperature, label = High
* add pictures/graphs for humidity inside and outside (need for farmers)
* add url for radar clouds with meteo60.fr because picture with departements borders/limits
* translate text in french when pictures are missing in web page

closed 2016-07-20 by tkeffer

yannick56, 2018-09-23:
[weeWX_v3.8.2-Standard-FR.tar.gz](https://github.com/weewx/weewx/files/2409004/weeWX_v3.8.2-Standard-FR.tar.gz)
Post to upload the skin archive for weewx 3.8.2 french standard skin and add the archive URL in the weewx's wiki.