weewx/weewx #128: rtfreq should not be fixed at 2.5

matthewwall, 2016-06-06:
rick pointed out that the rtfreq in the wunderground restful service is hard-coded to 2.5:

https://groups.google.com/forum/#!topic/weewx-user/DeTlOPVgCNc

although 2.5 is probably close enough for vantage stations, it is not correct for many others.

how do we fix this?

i'm pretty sure we don't want rtfreq to show up in any driver.  like many other wunderground issues (e.g., dealing with partial packets), the fix for this probably belongs in the wunderground restful service.

one option is to make the wunderground uploader remember the timestamp of the last rapidfire upload.  when another rapidfire packet comes in, set rtfreq to the difference between the current and last.

we'll quickly see whether wunderground actually does anything with rtfreq, and whether they can deal with non-constant values...

matthewwall, 2016-06-06:
btw, this is rick's suggestion:

> So, feature request, in order of convenience for the user:
> derive the actual rtfreq by timestamping packets
> allow a configurable rtfreq entry in the wunderground section of weewx.config
> only set the 'rtfreq' record if it is unset, allowing driver authors to set it in the driver

tkeffer, 2016-06-08:
It's probably not a good idea to leave `rtfreq` hard coded, but before we start patching code, we should know what it does.

Does it represent an aspiration on the part of the hardware?

Or, an average frequency of packets?

And, what does WU do if posts come in more frequently than `rtfreq`? Less frequently?

Let's be sure we understand what problem we are trying to solve before "fixing" this.

whorfin, 2016-06-11:
As I mentioned in the group posting [but I guess not with the checkin, sorry] rtfreq appears to only be used by the Flash-based "View full-screen RapidFire" window on the wunderground site.
It is also only used once per load of the Flash code, ie "Frame 1".
To elaborate, the RapidFire window will refresh every "the first value of rtfreq seen" seconds.
Subsequent values of rtfreq, if changed by the station, are passed along by wunderground for each
packet, but ignored by the Flash code.

Rather obviously to those of us with stations which refresh less frequenty than 2.5 seconds,
wunderground does nothing in particular if posts come in less frequently than rtfreq.
It appears to be a simple attempt to guide the Flash window refresh without bothering to compute anything server-side.

The regular non-Flash javascript dashboard appears not to use rtfreq.
The iOS mobile app appears to behave similarly to the javascript dashboard, though it gets the station update timestamp grossly wrong if you ask it.

So it's likely rather legacy-ish, but I contend that the behavior I'm proposing here is greatly superior to blindly using 2.5 "because that's what the example was".

Using a computed average results in the RapidFire window [in flash mode] behaving reasonably.

Cheers

tkeffer, 2016-06-11:
Wow! That's some careful sleuthing. Thanks for digging into this.

Forgive me, but perhaps you could walk me through the advantages of a computed refresh rate (which gets subsequently ignored). Because Flash would refresh less often? Or, something else? What's the advantage to the user if Flash refreshes at something closer to the actual data refresh rate?

whorfin, 2016-06-11:
The advantage of a more correct refresh rate is the improved likelihood of a refresh event [which is indicated in the window] corresponding to new data, and less network access for those who might wish to leave the weather pane open for long periods of time.

Apart from that, anyone attempting to write their own realtime clients might appreciate accurate data.

tkeffer, 2016-06-12:
I'd rather see a config file option, especially as the WU doesn't even use subsequent values. This would make the refresh more tunable, rather than just blindly sticking with the first calculated value, accurate or not. Plus, it is much simpler.

Also, these days, someone writing their own real-time client is more likely to use websockets than a timed polling scheme. I'm still not exactly sure what problem we are trying to solve, but whatever it is, it is definitely WU specific and, it appears, flash specific. As you say, `rtfreq` is legacy, and unlikely to be used by anything else.

whorfin, 2016-06-12:
The refresh is not the first calculated value, it is the first value at the time of launch of the Flash plugin in a given user's browser.  Each launch of the Rapid Fire window will use the value active at time of launch.
Yes the problem is WU specific, which is why I only touched the RapidFire subset of the WU module.

I could live with a config file option too, but I implemented, and gave to you, the version I believed would be the easiest for users.  With a config file option, a user will need to either time over some period, or hope that someone has already done so, and consult a table somewhere, unique to their combination of monitoring station hardware, including all intervening bridges and drivers.
Common wisdom is that the AcuLink Bridge is 18 seconds, for instance, but with no further work
but firing up the driver, my system determines something closer to 15.5 seconds, and the thing just works.

I was asked by Matthew to consider contributing a Pull request, so I did.
If you insist on a hard-coded config option, I'd still like to see my "just works" method also supported, as an option.

Cheers

matthewwall, 2016-06-12:
from a usability point of view, a config option is less than ideal.  the loop interval on some hardware may be constant, but it varies for many stations.  and for some hardware, the interval varies by design.  for example, the ws23xx changes depending on whether the sensors are wired or wireless, and depending on how long since the last sensor update.

so requiring a user to specify the interval is error-prone.

imho, doing an automatic rtfreq is the 'correct' thing to do.  i'd leave the rtfreq_sma_window out of the default weewx.conf, but document it in the wu section of the user guide (probably just call it rtfreq_window).

this is yet one more thing that makes weewx less 'wee'...  however, i do not see it as bloat as long it is contained in the wu class.  there are already a few fixes we've had to make that are wu-specific.

tkeffer, 2016-06-12:
One more point and then I'll shut up.

I guess what's bothering me is that in the years that weewx has supported Rapidfire, we have not had one bug report complaining that "Rapidfire is updating too frequently and using up my bandwidth!" Or, "Rapidfire is not updating as fast as my instrument is capable." 

The evidence suggest that the present hardwired value, perhaps tunable with a config file option, works for everyone.

wotid, 2016-06-19:
I'm not sure if it makes any difference or not to the final solution that is implemented, I see the Aercus Weather Sleuth / Observer / Fine Offset WH2600 and probably others that are meant to upload to WU directly emit an rtfreq value in their loops packets (if I have the terminology correct)

i.e.

> Jun  4 02:29:45 rpi3 weewx[27522]: interceptor: MainThread: raw data: ID=stationid&PASSWORD=password&tempf=49.8&humidity=98&dewptf=49.3&windchillf=49.8&winddir=347&windspeedmph=0.00&windgustmph=0.00&rainin=0.00&dailyrainin=0.05&weeklyrainin=0.09&monthlyrainin=0.09&yearlyrainin=0.09&solarradiation=0.00&UV=0&indoortempf=63.5&indoorhumidity=62&baromin=30.20&lowbatt=0&dateutc=2016-6-3%2016:29:44&softwaretype=Weather%20logger%20V2.1.9&action=updateraw&realtime=1**&rtfreq=5**

tkeffer, 2016-06-19:
Thanks. That's a useful data point.

And, it suggests that the driver should be providing that number. If not provided, then it would default to something small (like 2.5).

tkeffer, 2016-06-26:
Anything new on this PR? 

As I said earlier, I'd rather see `rtfreq` be a tuneable parameter in the config file.

tkeffer, 2016-07-20:
No activity.

closed 2016-07-20 by tkeffer