weewx/weewx #131: WMR300 Driver 0.10 Pressure Issue

enzogps, 2016-06-17:
Hi,

I use Weewx 3.50 on Raspberry Pi connected to WMR300.

I just update WMR300 driver from 0.9 version to 0.10 to fix rain counter but now i loose pressure data as you could see on

https://www.wunderground.com/personal-weather-station/dashboard?ID=ILEDEFRA202

take the 16 of june, for sample we have
9.25 PM 1000.6 hPa
9.30 PM -- hPa
9.35 PM 1000.6 hPa
9.40 PM 1000.6 hPa
9.45 PM -- hPa
9.50 PM 1000.6 hPa
9.55 PM -- hPa
10.00 PM 1000.6 hPa

if i replace 0.10 version by 0.9 there is no more pressure problem (but rain problem of course)

i join a zip with syslog file with debug option between 21:23 to 22:00 of the 16 of june and weewx.db file.

Thanks
[Wmr300_Driver_Problem_Pression.zip](https://github.com/weewx/weewx/files/321117/Wmr300_Driver_Problem_Pression.zip)

tkeffer, 2016-07-20:
Matthew: you're going to have to take a look at this one. The changes you made going to V0.10 look benign to me.

tkeffer, 2016-07-21:
The problem is that you have an archive interval of 5 minutes, but your hardware does not always emit a pressure value within that 5 minutes. So, weewx records a null value.

Version 0.9 did some caching which got around this, but this was eliminated in V1.0. The feeling was that drivers should not "make up" data. The caching algorithm was effectively adding data which did not exist.

My suggestion would be to watch the loop packets and see how often they show a pressure value, then increase your archive interval to something longer than that interval. For example, if a pressure value is being emitted only once every 6 or 7 minutes, then you will need a 10 minute or longer interval. Otherwise, we're making up data.

closed 2016-07-21 by tkeffer