weewx/weewx #137: Debian install set HTML_ROOT incorrectly.

tkeffer, 2016-07-27:
This thread pretty much describes the problem: https://groups.google.com/forum/#!topic/weewx-user/kzXjvLjWI-k

tkeffer, 2016-07-27:
mwall suggests two possible solutions:

> a) modify HTML_ROOT in weewx.conf:
> 
> sudo sed -i -e 's%HTML_ROOT =.*%HTML_ROOT = /var/www/html/weewx%' /etc/weewx/weewx.conf
> 
> b) create a symlink to the weewx report directory:
> 
> sudo ln -s ../weewx /var/www/html
> 
> i tend to prefer the latter, especially for systems such as rpi where the weewx reports are on a tmpfs to minimize sd card wear.
> 

It seems odd to put the HTML files in `/var/www` where almost nobody will use them. Why not put them in the right place?

matthewwall, 2016-07-27:
if we want the weewx .deb to work only with debian 8 and later, then all we have to do is modify the pkg/debian/rules file so that HTML_ROOT is /var/www/html/weewx instead of /var/www/weewx

i would prefer to find a solution that makes the weewx .deb version agnostic - we don't want to have one .deb for debian 7, another for debian 8, etc.

it would be even better if the solution works with nginx and lighthttpd, not just apache.

i was thinking of making HTML_ROOT be /var/weewx, then make a symlink to /var/weewx in /var/www/html (for apache), and whatever locations are appropriate for nginx and lighthttpd.  but where do we put that symlink step?  is that a postinst thing?  with conditional behavior in case lighthttpd or nginx or apache is not installed?

tkeffer, 2016-07-27:
Well, you're the install expert!

One thing I'd be careful of: Apache doesn't always follow symbolic links. Depends on a configuration file. Not sure what the default is.

One advantage of having `HTML_ROOT` point to `/var/weewx` is that we can then just symlink everything. That is, have `/var/www/weewx`, `/var/www/html/weewx`, and whatever nginx and lighttpd expect all point to the same place. No postinst processing necessary.

"There is no problem in computer science that can't be solved without another level of indirection."

matthewwall, 2016-09-08:
let's leave the HTML_ROOT as /var/www/weewx, then use the apache conf to make that visible.  this makes the instructions consistent across installation methods.  anyone installing on deb7 or older won't have to do the conf configuration, but doing the conf will not hurt.  implemented at 857342e

see these for the reason debian changed document root:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=730372

https://lists.debian.org/debian-devel/2012/04/msg00301.html

for weewx v4 we might want to make this structure:

/var/weewx/
/var/weewx/reports/
/var/weewx/data/
/var/weewx/log/

then symlink (and conf) to those directories.  that way the weewx structure is consistent across all DEB and RPM installs, everything is in a single directory tree, and any other app (apache, nginx, lighthttpd) simply symlinks or confs to the weewx data.  it also makes it a bit easier to configure for low-write - just tmpfs the reports and log partitions.

closed 2016-09-08 by matthewwall