weewx/weewx #139: Inconsistency in the way pressure is (not) calculated

gjr80, 2016-07-29:
I spent most of yesterday tracking down some issues I noticed when I was using `WXCalculate` to add derived obs to a fairly basic archive record. The record included `barometer` (and `outTemp` and `outHumidity`) but did not include `pressure` and `altimeter`. I expected to see `pressure` and `altimeter` added, but only `altimeter` was, but yesterdays PR fixed that. What I did notice in tracking down that issue was that `altimeter` was being calculated (well actually set to `None`) but `pressure` was being left out altogether, as it turned out becasue no valid `interval` could be found.

I can appreciate that if some of the required (measured) obs are not present then a value of `None` is returned, but it seems odd to me that `pressure` could be left out completely, even though we have a `barometer` value in the record. I would have expected the 'consistent' result would have been to return a `pressure` of `None`. Certainly, this is what happens when the boot is on the other foot when calculating `barometer` and `altimeter`; if their pre-requisites are not available then both are set to `None`.

I appreciate that leaving `pressure` out of the archive record has the same net effect in terms of what is stored in the archive, but it seemed to be a little inconsistent to me. I know there has been a lot of commentary on 'pressures' in the past, so I am happy to leave it alone, but also happy to raise the PR if required.

```
    def calc_pressure(self, data, data_type):  # @UnusedVariable
        interval = self._get_archive_interval(data)
        if (interval is not None and 'barometer' in data and
            'outTemp' in data and 'outHumidity' in data):
            temperature_12h_ago = self._get_temperature_12h(
                data['dateTime'], interval)
            if (data['barometer'] is not None and
                data['outTemp'] is not None and
                data['outHumidity'] is not None and
                temperature_12h_ago is not None):
                data['pressure'] = weewx.uwxutils.uWxUtilsVP.SeaLevelToSensorPressure_12(
                    data['barometer'], self.altitude_ft,
                    data['outTemp'], temperature_12h_ago, data['outHumidity'])
            else:
                data['pressure'] = None
```

```
    def calc_barometer(self, data, data_type):  # @UnusedVariable
        if 'pressure' in data and 'outTemp' in data:
            data['barometer'] = weewx.wxformulas.sealevel_pressure_US(
                data['pressure'], self.altitude_ft, data['outTemp'])
        else:
            data['barometer'] = None
```

```
    def calc_altimeter(self, data, data_type):  # @UnusedVariable
        if 'pressure' in data:
            algo = self.algorithms.get('altimeter', 'aaNOAA')
            if not algo.startswith('aa'):
                algo = 'aa%s' % algo
            data['altimeter'] = weewx.wxformulas.altimeter_pressure_US(
                data['pressure'], self.altitude_ft, algorithm=algo)
        else:
            data['altimeter'] = None

```

tkeffer, 2016-08-20:
I assume you are referring to the `else` clause in `calc_pressure`? The fix is something like... ?

```python
    def calc_pressure(self, data, data_type):  # @UnusedVariable
        interval = self._get_archive_interval(data)
        data['pressure'] = None
        if (interval is not None and 'barometer' in data and
            'outTemp' in data and 'outHumidity' in data):
            temperature_12h_ago = self._get_temperature_12h(
                data['dateTime'], interval)
            if (data['barometer'] is not None and
                data['outTemp'] is not None and
                data['outHumidity'] is not None and
                temperature_12h_ago is not None):
                data['pressure'] = weewx.uwxutils.uWxUtilsVP.SeaLevelToSensorPressure_12(
                    data['barometer'], self.altitude_ft,
                    data['outTemp'], temperature_12h_ago, data['outHumidity'])
```

matthewwall, 2016-08-20:
i think there is a more fundamental question to answer first:  in which situations should pressure=None and in which situations should pressure not be set?

how about this:
- if barometer is not None, then pressure should be either a calculated value or None
- if no barometer and altimeter is not None, pressure should be None (we do not calculate pressure from altimeter)
- in all other cases, pressure is untouched

the question is really around the last point - should pressure be untouched, or should it be None?

gjr80, 2016-08-20:
> I assume you are referring to the `else` clause in `calc_pressure`? The fix is something like... ?

I guess my concern was if we are set to calculate a derived obs and some of the pre-requisites are missing should we set the derived obs to `None` or omit it completely. `pressure` seemed to take a different approach to other (weewx) calculated derived obs. The 'fix' makes `pressure` consistet with other weewx calculated derived obs.

> how about this:
>-     if barometer is not None, then pressure should be either a calculated value or None

Would appear to make sense.

>-     if no barometer and altimeter is not None, pressure should be None (we do not calculate pressure from altimeter)

Again would make sense

>-     in all other cases, pressure is untouched 
>
>the question is really around the last point - should pressure be untouched, or should it be None?

Agree this is the the issue. In cases like this I like to take up pen and paper and map out a truth table; `barometer` and `altimeter` are the variables and `pressure` the output. Though in this case we are dealing with a trinary system; each variable can be either non-existant, `None` or a value. I can summarise the my truth table as follows:

- if `barometer` is `None` then `pressure` is `None`
- if `barometer` is a value then `pressure` is calculated (though the result may be `None` or a value)
- if `barometer` does not exist and `altimeter` does not exist then `pressure` does not exist
- if `barometer` does not exist and `altimeter` is a value then `pressure` is `None`
- if `barometer` does not exist and `altimeter` is `None` then `pressure` is `None`

I guess we are still left to philosophise, in this case over the last 2 sitauations. I guess I have the view that if `altimeter` exists (albeit `None` or a value) then so to should `pressure` (albeit `None`).

Or we could just leave it as it is after adopting 'the fix'...

matthewwall, 2016-10-03:
i implemented tom's suggestion at commit 

barometer = f(pressure, outTemp)
altimeter = f(pressure, altitude)
pressure = f(barometer, outTemp, outHumidity, 12h_temp)
12h_temp = f(interval)

just to be clear, this is the intended outcome:

1) packet: no pressure, no barometer, no altimeter
result: pressure=None, barometer=None, altimeter=None

2) packet: no pressure, barometer=None, no altimeter
result: pressure=None, barometer=None, altimeter=None

3) packet: pressure=None
result: pressure=None, barometer=None, altimeter=None

4) packet: pressure=x
result: pressure=x, barometer=f(x), altimeter=f(x)

5) packet: barometer=x
result: pressure=f(x), barometer=x, altimeter=f(x)

6) packet: altimeter=x
result: pressure=None, barometer=None, altimeter=x

7) packet: pressure=x, barometer=y
result: pressure=x, barometer=y, altimeter=f(x)

sorry to be pedantic, but i had to enumerate test cases so i could evaluate how each type of hardware would behave.

the only caveat is the behavior if we want to do the not-exist option, i.e., if a dependent value does not exist, then StdWXCalculate does nothing.  it would change case (1) to look like this instead:

1A) packet: no pressure, no barometer, no altimeter
result: no pressure, no barometer, no altimeter

and doing that would change *every* calculation in StdWXCalculate.

tkeffer, 2016-10-04:
Can we close out this issue?

closed 2016-10-04 by matthewwall