weewx/weewx #141: Refactor documentation so utilities are described separately

tkeffer, 2016-08-23:
Right now, a user must read the entire *Customizing Guide* to find out what utilities there are and how to use them --- there's no separate section describing them.

The descriptions should be factored out and either put in a separate *Executables* document (preferred), or in an appendix to the *Customizing Guide*. While we're at it, the main weewx executable, `weewxd`, should also be documented here.

The descriptions in the *Customizing Guide* would then be replaced with links to the appropriate utility.

See also Issue #97.

matthewwall, 2016-09-18:
should utilities.htm be called apps.htm?

tkeffer, 2016-09-18:
The moniker "apps" seems to have been co-opted to mean "applications for mobile devices," so I'd say we should stick with `utilities.htm`.

closed 2016-09-20 by tkeffer