weewx/weewx #150: map conventions

matthewwall, 2016-09-11:
this one is for gary.  the wee_import utility has a section [Map] for csv imports with entries:

weewx_database_field = csv_label, weewx_units

what do you think about this instead:

csv_label = weewx_database_field, weewx_units

my reasoning for this is to make the import syntax consistent with mappings in drivers and mappings in uploaders.  the general principle is:

input_field = output_field

so for drivers we have:

label_used_in_hardware = weewx_database_field

and for uploaders we have:

weewx_database_field = field_used_by_upload_destination

or for uploaders with units we have:

[[[inputs]]]
    [[[[weewx_database_field]]]]
        name = field_used_by_upload_destination
        units = units_used_by_upload_destination

matthewwall, 2016-09-11:
now that i have written it down, i see that input_field = output_field seems backwards.  unfortunately, that is how all of the drivers (te923, ws28xx, klimalogg, netatmo, interceptor) and uploaders are implemented.

so do we leave the import syntax as it is, make it consistent with the drivers and uploaders, or change the drivers and uploaders?

gjr80, 2016-09-11:
Consistency would be good. output=input seemed to make sense to me. As for the 3 options, I think we can rule out the last one; I would have expected it would be too much of an impact on the users. I think leaving it as is would not be that big an issue; I don't know the overlap between users who will import and users who use mapped drivers/uploaders would be that great, and then chances are their import would be a one off event. That being said while you are both on the water let me see what I need to do to flip `wee_import` around.

matthewwall, 2016-09-12:
here is the issue in more detail.  examples are for mapping sensor inputs in a driver, but they apply just as well to mapping fields in wee_import and for uploaders as well.

option 1: single-line-per-mapping; comma-delimited for multi-parameter

```
[map]
    output_field1 = input_field1
    output_field2 = input_field2, units2
```

option 2: single-line and/or stanza; stanza for multi-parameter

```
# generic example
[map]
    output_field = input_field
    [[output_field]]
        sensor = input_field2
        units = units2
```

```
# driver example
[map]
    outBatteryStatus = battery.1:9.HidekiTS04Packet
    [[rain_total]]
        sensor = rain_total.0026.FOWH1080Packet
        data_type = gauge
    [[rain]]
        sensor = rain.0026.FOWH1080Packet
        data_type = counter
```

```
# import example
[map]
    [[outTemp]]
        field_name = temperature
        units = degree_C
    [[inTemp]]
        field_name = temperature_inside
        units = degree_C
```

option 3: existing drivers

```
[map]
    input_field = output_field
```

```
# driver example
[sensor_map]
    battery.1:9.HidekiTS04Packet = outBatteryStatus
```

reasons we should consider this:

- make explicit the one-to-many relationships that are possible for output=input.  the converse is not true (you cannot have one-to-many input=output.  this is important for things where you might want to map a single sensor input into multiple database fields or functions.

- the stanza approach lets you add additional information such as sensor type (gauge vs counter).  single-line does as well, but it is more opaque (but more compact).

reasons we should not do this:

- this will break the existing [sensor_map] in some drivers.  those drivers include te923, klimalogg, interceptor, ws28xx.  only an issue for those who have customized their mappings, and it is possible to programmatically convert (e.g., during upgrade).

i am inclined to go with option 2 for any new development (including wee_import), then convert everything else when we do weewx v4.

thoughts?

gjr80, 2016-09-12:
Well if you have a way around the impact on existing users that puts a different light on things. Single line may be more compact but I think the stanza approach would be more user friendly. I agree with option 2. I'll get onto wee_import.

gjr80, 2016-09-13:
I have option 2 working for a CSV import under `wee_import` (in fact it works out quite well; has simplified a bit of the code plus the exercise highlighted a couple of non-critical errors/omissions in the `csvimport.py` code). WU requires no change as the user does not specify any map. Cumulus requires the user to specify the temperature, pressure, speed and rainfall UoM as these can't be determined from the Cumulus monthly log file. The UoM are currently specified by the user in a `[[Units]]` stanza as follows:

```
[[Units]]
    temperature = degree_C      # options are 'degree_F' or 'degree_C'
    pressure    = hPa           # options are 'inHg', 'mbar' or 'hPa'
    rain        = mm            # options are 'inch' or 'mm'
    speed       = km_per_hour   # options are 'mile_per_hour',
                                # 'km_per_hour', 'knot' or
                                # 'meter_per_second'
```

Strictly speaking this is a map but I don't know there is much to be gained in trying to treat it the same as the drivers/uploaders and CSV map. It isn't mapping any fields just specifying UoM. I don't know I want to see something like:

```
[[Map]]
    [[[outTemp]]]
        units = degree_C
    [[[inTemp]]]
        units = degree_C
    [[[poolTemp]]]
        units = degree_C
```

and we haven't even touched the multiple wind speed and pressure fields. I am inclined to leave it as is or how far do we take the 'map' standard?

gjr80, 2016-09-17:
Any further thoughts on this, if 3.6.0 is getting close it would be good to have the final solution included from the start.

matthewwall, 2016-09-18:
gary, could you post an example of what the map looks like for CSV?  WU has no map (names and units from WU never change).  Cumulus has no map (names and units from Cumulus never change).

here is a summary of current state of drivers and uploaders.

DRIVERS

the te923 driver does this:

```
    [[map]]
        t_in = inTemp
        h_in = inHumidity
        t_1 = outTemp
        h_2 = outHumidity
        ...
```

the cc3000 driver does this:

```
    [[sensor_map]]
        TEMP IN = inTemp
        TEMP 1 = extraTemp1
        TEMP 2 = extraTemp2
```

the wmr300 driver does this:

```
    [[map]]
        temperature_0 = inTemp
        temperature_1 = outTemp
        temperature_7 = extraTemp2
        ...
```

the klimalogg driver does this:

```
    [[sensor_map]]
        Temp0 = inTemp
        Humidity0 = inHumidity
        ...
```

the meteostick driver does this:

```
    [[sensor_map]]
        in_temp = inTemp
        soil_temp_1 = soilTemp1
        ...
```

the sdr driver does this (option 2):

```
    [[sensor_map]]
        windDir = wind_dir.0BFA.Acurite5n1Packet
        windSpeed = wind_speed.0BFA.Acurite5n1Packet
        outTemp = temperature.0BFA.Acurite5n1Packet
        ...
```

the vantage driver does not have a mapping, but probably should.

UPLOADERS

the emoncms, influx, and mqtt uploaders do this:

```
    [[inputs]]
        [[[barometer]]]
            units = inHg
            name = baromer_inHg
            format = %0.3f
        [[[outTemp]]]
            units = degree_F
        ...
```

the thingspeak uploader does this:

```
    [[fields]]
        [[[field1]]]
            obs = barometer
            format = %0.3f
        [[[field2]]]
            obs = outTemp
            units = degree_F
        ...
```

the xively uploader does this:

```
    [[channels]]
        [[[inTemp]]]
            name = inside temperature
            units = degree_F
            format = %.1f
        [[[outTemp]]]
            name = temperature
            units = degree_C
        ...
```

gjr80, 2016-09-18:
>could you post an example of what the map looks like for CSV?

As currently committed CSV uses the following:

```
[CSV]
    [[Map]]
        dateTime    = csv_field_name1, unix_epoch
        barometer   = csv_field_name2, inHg
        outTemp     = csv_field_name3, degree_F
        inTemp      = csv_field_name4, degree_F
```
Based on your earlier comments I have a branch that instead uses:
```
[CSV]
    [[Map]]
        [[[dateTime]]]
            field_name = csv_field_name1
            units = unix_epoch
        [[[barometer]]]
            field_name = csv_field_name2
            units = inHg
        [[[outTemp]]]
            field_name = csv_field_name3
            units = degree_F
        [[[inTemp]]]
            field_name = csv_field_name4
            units = degree_F
```

>WU has no map (names and units from WU never change)

Agreed, sort of. The names and units actually do change but there is no need for a map as the column name includes the units and there are a finite (and fortunately small) number of name/units column names that are used. It appears that WU geolocates the requestor and returns units appropriate to the requestors location. For example, here in AS if I do a query on a WU PWS, whether the station be here is AS or anywhere overseas, I am returned a column `TemperatureC` that contains the outside temperature. Assuming you are in the US, if you do a query on any PWS in the world you will see a column `TemperatureF`. I have VPN'd to a point in the US and was returned `TemperatureF`. So for WU the names do change but you can glean the units from the column name and since there are only a small number of different columns it is easy just to have a fixed map covering all and you just use those that are returned in your query. No need for user input.

>Cumulus has no map (names and units from Cumulus never change)

Again, agreed, sort of. Cumulus names do not change but unlike WU they do not give us the units. The units that appear in a Cumulus monthly log file are dependent on what the Cumulus user set at the _Cumulus Station Configuration Screen_ as the units to use for temperature, pressure, rain and windspeed. Refer to this Cumulus wiki [entry](http://wiki.sandaysoft.com/a/Setup#Units_of_measure_panel) and this [screenshot](http://wiki.sandaysoft.com/a/Cumulus_Screenshots#Station). So it is true that for a given import the Cumulus names and (most likely) the units will not change, but chances are that if you and I were Cumulus users we would be using a different suite of units. So for Cumulus we do not need to specify the names just the units. That was the point of my 2nd last comment; it seems (at least visually) a little odd to do this:

```
[[Units]]
    [[[outTemp]]]
        units = degree_C
    [[[inTemp]]]
        units  = degree_C
    [[[barometer]]]
        units  = hPa
```
compared to
````
[[Units]]
    temperature = degree_C
    pressure = hPa
```

matthewwall, 2016-09-20:
for cumulus, i would go with:

```
[[Units]]
    temperature = degree_C
    pressure = hPa
```

assuming that:
- every database field name must have both a csv field and units, 
- commas are not permitted in field names, 
- field names will not have leading or trailing spaces, 
then i lean toward the more concise approach for csv:

```
[CSV]
    [[Map]]
        dateTime    = csv_field_name1, unix_epoch
        barometer   = csv_field_name2, inHg
        outTemp     = csv_field_name3, degree_F
        inTemp      = csv_field_name4, degree_F
```

since this brings us full circle, i hope this whole mapping discussion has not been a superfluous distraction.  i think it was needed to ensure consistency across different parts of weewx.

matthewwall, 2016-09-20:
you might consider using [[field_map]] instead of simply [[Map]]

tkeffer, 2016-09-20:
I don't like the idea of mixing single line specs with stanza specs. It should be one or the other.

How likely is it that a mapping will need more than (input, unit)? Matthew gave an earlier example with an additional parameter `data_type`. Is that a real example? Or, a theoretical one?

Assuming it's mostly theoretical, I prefer the compact style. It's what we're already using elsewhere in weewx and it's a lot less verbose.

Clear preference for `output = input`. I know that means that some drivers will have to be changed, but that should  be mostly our problem. For those that have customized drivers, they should just, well... re-customize. It's easier and probably less error prone than us trying to write a tool to change it automatically for the half-dozen people involved.

gjr80, 2016-09-20:
Ok, from this mornings comments I think I have enough to finish off wee_import (for 3.6 anyway). It seems that what is presently in master is almost there, just a few cosmetic changes. Tonight's job.

I don't see data_type coming into play for wee_import, I think it will be limited to input and units.

gjr80, 2016-09-20:
Now my wife tells me I am a pedant so I apologise upfront...

I know Matthew has a liking for lower case only in most places :) , but do we have any sort of convention for section names in config files? Matthew's driver/uploader examples above are all lowers, if I look in 
`weewx.conf` I see all uppers (where it is an abbrviation eg `[[WOW]]`) or camel case or initial capital.

So to cut to the chase, do we use `[[field_map]]`, `[[Field_map]]`, `[[FieldMap]]` or some other variant?

tkeffer, 2016-09-20:
I try to use CamelCase for larger "collective" things like Python class names and section names, snake_case for individual variables and options. So, it would be `[[FieldMap]]`.

I know I haven't always been very consistent with this.

matthewwall, 2016-09-30:
resolved for use in import.

uploaders are ok as they are for now, but probably should be converted at some point.

drivers definitely should be converted at some point.

closed 2016-09-30 by matthewwall