weewx/weewx #155: Inconsistent units used for wee_device --set-interval

tkeffer, 2016-09-18:
As gleaned from the documentation. The actual driver code might be different.

| *driver* | *units*|
|----------|------------|
| CC3000      | minutes|
|fineoffset | minutes |
|TE923|seconds|
|Vantage|seconds|
|WS23xx|minutes|
|WS28xx|minutes|

With few exceptions, weewx uses seconds throughout, so this is what I'd prefer. But, we ought to stick with a single time unit.

matthewwall, 2016-09-18:
i can fix this one.  however, to do so i would propose the following standard for `wee_device --set-interval`

use 's', 'm' and 'h' to indicate seconds, minutes, and hours.  if no character specified, then the value is in seconds.

```
# set to 5 minutes
--set-interval=300
```

```
# set to 5 minutes
--set-interval=5m
```

```
# set to 60 minutes
--set-interval=1h
```

the reason for this is that many stations do everything in minutes, not seconds.  so from the station point of view, it is less foreign to specify minutes.  also, some of the stations support distinct archive intervals, e.g., 30m, 1h, 2h, instead of e.g., 1-3600 seconds.

use m and h means users won't have to screw up the conversion to seconds.

tkeffer, 2016-09-18:
I dunno. If we do this, then we should also change other parameters, such as altitude (the Vantage requires that it be in feet).

Also, it would prevent `optparse` from doing integer validation.

I don't care about the station point of view. I care about the user point of view. Being consistent in units may make it harder for the station, but easier for the user.

I get your point about distinct archive intervals, but that doesn't go away by allowing different unit systems.

matthewwall, 2016-09-18:
i'm thinking of the user.  for example, the te923 valid intervals are 5m, 10m, 20m, 30m, 60m, 90m, 2h, 3h, 4h, 6h, 1d"

if --set-interval is only seconds, the user must convert to seconds before setting the interval.

so we have ease-for-user versus integer-validation-by-optparse.  the latter could be solved by a weeutil.parse_interval() utility, invoked by any wee_device stub.

tkeffer, 2016-09-18:
I see your point.

I guess I'm having an allergic reaction because this introduces yet another way of specifying units. The mapping algorithm threatens to introduce still another. 

Oh, and the abbreviations `m` and `d` introduce new unit specifiers.

tkeffer, 2016-09-20:
Just thought of another approach. What if we had:

| *option* |
|----------|
|`--set-interval-seconds`|
|`--set-interval-minutes`|
|`--set-interval-hours`|

They would be mutually exclusive, of course.

tkeffer, 2016-09-26:
Fixed in https://github.com/weewx/weewx/commit/2b96d101aa7a99111a217f9bc7d267330e9afae0

closed 2016-09-26 by tkeffer