weewx/weewx #164: wmr200 driver can never use extraTemp1, extraHumid1

mdecker, 2016-10-08:
Hi,

I'm currently working on adding additional sensors on my WMR200 and happened to notice that the sensor on channel 2 is assigned to extraTemp2 - even though I would have expected it to be on extraTemp1. A look at the source code (https://github.com/weewx/weewx/blob/master/bin/weewx/drivers/wmr200.py#L1033) shows that the way the extra fields are calculated uses a flawed logic that will never be able to assign values to extraTemp1 and extraHumid1. While the fix in itself is pretty easy, it comes with potentially broken backward compatibility.

I see two ways to address this:
1: calculate the field numer like the WMR100 driver does (`sensor_id -1`). This will potentially break existing setups, though! (What is currently extraTemp2 will become extraTemp1, etc.)
2: add a special case for `sensor_id == 4`  to assign it to extraTemp1. This will make it a little more confusing for starters, but it won't break existing setups. This deviation from expected procedures should thus be clearly documented. For `sensor_id >=5`  the `sensor_id -1` approach could be used (only relevant if you update your database schema).

matthewwall, 2016-10-08:
perhaps it is time to introduce the sensor_map pattern to the wmr drivers (wmr100, wmr200, wmr300, wmr9x8)

in the sensor_map pattern, the driver uses hardware-appropriate labels for the observations, then the sensor_map associates database schema with those labels.  for example, the wmr200 would have a default sensor_map that associates with an extended, wview-compatible schema:

```
[WMR200]
    [[sensor_map]]
        inTemp = temperature_0
        inHumidity = humidity_0
        outTemp = temperature_1
        outHumidity = humidity_1
        extraTemp1 = temperature_2
        extraHumid1 = humidity_2
        extraTemp2 = temperature_3
        extraHumid3 = humidity_3
        extraTemp3 = temperature_4
        extraHumid3 = humidity_4
```

so if you wanted to re-map the observations to a different schema, you simply specify a sensor_map in your weewx configuration file.

mdecker, 2016-10-09:
That sounds like a reasonable proposal to me.

matthewwall, 2016-10-11:
temporarily fixed at commit fc97406 using sensor_id -1

proper fix using the sensor_map pattern will come in a bigger weewx release when we do the mapping for all drivers.

closed 2016-10-11 by matthewwall