weewx/weewx #174: Consider a 'minimum required weewx version' check when installing an extension

gjr80, 2016-10-27:
As each weewx version introduces new capabilities and as these capabilities are used in weewx extensions, some extensions will have a minimum required weewx version if they are to operate error free. This check woudl logically be ether in the extension itself (checking and catching the error after install at runtime) or within `wee_extension` (checking and catching the error before the extension is installed).

The following email extracts capture discussion so far.

hi gary,
i took the lazy approach - the installer does no version checking.
however, each extension can do a version check.  one way to do it looks like this:

if weewx.__version__ < "3":
    raise weewx.UnsupportedFeature("weewx 3 is required, found %s" %
                                   weewx.__version__)

there is also a utility function require_weewx_version in bin/weewx/__init__.py that provides more fine-grained comparison.

i typically put the version check in the extension's primary .py file after the imports.

the end result is that anything will install, and compatibility is checked by the extension itself at runtime.  if there is incompatibility, the extension should fail fast and hard, but if there is partial compatibility then the extension can deal with it however it wants.

i am leery of putting version/compatibility checks in the extension installer due to the potentially huge combinatorial issues that could result.  take a look at the state diagrams for debian dpkg/apt.

but if you want to discuss some options that would make the user experience better, let's discuss.

m
On 26 Oct 2016, at 13:35, Gary wrote:
> Matthew,
>  
> Having a short weewx break and am touching up the weewx-wd extension so we can put out a new version. This version will require weewx 3.4.0 or later and as far as I can tell the extension installer does not have the ability to do this sort of check. I had a look through the extension installer code and had a few ideas as to how it can be done, but I am not overly familiar with code. Is this a good idea and if so what do you believe is the best approach? I am happy to do the coding just need a pointer in the right direction.
>  
> So much for the short weewx break.
>  
> Gary

On 26 Oct 2016, at 19:10, Gary wrote:

> What I had thought was
> that if the 'extension engine' had a handler to catch the such an 
> exception and gracefully fail/exit (with the nice informative message 
> rather than the stack trace)  that would be a better way to handle 
> things. It puts the onus on the extension writer/packager, requires 
> little in way of additional code and complexity in the 'extension 
> engine' and prevents the user installing something which will fail and result in support issues.

i was assuming that the only compatibility issues would be for services/drivers (extensions with code).  but you are right - there will be skins that require certain features/capabilities, so there should be a way to do version checks for them.  it makes sense to do that in the installer.

it could be as simple as another arg to the installer __init__, say 'requires'.  for example:

class ComputerMonitorInstaller(ExtensionInstaller):
    def __init__(self):
        super(ComputerMonitorInstaller, self).__init__(
            version="0.14",
            name="cmon",
            ...
            requires="3.6") # requires weewx 3.6 or later

if there is a 'requires' argument, then the weewx version must be equal to or greater than that version number.  if no 'requires' is specified, then no version comparison is made.

is there a use case for specifying a handler (i.e., code that the extension's install must run) instead of just a 'requires' version string?

i can imagine a comma-delimited argument to 'requires' so that one extension could require another extension.  but that is the dpkg/apt-like dependency rabbit hole that i'm trying to avoid.

note that this approach will only work for 3.7 and later.  using this approach with older weewx releases won't break them, they'll just fail at runtime instead of install time.

m=

gjr80, 2016-10-27:
I had considered another installer arg but had the impression you did not want to extend the extension engine too much. I agree it is a neater solution.

> is there a use case for specifying a handler (i.e., code that the extension's install must run) instead of just a 'requires' version string?

There probably is, but those extensions are probably few and far between. Would be easy enought to ad later if the need arises.

>i can imagine a comma-delimited argument to 'requires' so that one extension could require another extension. but that is the dpkg/apt-like dependency rabbit hole that i'm trying to avoid.

Starting to get a bit sophisticated now. I say just don't go down the rabbit hole or go down a rabbit hole that you have control of. If something like this is implemented you could get by with just identifying that one or more pre-requisites are missing, notify the user and gracefully exit. Then let the user fix it up, a bit less sophisticated than dpkg/apt.

>note that this approach will only work for 3.7 and later. using this approach with older weewx releases won't break them, they'll just fail at runtime instead of install time.

Yep, so it will be no worse than it is at the moment. The target audience is probably more the new users, they will probably be using the latest version and by having something that is a little more user friendly it might save a few support issues.

Are we at the stage where I can scratch something together with a PR for review?

matthewwall, 2016-11-02:
On 27 October 2016 10:04:12 EDT tkeffer wrote:

So long as we're having a discussion about installers, I have come to admire npm, the installer used by node. With npm, an application specifies the dependencies. For example, WeeRT, has the following dependencies:
"dependencies": {
  "argv": "0.0.2",
  "bcrypt": "^0.8.5",
  "body-parser": "^1.13.2",
  "debug": "^2.2.0",
  "express": "^4.13.3",
  "express-jwt": "^3.3.0",
  "hbs": "^4.0.0",
  "jsonwebtoken": "^5.5.4",
  "mongodb": "^2.0.40",
  "morgan": "^1.6.1",
  "normalize-url": "^1.3.1",
  "socket.io": "^1.3.6",
  "url": "^0.11.0"
},
These can have recursive dependencies. The up arrow ("^") means any version greater than, but you can also constrain versions in other ways.

So, to install an extension, you tell npm what you want. It then downloads the package, then annotates the dependency list to reflect the version it downloaded. By default, it uses "greater than" the version it downloaded, but the user can change that.

In my dream world, if a user wanted to install cmon, s/he would use wee_install to ask for cmon, which would then download the package, then annotate weewx.conf to reflect the version it found:

[Extensions]
  cmon = "^0.14"

I'm not sure if npm can handle the reverse dependency: what version of node (or weewx, in our case) does the extension require? I'd have to research that.

-tk

matthewwall, 2016-11-02:
so there are (at least) three parts to make this happen.

1) mechanism for specifying dependencies/requirements.  if we use a dict instead of a string for the requires/dependency arg, then we are future-proof for npn-style dependencies.  for example,

```
class ComputerMonitorInstaller(ExtensionInstaller):
    def init(self):
        super(ComputerMonitorInstaller, self).init(
            version="0.14",
            name="cmon",
            ...
            requires={
                weewx="^3.6"
            }) 
```

2) local registry of what has been installed.  the user/.installer/ directory could work as-is.  another option would be to keep the information in the weewx configuration file.  we probably want to review/formalize/experiment/document it to be certain.

3) mechanism to download required extensions.  this probably needs a central listing of which extensions are available, or perhaps a local listing of listings, analogous to the way apt sources work.  downloading is easy, and local caching/extraction is easy.

1 and 2 could be implemented before 3, and would provide significant value on their own.

for that matter, an implementation of 1 with only the weewx version check would provide the value that gary initially proposed.

PeterQuinn925, 2019-04-03:
Is there documentation somewhere that says what version of weewx is currently supported? I'm still running 3.5 (it wasn't broken) but with my forecast stuff breaking with the WU API key changes, I'm considering upgrading forecast.py or all of weewx if I have to.

matthewwall, 2019-04-03:
we typically 'support' only the latest release, but we do not discourage anyone from running older releases.  if it works, use it.  don't upgrade until you need a feature/bugfix in a later release.

we do everything we can to maintain backward compatibility - that is what the x.y.z versioning means.  3.7 is backward compatible to things written for 3.0.

the wu api stuff is completely independent of weewx versions/releases

the forecast extension works with any weewx release since 3.0

matthewwall, 2019-04-03:
btw, the next forecast release will include support for the new wu api, and it will work with any weewx release back to 3.0.  it *should* also work with the python3 stuff in weewx 4.0

PeterQuinn925, 2019-04-04:
Thanks. I installed the current forecast.py version and in the processes screwed somethings up. I switched it to NWS forecast and fixed my screwups and it's all good now.

When 4.0 comes out, I'll probably do an upgrade.

djenk41us, 2022-07-10:
Hello, I was looking to contribute to the WeeWX software. I was reviewing this issue and wondering if it is either stale or should be addressed. I am willing to address the issue and provide a solution.

Please let me know.

closed 2024-07-03 by github-actions[bot]