weewx/weewx #175: Consistent use of None

tkeffer, 2016-11-02:
Matthew made a proposal:

> 1. Driver should insert an observation name only if there is a known value for that observation. None is a valid known value, but only if the value is known to be undefined. for example, if a sensor is known to be reporting a bad value, then None is an appropriate value for that sensor and it should be included in the packet/record. if the driver does not know whether a sensor exists, then it should not include that sensor in the packet/record, and it should definitely not include that sensor with a value of None.
> 
> 2. The StdWXCalculate service should insert a value of None for a derived value only if it has values for all of the independent variables on which derived depends. for example, if a packet contains temperature=None and humidity=50 then a dewpoint=None is appropriate. if a packet does not contain temperature but contains humidity=50, then dewpoint=None is not appropriate; there should be no dewpoint in the packet.

The haphazard treatment of `None` has bugged me for a while. This is a sensible proposal. 

However, this means that all drivers will emit far fewer keys than they did before. Is this going to break anything? That is, is there a service that depends on `None` values appearing for any keys?

tkeffer, 2016-11-30:
Let's start with changing the vantage driver to emit the minimum and see how it goes.

tkeffer, 2016-12-02:
Branch `None` has a version of the Vantage driver that only emits a value if there is a known value for an observation.

It seems to work just fine.

Worth noting: here's the payoff of changing the drivers so they emit `None` if wind speed is zero. It's now possible to distinguish between a genuine null value (wind speed is zero), and a sensor that's just plain missing.

gjr80, 2016-12-29:
Hmm, have been pondering this with a driver I am writing for my PV inverter. Once the sun sets and the PV voltage to the inverter drops below some threshhold the inverter goes to sleep. No signal or anyhting, it just sleeps and ignores all attempts at contact. It wakes again around sun rise when the PV voltage rises above some threshhold. The driver only emits loop packets with weewx using software record generation. So what should the driver emit for say the `energy` field. The driver knows there is an inverter there and knows that it when it is on it provides data for the `energy` field. The driver also knows that at times the inverter sleeps and will provide no data whatsoever. The driver does not now when the inverter is asleep since the inverter does not respond to any polls and it does not signal 'I am going to sleep' before it does so. The driver makes an assumption that when the inverter ignores the driver it is becasuse the inverter is asleep.

So when the inverter is asleep should the driver emit the `energy` field with the value `None` or should it leave the `energy` field out of the loop packet altogether? At first I thought it was pretty clear, leave it out. But the more I think about it the more I am not so sure.

tkeffer, 2016-12-29:
The driver should emit `None`.  The hardware is still there; it's just non-functional. 

Leaving the field out entirely is appropriate for when the sensor just plain isn't there.

matthewwall, 2017-01-06:
should i apply the 'None' treatment to all drivers in the None branch?

also, the StdWXCalculate service needs to be evaluated (and probably modified) for proper 'None' behavior.  should i do that on the None branch as well?

tkeffer, 2017-01-06:
I feel like we are juggling too many branches right now. We have None, schema, weighted_summaries and (possibly) newskin.

I say we include the None, schema, and weighted_summaries changes into the next, V3.7, release. Can we consolidate their respective branches into one "development" branch, and work off of that?

gjr80, 2017-01-06:
Doesn't cause me a problem

gjr80, 2017-03-15:
Another use of None to consider is in the accumulator extractor functions. At the moment if a field is None in every loop packet in an acumulators lifetime, the sum extractor returns 0.0, should it return None? The last and avg extractors 'respect' this approach.

Traditionally the sum extractor is used for `rain`, and returning 0.0 for a period period where no rainfall data was reported makes sense. The practical case that I am dealing with is a solar inverter where the energy generated may be None (if the inverter is asleep) or 0.0 (the inverter is awake but there is insufficient sunshine to produce any electrical energy) or some value > 0.0 (if the inverter is awake and generating). In this case there is a distinction between None and 0.0 and this distinction is lost through the present accumulator behviour.

tkeffer, 2017-03-15:
I'd say the accumulator should return `None`.

The value `None` means "I don't know," which is definitely different from a sum of zero. In your case, the inverter was offline, so it produced an unknown amount of electricity, which may be zero, or it may not be. If the inverter is awake, but there is insufficient sun to produce electricity, then it should return zero.

Having said this, I worry about the accumulators returning `None` for a sum. The downstream algorithms may not be prepared to accept it.

gjr80, 2017-03-15:
My workaround to get my inverter running with 3.7.0 was to define a new extractor `sum_none` that returns `None` if `count==0` otherwise it returns the accumulator sum property. I set the extractor function in `weewx.conf` for the field concerned and all works well. Obviously I have an orphan `accum.py` (that has already caught me once overwriting changes). Could we have two sum extractors defined in `accum.py`, one as it is now that return 0.0 and another that can return `None`. The latter would be there if required but the default behaviour is to use the existing 0.0 sum extractor?

nellyt, 2017-08-02:
See also [https://groups.google.com/forum/#!topic/weewx-user/GAy8mnRO3N0] for related effects of this issue

tkeffer, 2017-08-07:
@gjr80 Is there any reason not to define `sum` as a property, which returns a value of `None` if `count` is zero? Of course, to make this work, the name of the current attribute `sum` would have to be changed to something else.

This way, there would be only one extractor for sum.

gjr80, 2017-08-07:
> @gjr80 Is there any reason not to define sum as a property, which returns a value of None if count is zero? Of course, to make this work, the name of the current attribute sum would have to be changed to something else.

I guess that would work, but as you said a couple of posts back will that create downstream issues? Thinking aloud. `rain` is the obvious candidate for any issues, `$day.rain.sum` at the moment always displays 0.0 if only 0.0 or `None` values are received from the driver for the day. Under the 'new' arrangement if the driver only outputs `None`, then `$day.rain.sum` would output `None` or `N/A`. Technically correct but does not seem intuitive. Though the more I think about it the more accepting it is to me - really why should `rain` be any different to `outTemp` in terms of what is displayed if there is no data.

Still thinking aloud, let's say the existing `sum` attribute is changed to `new_sum`, are their any implications for the `wsum` attribute and `avg` property? `wsum` as it is now cannot be `None`, so someone using the `new_sum` property may get `None`, 0.0 or a numeric whereas `wsum` will be 0.0 or numeric. Probably not an issue as `wsum` tends to be used in calculating other aggregates only. The existing `avg` property uses `wsum`, if `new_sum` is `None` aware it stands to reason `avg` will need to be too, do you then make `wsum` `None` aware as well?

What about `xsum` and `ysum` in class `VectorStats`? Also think some of the aggregate calcs in `weewx.manager` would need to be `None` aware as well? 

Have I read too much into what you propose, or is it much simpler then my interpretation? 

Hmm, a bit longer than I thought when I sat down to write this. I am happy to make a new branch in my fork and see how it goes? Though I won't be coming up with a new name for `sum` :-)

tkeffer, 2017-08-08:
I'm being pretty stupid. My proposal isn't going to work at all, because the daily summaries are used with SQL statements such as

```
"SELECT SUM(sum) FROM %(table_name)s_day_%(obs_key)s WHERE dateTime >= %(start)s AND dateTime < %(stop)s"
```
SQL cannot sum a bunch of null values and get anything sensible. 

Instead, we need to check the total count within the `DaySummaryManager.getAggregate()` function.

gjr80, 2017-08-08:
> SQL cannot sum a bunch of null values and get anything sensible.

Won't the sum of a bunch of nulls return a null and the sum of some nulls and some numerics return the sum of the numerics? I would have thought you could return a sensible result (`None` or the sum of the numerics) from that. Mind you I just did a few quick queries with SQLite, don't know if MySQL behaves the same.

tkeffer, 2017-08-08:
That's what sqlite does for me as well. But, if that's the case, why doesn't `$day.rain.sum` work?

gjr80, 2017-08-08:
Because the `sum` attribute is initialised as 0.0 and the extractor extracts the value of the `sum` attribute irrespective of `count`

tkeffer, 2017-08-08:
Of course. I've been spending too much time out in the sun.

gjr80, 2017-08-08:
> Of course. I've been spending too much time out in the sun.

Ha ha, plenty of sun here too but unfortunately too much wind mixed with it! You have given me a way ahead that seems worthwhile pursuing, I will see where it takes me. Have just about had enough playing with MQTT anyway.

gjr80, 2017-08-09:
Implementing `sum` as a property was easy enough:
- defined a `sum` property for each of the accumulators that returns the existing `sum` field if count>0 else None
- renamed the `sum` field to `nsum`
- the `nsum` field of each accumulator is now initialised as None instead of 0.0
- changed the sql create strings in wxmanager.py and manager.py to use `nsum` in lieu of `sum`
- changed the sqlDict select statements in DaySummaryManager to use `nsum` in lieu of `sum`

A default weeWX install running simulator with the loop packet rain field set to None runs without error. Day/week/month/year rainfall is shown as N/A as would be expected. Setting rain to 0.01 inch in each loop packet appears to display as expected, resetting it to None keeps the previous (non-None) total as expected.

I wonder though if anything needs be done with the other 'sums', ie `wsum`, `xsum`, `ysum`. As far as I know weeWX only uses `wsum` to calculate `avg` and the `avg` property already has an else clause that picks up `count=0` and returns None. `wsum` is not typically displayed by the user so leaving it as is should not cause an issue (though as history would show I am sure someone will want to use it directtly). Changing `wsum` to initialise to None rather than 0.0 is easier than it was for `sum`. The other issue is vectors. I don't believe the vectors have any code to handle/return None. `xsum` and `ysum` are initialised to 0.0 so the vector is returned as 0.0 speed, 0.0 direction. Should the returned vector be (None, None); common sense would say if you have had no (for example) `windSpeed`/`windDir` data, or the data was None/None, then you should return (None, None) not (0,0). Not a comprehensive analysis but I guess the question now is how far do we take this?

tkeffer, 2017-08-10:
The problem with this approach is that the semantics of `sum` changes in mid-stream in the database, and I sure don't want to go through that database patching exercise again!

An alternative approach is to make the change in method `getAggregate()`. Basically, if `count` is zero, then `sum` is `None`. Otherwise, you return the actual sum.

gjr80, 2017-08-11:
> The problem with this approach is that the semantics of `sum` changes in mid-stream in the database, and I sure don't want to go through that database patching exercise again!

Hmm, I guess that is why you suggested this a few days back, my turn for too much sun. Then again, you could take the alternative view, after the 3.7.0 release we are now well versed in patching the database :-)

gjr80, 2017-08-11:
Ah, mention of patching the database got me thinking, if implementing `sum` as a property we will need to rename the `sum` field to something else. Not a problem for a new user but existing users will require a(nother) database patch won't they? If this is correct do we really want to go there?

tkeffer, 2017-08-11:
Forget the accumulators. Use them as is. Make the changes in `getAggregate()`. Something like (NOT TESTED):

```
diff --git a/bin/weewx/manager.py b/bin/weewx/manager.py
index f20ee2c..7755a6f 100644
--- a/bin/weewx/manager.py
+++ b/bin/weewx/manager.py
@@ -1112,7 +1112,7 @@
                               "sum = (SELECT MAX(sum) FROM %(table_name)s_day_%(obs_key)s WHERE dateTime >= %(start)s AND dateTime <%(stop)s)",
                'gustdir'    : "SELECT max_dir FROM %(table_name)s_day_%(obs_key)s  WHERE dateTime >= %(start)s AND dateTime < %(stop)s AND " \
                               "max = (SELECT MAX(max) FROM %(table_name)s_day_%(obs_key)s WHERE dateTime >= %(start)s AND dateTime < %(stop)s)",
-               'sum'        : "SELECT SUM(sum) FROM %(table_name)s_day_%(obs_key)s WHERE dateTime >= %(start)s AND dateTime < %(stop)s",
+               'sum'        : "SELECT SUM(sum), SUM(count) FROM %(table_name)s_day_%(obs_key)s WHERE dateTime >= %(start)s AND dateTime < %(stop)s",
                'count'      : "SELECT SUM(count) FROM %(table_name)s_day_%(obs_key)s WHERE dateTime >= %(start)s AND dateTime < %(stop)s",
                'avg'        : "SELECT SUM(wsum),SUM(sumtime) FROM %(table_name)s_day_%(obs_key)s WHERE dateTime >= %(start)s AND dateTime < %(stop)s",
                'rms'        : "SELECT SUM(wsquaresum),SUM(sumtime) FROM %(table_name)s_day_%(obs_key)s WHERE dateTime >= %(start)s AND dateTime < %(stop)s",
@@ -1298,10 +1298,14 @@
         
         # Do the required calculation for this aggregat type
         elif aggregate_type in ['min', 'maxmin', 'max', 'minmax', 'meanmin', 'meanmax', 
-                               'maxsum', 'sum', 'gustdir']:
+                               'maxsum', 'gustdir']:
             # These aggregates are passed through 'as is'.
             _result = _row[0]
         
+        elif aggregate_type == 'sum':
+            # Return None if count is zero.
+            _result = _row[0] if _row[1] else None
+            
         elif aggregate_type in ['mintime', 'maxmintime', 'maxtime', 'minmaxtime', 'maxsumtime',
                                'count', 'max_ge', 'max_le', 'min_ge', 'min_le', 'sum_ge']:
             # These aggregates are always integers:
```

gjr80, 2017-08-12:
> Make the changes in `getAggregate()`. Something like (NOT TESTED):

I have that, well similar but (was) not quite as elegant, you also need to apply the same treatment to `maxsum`.

> Forget the accumulators.

I don't think you can. Consider field `rain`, the problem with only changing `getAggregate()` is that if the driver is putting field `rain` == `None` in each loop packet the accumulator will 'extract' an archive record from the accumulator with field `rain` == 0.0 not `None`. The reason being that the `sum` field in the accumulator is initialised to 0.0 and the extractor extracts the `sum` field, which after an archive period of loop packets with `rain` == `None` leaves the `sum` field == 0.0. Hence the archive record has `rain` == 0.0. When the daily summary is updated with this archive record, `count` is incremented and our `getAggregate()` changes cannot return `None` as `count` is now > 0. I have worked around this with the following changes to `accum.py`:
```
diff -u /home/gary/src/weewx/bin/weewx/accum.py /home/weewx/bin/weewx/accum.py
--- /home/gary/src/weewx/bin/weewx/accum.py     2017-08-11 11:17:12.304000000 +1000
+++ /home/weewx/bin/weewx/accum.py      2017-08-12 10:35:50.220000000 +1000
@@ -94,6 +94,14 @@
     def avg(self):
         return self.wsum / self.sumtime if self.count else None

+    @property
+    def none_sum(self):
+        return self.sum if self.count else None
+
+#===============================================================================
+#                             VectorStats
+#===============================================================================
+
 class VecStats(object):
     """Accumulates statistics for a vector value.

@@ -194,6 +202,10 @@
         return self.wsum / self.sumtime if self.count else None

     @property
+    def none_sum(self):
+        return self.sum if self.count else None
+
+    @property
     def rms(self):
         return math.sqrt(self.wsquaresum / self.sumtime) if self.count else None

@@ -366,7 +378,7 @@
         record['windGustDir'] = self[obs_type].max_dir

     def extract_sum(self, record, obs_type):
-        record[obs_type] = self[obs_type].sum
+        record[obs_type] = self[obs_type].none_sum

     def extract_last(self, record, obs_type):
         record[obs_type] = self[obs_type].last
```
ie new `none_sum` property in the accumulator and chnage the default sum extractor function. I don't believe these chnages chnage the context of the `sum` field and will be backward compatible. The only possibility of an issue is if someone is relying on a '`None` fed field' to produce 0.0 as it's sum or if (say) an extension is reaching into the acumulators and using the `sum` field. 

This is giving me archive records with `rain` == `None` when all loop have `rain` == `None`.

tkeffer, 2017-08-12:
Thanks for the explanation. I get it now.

Suggestion: Change the name of the attribute `sum` to, say, `total`, then name the property `sum` (instead of `none_sum`). This is less likely to break an existing extractor, which might be expecting the name of `sum`. 

Alternatively, keep the name `sum`, but use setters and getters.

gjr80, 2017-08-12:
If we change the name of any attribute are we not going to end up patching the db again?

tkeffer, 2017-08-12:
I don't think so. The name of the attribute in the accumulator and the name of the column in the daily summaries are independent. There's no code that queries the accumulator for attribute names, then retrieves the same name from the database. Or *vice versa*.

 I just gave them the same names to reduce confusion. 

Of course, at this point, I'm not sure of much of anything. :-)

gjr80, 2017-08-12:
Like I said my head was spinning. I guess we suck it and see; will make the changes, drop in an existing db and see if anything complains.

tkeffer, 2017-08-12:
Second thought, don't use the name `total`. Use `_sum`. Less likely to cause confusion.

gjr80, 2017-08-12:
Good idea

gjr80, 2017-08-13:
Changed accumulator `sum` attribute to `_sum` and implemented `sum` property. Seems to be working OK with a database of historical data. Will do some more in-depth testing (as opposed to go/nogo testing done so far) later tonite or tomorrow. In the meantime, and even though the changes are only a handful of lines in `accum.py` and `manager.py`, I have created the `none_sum` branch on GitHub with these chnages. I know the `rainRate` calcs in `wxservices.py` need some touching up as they return 0.0 when `rain` is `None`. Hopefully that will not upset the natives.

vbabiy, 2017-12-02:
Should this fix:

```
Dec  2 16:24:59 weewx weewx[12316]: restx: raw packet: altimeter: 30.2948542808, appTemp: 49.1138072283, barometer: 30.3139428559, channel: None, cloudbase: 3212.45042708, dateTime: 1512231899, dewpoint: 41.2833361071, heatindex: 52.3, humidex: 52.3, inDewpoint: None, inTemp: 72.95, maxSolarRad: None, outHumidity: 66, outTemp: 52.3, outTempBatteryStatus: 0, pressure: 29.5373699969, rainRate: 0, rssi: 3, rxCheckPercent: 100.0, sensor_battery: 0, sensor_id: 3309, usUnits: 1, windchill: 52.3, windGustDir: None, windSpeed: 2.16448979085
Dec  2 16:24:59 weewx weewx[12316]: restx: cached packet: altimeter: 30.2948542808, appTemp: 49.1138072283, barometer: 30.3139428559, channel: None, cloudbase: 3212.45042708, dateTime: 1512231899, dewpoint: 41.2833361071, heatindex: 52.3, humidex: 52.3, inDewpoint: None, inTemp: 72.95, maxSolarRad: None, outHumidity: 66, outTemp: 52.3, outTempBatteryStatus: 0, pressure: 29.5373699969, rain: 0.0, rain_total: 0.0, rainRate: 0, rssi: 3, rxCheckPercent: 100.0, sensor_battery: 0, sensor_id: 3309, usUnits: 1, windchill: 52.3, windDir: 315.0, windGustDir: None, windSpeed: 2.16448979085
Dec  2 16:24:59 weewx weewx[12316]: acurite: next read in 18 seconds
Dec  2 16:24:59 weewx weewx[12316]: restx: Ambient: url: http://rtupdate.wunderground.com/weatherstation/updateweatherstation.php?action=updateraw&ID=KNCINDIA32&PASSWORD=XXX&softwaretype=weewx-3.8.0&realtime=1&baromin=30.314&dewptf=41.3&humidity=066&dateutc=2017-12-02%2016%3A24%3A59&rtfreq=2.5&dailyrainin=0.00&winddir=315&tempf=52.3&windspeedmph=2.2&rainin=0.00
Dec  2 16:25:17 weewx weewx[12316]: acurite: Found station at bus= device=
Dec  2 16:25:17 weewx weewx[12316]: restx: raw packet: altimeter: None, appTemp: None, barometer: None, channel: None, cloudbase: None, dateTime: 1512231918, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None, outTempBatteryStatus: 0, pressure: None, rain: 0.0, rain_total: 0.0, rainRate: 0, rssi: 3, rxCheckPercent: 100.0, sensor_battery: 0, sensor_id: 3309, usUnits: 1, windchill: None, windDir: 292.5, windGustDir: None, windSpeed: 1.65011743945
Dec  2 16:25:17 weewx weewx[12316]: restx: cached packet: altimeter: None, appTemp: None, barometer: None, channel: None, cloudbase: None, dateTime: 1512231918, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, inTemp: 72.95, maxSolarRad: None, outHumidity: 66, outTemp: 52.3, outTempBatteryStatus: 0, pressure: None, rain: 0.0, rain_total: 0.0, rainRate: 0, rssi: 3, rxCheckPercent: 100.0, sensor_battery: 0, sensor_id: 3309, usUnits: 1, windchill: None, windDir: 292.5, windGustDir: None, windSpeed: 1.65011743945
```

vbabiy, 2017-12-02:
I should note, I am on 3.8.0

tkeffer, 2018-04-28:
Any progress on this one?