weewx/weewx #179: parse passwords as strings

matthewwall, 2016-11-12:
weewx should parse any characters for a password field as a string, not as a list.  otherwise, this causes problems:

`password = hdw,!ge`

this affects the ftp password, probably others.

we might need a weeutil.parse_as_password function to make this consistent throughout weewx and any extensions.

tkeffer, 2016-11-26:
There's already a function `weeutil.weeutil.list_as_string()` that does this, except it adds a space after every comma. It would work with your example above.

The problem is that the conversion to a list happens within `ConfigObj`, so we are reduced to converting it back. Unfortunately, we don't know where to put any spaces.

I'm afraid getting this right 100% of the time would require a change to `ConfigObj`.

tkeffer, 2016-11-30:
Matthew and I talked this one out and concluded the way to go is to offer a `parse_as_password` function, which, if handed a list, will do its best to reconstruct the original string, then emit a warning to the log.

tkeffer, 2016-12-03:
So, remind me why we can't just tell the user to put any password in quotes.

~~~python
>>> from configobj import ConfigObj
>>> from StringIO import StringIO
>>> print ConfigObj(StringIO("""password='a, password' """))
{'password': 'a, password'}
~~~

For example:

~~~ini
    [[Wunderground]]
        # This section is for configuring posts to the Weather Underground.
        
        # If you wish to do this, set the option 'enable' to true,
        # and specify a station (e.g., 'KORHOODR3') and password.
        enable = false
        station  = replace_me
        # To guard against parsing errors, put your password in quotes:
        password = "replace_me"
~~~

matthewwall, 2016-12-03:
because they won't? :)

since there are characters other than comma that will cause grief (e.g., #) i'm inclined to agree with the just-put-it-in-quotes approach.

gotta see how that affects wee_debug...

tkeffer, 2016-12-03:
By putting the `replace_me` in quotes, I'm hoping to encourage retaining them.

tkeffer, 2016-12-06:
Sort of fixed in commit https://github.com/weewx/weewx/commit/2d6d186c065bba5bcdb05ea4865ee7acef28fe28

closed 2016-12-06 by tkeffer