weewx/weewx #180: restx needs to be refactored for ssl connections

matthewwall, 2016-11-14:
a typical way to establish an ssl connection using urllib is to specify an ssl context as an argument to urllib2.urlopen

unfortunately, urllib2.urlopen is buried in post_request, which is in turn buried in post_with_retries

we probably need a refactoring that passes **args through so that a derived can have control over what is passed to urlopen

this came up in the context of sending data to an influx server over https connection using the influx extension

matthewwall, 2016-11-14:
nevermind.

i just overrode the post_request method.

the restful stuff has turned into one very complicated beast.  that and database continue to be the most confusing abstractions for me, probably due to the sheer number of options and permutations.

closed 2016-11-14 by matthewwall

tkeffer, 2016-11-14:
I agree. The complexity in restx is because of a desire to maximize code reuse. I think if it was documented better, with cookbook "how to" instructions, it wouldn't be so bad.

The complexity in the databases comes from two places: the complicated aggregation strategies, and the need to create a simple binding lookup (function `db_lookup()`) for SLEs. The former could be factored out and put someplace else. Then at least you wouldn't get distracted by it when looking through `manager.py`. Not sure what could be done about the latter. I played around with lots of stuff when I wrote it, and that's the best I could come up with.